Koenkk/zigbee2mqtt #24745: GLEDOPTO is reporting "Unsupported device" - Support was dropped after update of zigbee2mqtt

EIBHomeControl, 2024-11-13:
### What happened?

please see https://github.com/Koenkk/zigbee2mqtt/issues/4134

after update to 1.40.1 the GLEDOPTO LED drive is detected as unkown device

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

Sonoff

### Setup

plain, RPi 4

EIBHomeControl, 2024-11-13:
[image]

Koenkk, 2024-11-14:
What do you see in the "About" tab of the device in the z2m frontend?

EIBHomeControl, 2024-11-15:
[image]

Koenkk, 2024-11-17:
Could you try with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition) instead?

EIBHomeControl, 2024-11-17:
Just found the model number. Its GL-C-007-2ID
If i interview the device again it show following info
[image]

Koenkk, 2024-11-18:
Could you provide the data/database.db entry of this device?

EIBHomeControl, 2024-11-19:
{"id":15,"type":"Router","ieeeAddr":"0x00124b0019c5dea7","nwkAddr":10296,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO","epList":[10,11,13],"endpoints":{"10":{"profId":260,"epId":10,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GLEDOPTO","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":1}},"lightingColorCtrl":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"11":{"profId":260,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorMode":1,"currentX":29881,"currentY":26946,"enhancedCurrentHue":9472,"currentSaturation":187,"colorTemperature":293}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":1}},"genBasic":{"attributes":{"modelId":"GLEDOPTO","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2"}}},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2","zclVersion":1,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1732001406905}

Koenkk, 2024-11-20:
fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-11-20 by Koenkk