Koenkk/zigbee2mqtt #24747: [New device support]:  Tuya 1 gang light switch _TZE200_ojtqawav

ehwwong, 2024-11-13:
### Link

https://vi.aliexpress.com/item/1005006390186459.html?spm=a2g0o.productlist.main.19.322e6025t285xq&algo_pvid=e3d473d1-bb3c-471a-876c-c362d27d9ad4&algo_exp_id=e3d473d1-bb3c-471a-876c-c362d27d9ad4-9&pdp_npi=4%40dis%21SGD%2119.75%219.87%21%21%2114.34%217.17%21%4021411ec417314981626461675e79a6%2112000036995300048%21sea%21SG%210%21ABX&curPageLogUid=Dr09s2MgzdHW&utparam-url=scene%3Asearch%7Cquery_from%3A

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0xe498bbe02e07416f","nwkAddr":27753,"manufId":4660,"manufName":"_TZE200_ojtqawav","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,61184],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ojtqawav","powerSource":1,"zclVersion":2,"appVersion":100,"stackVersion":2,"hwVersion":3,"dateCode":"20210625","swBuildId":"600+TZZUT11HH102"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":2,"hwVersion":3,"dateCode":"20210625","swBuildId":"600+TZZUT11HH102","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1731498098119}

### Zigbee2MQTT version

1.41.0

### Comments

Following my earlier report, from the same seller.. whether its 1 gang or 4 gang, it doesnt work. Not sure whether this is a new firmware? 

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_ojtqawav',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Unfortunately no external definition available now

ehwwong, 2024-11-13:
I bought like 4 of this switch and none of them works with ZHA or z2m... feeling very lost right now.. hopefully someone can point me in the right direction on how to "self help"

Koenkk, 2024-11-14:
Can you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE200_ojtqawav` to `_TZE204_ojtqawav`
- Start Z2M and see if the device now works

h0m3d0m0, 2024-11-16:
@Koenkk 
i have the same problem  #24650
Try your solution and now i can power the switch on and off but it's not marked tuya anymore
Anyway for now i can start my automations again so thx

[image]

Koenkk, 2024-11-18:
Added, you can change it back after the next z2m release (it will be detected as Tuya then)

closed 2024-11-18 by Koenkk