Koenkk/zigbee2mqtt #24754: Adding New Devices Is Difficult - Regularly Getting "Failed to remove rejected device: AREQ - ZDO - mgmtLeaveRsp..." Errors

Inshakoor, 2024-11-13:
### What happened?

About 6-8 weeks ago or so, devices disappeared from Z2M.  That had never happened before but I figured I'd just re-add them.  I reset the devices and enabled joining and some came back quickly and others took 5-10 tries and some still won't rejoin.  I noticed recurring "Failed to remove rejected device: AREQ - ZDO - mgmtLeaveRsp..." errors.  I continue to try to add things with very little luck, especially in the first few tries.

I researched the error and found similar issues but no solutions that worked.

I downgraded to 1.39.0 (wanted to go back when I knew it was working) but Z2M wouldn't launch with error "Attempt to access memory outside buffer bounds"

As an aside, since I setup Z2M 8 years ago, there have always been about 8 Zigbee devices nearby that always want to add and I have no idea what they are (meters perhaps?).  They have been in the block list since then.

### What did you expect to happen?

I expected the Zigbee devices that I'm trying to add, to successfully add within a reasonable amount of time.

### How to reproduce it (minimal and precise)

I enable joining and set a device to pairing mode and see if it actually joins, much less joins successfully

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

CC26X2R1

### Setup

Docker Container on Raspberry Pi 4

### Debug log

[log.log](https://github.com/user-attachments/files/17736636/log.log)

toto9262, 2024-11-26:
Hello i have the same problem

toto9262, 2024-11-30:
my problem is solved, I reconnected my Lidl HG08673

Inshakoor, 2024-12-01:
I still have devices that I can't get to rejoin, and devices that seem to become unresponsive.  During my searching for solutions, I did read that sometimes you can enable joining and that some unresponsive devices will then become responsive.  I have done this a few times and that has worked, but I still have devices that won't join.

Is there something I can do so that I don't have to periodically turn on enable joining when devices become unresponsive?  I've thought about rebuilding a whole new container and restoring the data folder but I imaging the problems will just follow.  I'm open to any kind of housekeeping ideas or anything of the like that may help.  Thanks all.

closed 2025-02-07 by github-actions[bot]