Koenkk/zigbee2mqtt #24755: Wrong current_position value (inverted values)

marcopaggioro, 2024-11-13:
### What happened?

For a few days now I have seen the "open" and "closed" values ​​of my roller shutters reversed. When the roller shutter is completely up (raised) I see "closed" from the interface and when the roller shutter is completely down (lowered) I see "open" from the interface. Since the command detects that the roller shutter is already lowered, the button to lower it is deactivated and vice versa

[image]

When the shutter is up I expect current_position to 100 ([as described here](https://github.com/piitaya/lovelace-mushroom/issues/1581#issuecomment-2473493237)) but It is:
```
current_position: 0
device_class: shutter
icon: mdi:window-shutter
friendly_name: Tapparella studio
supported_features: 15
```

Device: https://it.aliexpress.com/item/1005002353449152.html?spm=a2g0o.productlist.main.11.77f67266RdXYiq&algo_pvid=c5202f23-937a-42b3-83bf-e8c2258dab87&algo_exp_id=c5202f23-937a-42b3-83bf-e8c2258dab87-5&pdp_npi=4%40dis%21EUR%2128.22%2114.39%21%21%2129.29%2114.94%21%40211b816617315194529965001e2f50%2112000030751917120%21sea%21IT%21820641017%21X&curPageLogUid=J8jaKcCSbAUO&utparam-url=scene%3Asearch%7Cquery_from%3A

### What did you expect to happen?

Until a few days ago (maximum 2/3 weeks) when the roller shutter was completely up (raised) I could see "open" from the interface

### Zigbee2MQTT version

1.41.0 commit: cc31cea

### Adapter firmware version

20210708

### Adapter

https://www.amazon.it/gp/product/B09KXTCMSC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

### Setup

Z2M and HA with Docker in Debian 12.8 and SBC is LibreComputer Alta AML-A311D-CC

### Debug log

[log.log](https://github.com/user-attachments/files/17736938/log.log)

BeFygo, 2024-11-13:
I just noticed the same for my "_TZ3000_1dd0d5yi" these devices have been working correctly for over a year and now (assuming with the upgrade to V1.41.0) it looks like the feedback position is reverse.

marcopaggioro, 2024-11-13:
@BeFygo Did you open another bug?
have you tryed to downgrade to 1.40 to see If the problem was introduced with 1.41?

marcopaggioro, 2024-11-13:
I can confirm that the bug is NOT present in 1.40.2

Koenkk, 2024-11-14:
It has been fixed in latest-dev indeed and will be included in the next z2m release.

BeFygo, 2024-11-15:
I also confirm that its fixed in latest-dev.
Great work, like usual :)

closed 2024-11-17 by Koenkk