Koenkk/zigbee2mqtt #24757: npm start doesn't work and crashes without any log

chuck2kill, 2024-11-13:
### What happened?

when i launch the command `npm start` in the zigbee2mqtt directory, it crashes, always at the same line:

[image]

after that, the raspberry seems to crash and I have to restart it

I checked the rights and it should be ok as the user as the permissions in all the directory zigbee2mqtt, and he is member of the dialout group, so it should open the usb

I managed to launch it once, but it seems impossible to reproduce

thanks for the help

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20211115

### Adapter

cc2531 stick

### Setup

raspberry pi 3

lux73, 2024-11-14:
CC2531 is considered as crap in 2024 and its not worth to deal with it

Please update your Coordinator to an CC2652x based Stick

Ricc68, 2024-11-16:
Can it be related to #22881?

chuck2kill, 2024-11-25:
not sure
I ordered a new stick based on CC2652, I'll let you know if it fixes it.

closed 2025-02-01 by github-actions[bot]