Koenkk/zigbee2mqtt #24759: [New device support]: Tuya Zigbee Water Leak Detector

MilesTEG1, 2024-11-13:
### Link

https://fr.aliexpress.com/item/1005006512619030.html

### Database entry

{"id":74,"type":"EndDevice","ieeeAddr":"0xa4c1382c55337701","nwkAddr":59608,"manufId":4742,"manufName":"_TZ3000_upgcbody","powerSource":"Battery","modelId":"SNZB-05","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"ssIasZone":{"attributes":{"iasCieAddr":"0x847127fffedfe3cd","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":114}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x847127fffedfe3cd","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x847127fffedfe3cd","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1731510955140,"checkinInterval":3600}

### Zigbee2MQTT version

1.41.0

### Comments

I paired the device, a [Tuya Zigbee Water Leak Detector](https://fr.aliexpress.com/item/1005006512619030.html) that is a [Tuya TS0207_water_leak_detector_1](https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector_1.html#tuya-ts0207-water-leak-detector-1), but it doesn't been supported, nor detected as TS0207_water_leak_detector_1.
Instead, Z2M says that it's unsupported and this:
- Modèle Zigbee: SNZB-05
- Constructeur Zigbee: _TZ3000_upgcbody

It should be detected as TS0207_water_leak_detector_1.

### External definition

```shell
const {battery, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SNZB-05'],
    model: 'SNZB-05',
    vendor: '_TZ3000_upgcbody',
    description: 'Automatically generated definition',
    extend: [battery(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

alarm_1 is working: a water leak is detect when sensor in water.
alarm_2 isn't working: nothing happens when sensor in water.

tamper: I don't know how to trigger this.
battery_low: my battery is at 57%, so I don't know if it's working for now.
battery indicate a %, so it seems to be working.

Koenkk, 2024-11-14:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-11-14 by Koenkk

MilesTEG1, 2024-11-14:
@Koenkk 
I just switched to the Edge version on HA Addon (after refreshing the repository as described in the Edge version doc).
I put my configuration from stable version, and launched it.
No problems, all my devices are working fine.

But... my leak sensor isn't detected as a [`TS0207_water_leak_detector_1`](https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector_1.html#tuya-ts0207-water-leak-detector-1) as it should be, but it is detected as a [IH-K665](https://www.zigbee2mqtt.io/devices/IH-K665.html#aubess-ih-k665) which is not.

Can you do a correction of that please?
and reopen this issue for now until this is fixed?

By the way, the exposes sensors are OK.
[image]
If I simulate a water leak, it's being detected:
[image]

Thanks very much.

Koenkk, 2024-11-18:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

MilesTEG1, 2024-11-19:
Thanks you very much! 😇

Peter915671, 2024-12-02:
Isn’t SNZB-05 Sonoff?
Wouldn’t it be better to use that name instead of Tuya?

cypherbits, 2024-12-28:
Upgraded to 1.42.0 and my device (https://es.aliexpress.com/item/1005005942065429.html) changed its image to this commit one. I mean, my device is round like the link...

vincentscode, 2025-01-17:
@Koenkk There appears to have been a regression due to https://github.com/Koenkk/zigbee-herdsman-converters/pull/8385 causing some devices to be no longer detected as supported:
```
Received message from unsupported device with Zigbee model 'SNZB-05' and manufacturer name '_TZ3000_upgcbody'
```
Should this issue be reopened or a new one be created?