Koenkk/zigbee2mqtt #24768: OSRAM Smart+ Switch Mini Interview failed

xTim18, 2024-11-14:
### What happened?

When paring the zigbee2mqtt supported [OSRAM AC0251100NJ/AC0251600NJ/AC0251700NJ](https://www.zigbee2mqtt.io/devices/AC0251100NJ_AC0251600NJ_AC0251700NJ.html#osram-ac0251100nj-ac0251600nj-ac0251700nj) switch the paring gets initiated, and ends up with the error massages below.
[image]
Anyone smarter then me with an idea how to fix that beside the hundert reset and retry to connect?

### What did you expect to happen?

Before moving to a new flat I had two the switches running on ZHA fairly well, now I bought some in addition and all of them expiriance the same behaviour. Usually they should be recogniced as a switch with 3 buttons and 3 functions (press, hold, double tap) per button.

### How to reproduce it (minimal and precise)

Everytime the device has been resettet and deleted from Z2M the error occurs whenever paring the switches.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Add-on on Home Assistant OS running on Proxmox VM

Great-Chart, 2024-11-14:
I had a comparable issue with a Develco smart plug that mysteriously dropped off my network and every attempt to re-pair failed. Usually a force delete in Z2M and reset the device sorts things.
But not this time. I found some advice to force remove then restart Z2M addon. 

Not sure that worked first time but I also took it to the coordinator and also attempted to pair directly to it. I recall it seemed to flash up with a number of pair attempts and I left it alone long enough for the pairing yo be successful.
Generally doing something different and leaving it be can have results - but like you are experiencing it's quite frustrating and seemingly for no good reason!

I'll see if I can find the link to the suggestion I found as it did have a hint as to what the force delete and Z2M accomplished

xTim18, 2024-11-18:
Actually doing things different helped. I layed the switch tip to tip with my zigbee coordinator. Lets see if the range of those things is only this poor for initialising or during run time too. But pairing worked - For now. Thanks man!

closed 2024-11-18 by xTim18