Koenkk/zigbee2mqtt #24770: Danalock V3 does not update state with latest FW 0.27.0

FredyReady, 2024-11-14:
### What happened?

I have been using Danalock V3 for over a year and half now and without major issues. I have just updated to the newest firmware 0.27.0 via app due to issues with calibration. The new firmware has more detailed calibration - great.

The firmware for zigbee module in Z2M is latest - 20231030.

Now the issue, **since the firmware update the state of the lock does not update automatically.**
I can control the lock without issue, but the state does not update, I have to trigger manual state update via Z2M UI then the state updates.

I do suspect that the firmware version for Z2M OTA might be outdated, when I access [Danalock website](https://danalock.com/zigbee-firmware-update?customize_changeset_uuid=#download ) latest firmware is:

"ZigbeeXM_101-029_E1_DanalockV3_17.7.0_20240201155008-secureboot"

It seems there has been also zigbee update in 2024.

There are steps on how to update the Zigbee firmware manually but I do not dare to do that as it might not be compatible with Z2M anymore and there is no way to revert back to 20231030.

Screenshots from Danalock app and Z2M UI when lock was unlocked via Z2M - State does update in the app, but not Z2M. If I reload the state it updates in 3s.

[image]
[image]

### What did you expect to happen?

Danalock state will udpateautomatically after it changes either via app or Z2M or by manual operation.

### How to reproduce it (minimal and precise)

Update the lock to latest FW 0.27.0 via app.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Add-on on Home Assistant OS on Intel NUC

BeFygo, 2024-11-15:
I've been using FW build 20240309 as shown in zigbee2mqtt (I assume its the zigbee FW) for some months without issues,
today I updated the device to FW 0.27.0 with the android app and tested without issues,
To be sure I used the windows APP to update the lock and with testing no issues so far.

Using zigbee2mqtt.latest-dev from today

FredyReady, 2024-11-18:
> I've been using FW build 20240309 as shown in zigbee2mqtt (I assume its the zigbee FW) for some months without issues, today I updated the device to FW 0.27.0 with the android app and tested without [...]

Did you get the 20240309 OTA via Z2M or manually via the Danalock firmware updater like [here](https://danalock.com/zigbee-firmware-update)?

BeFygo, 2024-11-18:
> Did you get the 20240309 OTA via Z2M or manually via the Danalock firmware updater like [here](https://danalock.com/zigbee-firmware-update)?

I actually don't remember, I've done this several months back.
My guess is via Z2M.

FredyReady, 2024-11-18:
Weird, Z2M is not offering me this update OTA...

Oliviakrkk, 2024-11-24:
I have the same problem - I updated via Danalock app.

BeFygo, 2024-11-24:
> Weird, Z2M is not offering me this update OTA...

Then I think I just followed the guide from danalock
https://danalock.com/zigbee-firmware-update

FredyReady, 2024-11-26:
Updated to the latest FW via Danalock update utility, Z2M still reports FW build "20231030".

Issue is gone, lock reports its state as it should, automatically.

**Result**
Update via Danalock utility helped and the latest FW is compatible with Z2M.

Thanks everyone for help.

closed 2024-11-26 by FredyReady