Koenkk/zigbee2mqtt #24775: Bosch Twinguard and Bosch Smoke Detector 2

leonw-04, 2024-11-14:
### What happened?

Good evening,
I have tried to add the Bosch Twinguard and also the Bosch Smoke Detector 2 to Z2M. I have also tried several of these devices. The interview is started, but it fails with: 'Error: Interview failed because can not get node descriptor'. According to the wiki, both devices are supported. I have also tried many places in the house. I tried it on the stable branch and dev branch and with ezsp and ember

### What did you expect to happen?

Successful interview

### Zigbee2MQTT version

1.41.0-dev commit: 36e452b

### Adapter firmware version

7.4.4 [GA]

### Adapter

Sonoff Dongle-E

### Setup

Home Assistant Add-on

### Debug log

[45df7312_zigbee2mqtt_edge_2024-11-14T21-24-37.098Z.log](https://github.com/user-attachments/files/17758171/45df7312_zigbee2mqtt_edge_2024-11-14T21-24-37.098Z.log)
ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=48 Len=8]
[2024-11-14 22:22:45] [34mdebug[39m: 	zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY], [target=5068]
[2024-11-14 22:22:45] [32minfo[39m: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "5068".
[2024-11-14 22:22:45] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-11-14 22:22:45] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-11-14 22:22:45] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=0](frmRx=0) Added to rxQueue
[2024-11-14 22:22:45] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=1)
[2024-11-14 22:22:45] [34mdebug[39m: 	zh:ember:ezsp: <=== [CBFRAME: ID=128:"INCOMING_ROUTE_ERROR_HANDLER" Seq=48 Len=8]
[2024-11-14 22:22:45] [34mdebug[39m: 	zh:ember:ezsp: ezspIncomingRouteErrorHandler(): callback called with: [status=MAC_UNKNOWN_HEADER_TYPE], [target=5068]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=1](frmRx=1) Added to rxQueue
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=1)
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:ezsp: <=== [CBFRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=48 Len=18]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=14381], [sourceEui=0x001e5e09029fc24a], [lastHopLqi=108], [lastHopRssi=-73], [relayCount=0], [relayList=]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=2](frmRx=2) Added to rxQueue
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=1)
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:ezsp: <=== [CBFRAME: ID=36:"TRUST_CENTER_JOIN_HANDLER" Seq=48 Len=19]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:ezsp: ezspTrustCenterJoinHandler(): callback called with: [newNodeId=11345], [newNodeEui64=0x000d6f0019107c11], [status=STANDARD_SECURITY_UNSECURED_JOIN], [policyDecision=USE_PRECONFIGURED_KEY], [parentOfNewNodeId=14381]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:controller: Device '0x000d6f0019107c11' joined
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:controller: Device '0x000d6f0019107c11' accepted by handler
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:controller: Device '0x000d6f0019107c11' is already in database with different network address, updating network address
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:controller: Not interviewing '0x000d6f0019107c11', completed 'false', in progress 'true'
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=3](frmRx=3) Added to rxQueue
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=1)
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:ezsp: <=== [CBFRAME: ID=36:"TRUST_CENTER_JOIN_HANDLER" Seq=48 Len=19]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:ezsp: ezspTrustCenterJoinHandler(): callback called with: [newNodeId=11345], [newNodeEui64=0x000d6f0019107c11], [status=STANDARD_SECURITY_UNSECURED_JOIN], [policyDecision=USE_PRECONFIGURED_KEY], [parentOfNewNodeId=14381]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:controller: Device '0x000d6f0019107c11' joined
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:controller: Device '0x000d6f0019107c11' accepted by handler
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:controller: Not interviewing '0x000d6f0019107c11', completed 'false', in progress 'true'
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=4](frmRx=4) Added to rxQueue
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=1)
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:ezsp: <=== [CBFRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=48 Len=20]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=52279], [sourceEui=0x90395efffe0a8f8c], [lastHopLqi=96], [lastHopRssi=-76], [relayCount=1], [relayList=14381]
[2024-11-14 22:22:47] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[... 287 lines cut ...]
[2024-11-14 22:23:10] [34mdebug[39m: 	zh:controller:device: Interview - quirks did not match
[2024-11-14 22:23:10] [34mdebug[39m: 	zh:controller:device: Interview - failed for device '0x000d6f0019107c11' with error 'Error: Interview failed because can not get node descriptor ('0x000d6f0019107c11')'
[2024-11-14 22:23:10] [34mdebug[39m: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2024-11-14 22:23:10] [31merror[39m: 	zh:controller: Interview failed for '0x000d6f0019107c11 with error 'Error: Interview failed because can not get node descriptor ('0x000d6f0019107c11')'
[2024-11-14 22:23:10] [31merror[39m: 	z2m: Failed to interview '0x000d6f0019107c11', device has not successfully been paired
[2024-11-14 22:23:10] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0019107c11","ieee_address":"0x000d6f0019107c11","status":"failed"},"type":"device_interview"}'
[2024-11-14 22:23:13] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-11-14 22:23:13] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=4)
[2024-11-14 22:23:13] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=3](frmRx=3) Added to rxQueue
[2024-11-14 22:23:13] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=4)
[2024-11-14 22:23:13] [34mdebug[39m: 	zh:ember:ezsp: <=== [CBFRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=51 Len=8]
[2024-11-14 22:23:13] [34mdebug[39m: 	zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY], [target=522]
[2024-11-14 22:23:13] [32minfo[39m: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "522".
[2024-11-14 22:23:13] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-11-14 22:23:13] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=4)
[2024-11-14 22:23:13] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=4](frmRx=4) Added to rxQueue
[2024-11-14 22:23:13] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=4)
[2024-11-14 22:23:13] [34mdebug[39m: 	zh:ember:ezsp: <=== [CBFRAME: ID=128:"INCOMING_ROUTE_ERROR_HANDLER" Seq=51 Len=8]
[... cut ...]
[2024-11-14 22:23:15] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=5](frmRx=5) Added to rxQueue
[2024-11-14 22:23:15] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=4)
[2024-11-14 22:23:15] [34mdebug[39m: 	zh:ember:ezsp: <=== [CBFRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=51 Len=8]
[2024-11-14 22:23:15] [34mdebug[39m: 	zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY], [target=522]

Mintzki, 2025-01-16:
I've the same issue, but the first Twinguard got paired, the second one not
smoke detector II also paired

Koenkk, 2025-02-10:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/25885#issuecomment-2649088933

closed 2025-02-10 by Koenkk