Koenkk/zigbee2mqtt #24776: INSPELNING Smart plug (IKEA) wrong Voltage 23.3V instead of 233V

njordan77, 2024-11-14:
### What happened?

Latest Version installed, paired new INSPELNING Smart plug. seems either a wrong value 23.3V or on purpose? don't think so.

### What did you expect to happen?

expected something arround 230V not 23.3V

### How to reproduce it (minimal and precise)

happens any time.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

latest version

### Adapter

ElectroLama zigzig

### Setup

Raspi5

MaxRower, 2024-11-17:
You need to update the firmware of the plug.
Currently, this is a little bit complicated, so you could wait for the next release.
See also: https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2354118090

icheered, 2024-12-01:
This just got fixed! https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0

closed 2025-02-07 by github-actions[bot]