Koenkk/zigbee2mqtt #24777: [New device support]:

prhodey, 2024-11-15:
### Link

https://www.aliexpress.us/item/3256807565817715.html?spm=a2g0o.order_list.order_list_main.5.4aa318022ioDIo&gatewayAdapt=glo2usa

### Database entry

{"id":43,"type":"Router","ieeeAddr":"0xa4c138cc9a3baff8","nwkAddr":61556,"manufId":4417,"manufName":"_TZE204_eaulras5","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1731632401740}

### Zigbee2MQTT version

1.41.0

### Comments

Tried to model it after other similar Tuya Presence detection models but no luck

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_eaulras5',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Nothing, keep getting a "device not supported" message despite attempting to use various external converter methods

djXonePL, 2024-11-25:
can someone convert this for zigbee2mqtt?
https://community.home-assistant.io/t/how-i-got-it-working-wz-m100-tuya-presence-sensor-ts0601-tze204-7gclukjs/737497

odx, 2024-11-26:
I am trying to get the same device running and ran into the same issue.
Is my assumption correct that with the empty external converter configuration above at least the device should not show up as unsupported?
[image]

odx, 2024-11-28:
I had some progress reading the sensor values.

What I found so far to be reasonable. Allthough presence sometimes hangs in "True" even if the target passes quickly beyond the maximum_range.

[9, 'target_distance', tuya.valueConverter.divideBy100],
[112, 'presence', tuya.valueConverter.trueFalseInvert],
[4, 'maximum_range', tuya.valueConverter.divideBy100],
[3, 'minimum_range', tuya.valueConverter.divideBy100],

odx, 2024-11-29:
@prhodey can you try if this converter works for you?

```
const { } = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_eaulras5',
        },
    ],
    zigbeeModel: ['TS0601'],
    model: 'TS0601_foo',
    vendor: '_TZE204_eaulras5',
    description: 'Automatically generated definition',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.presence(),
        e.occupancy(),
        e.numeric('target_dis_closest', ea.STATE).withDescription('the target distance away from the sensor').withUnit('m'),
        e
            .numeric('mov_delay', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(43200)
            .withValueStep(1)
            .withDescription('setting the delay time to keep moving status  (the time from moving -> None)')
            .withUnit('s'),
        e
            .numeric('none_delay', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(43200)
            .withValueStep(1)
            .withDescription('setting the delay time to keep present status (the time from present -> None)')
            .withUnit('s'),
        e.illuminance_lux(),
        e
            .numeric('far_mov_sensitivity', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withDescription('the moving detecting sensitivity 1 meter away'),
        e
            .numeric('near_mov_sensitivity', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withDescription('the presence detecting sensitivity  within 1 meter'),
        e
            .numeric('near_occ_sensitivity', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withDescription('the moving detecting sensitivity 1 meter away'),
        e
            .numeric('far_occ_sensitivity', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withDescription('the presence detecting sensitivity  1 meter away'),
        e.numeric('near_detection', ea.STATE)
            .withDescription('The closest distance that can be detected')
            .withUnit('m'),
        e.numeric('far_detection', ea.STATE)
            .withDescription('The farest distance that can be detected (moving)')
            .withUnit('m'),
        e.numeric('occ_detection', ea.STATE)
            .withDescription('The farest distance that can be detected (present)')
            .withUnit('m'),
        e.binary('restore_factory', ea.STATE)
            .withDescription('restore_factory'),
        e.binary('led_indicator', ea.STATE)
            .withDescription('turn on or off the led '),
    ],
    meta: {
        tuyaDatapoints: [
            [104, 'presence', tuya.valueConverter.trueFalse],
            [112, 'occupancy', tuya.valueConverter.trueFalseInvert],
            [9, 'target_dis_closest', tuya.valueConverter.divideBy100], 
            [101, 'mov_delay', tuya.valueConverter.raw],
            [102, 'none_delay', tuya.valueConverter.raw],
            [103, 'illuminance_lux', tuya.valueConverter.raw],
            [105, 'far_mov_sensitivity', tuya.valueConverter.raw],
            [110, 'near_mov_sensitivity', tuya.valueConverter.raw],
            [109, 'near_occ_sensitivity', tuya.valueConverter.raw],
            [111, 'far_occ_sensitivity', tuya.valueConverter.raw],
            [3, 'near_detection', tuya.valueConverter.divideBy100],
            [4, 'far_detection', tuya.valueConverter.divideBy100],
            [108, 'occ_detection', tuya.valueConverter.divideBy100],
            [106, 'restore_factory', tuya.valueConverter.trueFalse],
            [107, 'led_indicator', tuya.valueConverter.trueFalse],
        ],
    },
};

module.exports = definition;

```

prhodey, 2024-11-30:
@odx thank you!
Seems to be working well:

{
    "far_detection": 3,
    "far_mov_sensitivity": 5,
    "far_occ_sensitivity": 5,
    "illuminance_lux": null,
    "led_indicator": null,
    "linkquality": 192,
    "mov_delay": 60,
    "near_detection": 0.1,
    "near_mov_sensitivity": 5,
    "near_occ_sensitivity": 5,
    "none_delay": 60,
    "occ_detection": 2.5,
    "occupancy": true,
    "presence": true,
    "restore_factory": null,
    "target_dis_closest": 1.2
}

Wikibear, 2024-11-30:
I try, but somethings aren't working:
presence stay N/A
Illuminance  stay N/A
no effect of factory reset
no effect of switch LED ON/OFF

`{
    "last_seen": "2024-11-30T15:48:52+01:00",
    "linkquality": 255,
    "occupancy": false,
    "target_dis_closest": 1,
    "device": {
        "applicationVersion": 74,
        "dateCode": "",
        "friendlyName": "WZRadar",
        "hardwareVersion": 1,
        "ieeeAddr": "0xa4c1385e2da497d7",
        "manufacturerID": 4417,
        "manufacturerName": "_TZE204_eaulras5",
        "model": "TS0601_TZE204_eaulras5",
        "networkAddress": 33658,
        "powerSource": "Mains (single phase)",
        "stackVersion": 0,
        "type": "Router",
        "zclVersion": 3
    },
    "mov_delay": 0,
    "none_delay": 0,
    "far_mov_sensitivity": 4,
    "near_mov_sensitivity": 4,
    "near_occ_sensitivity": 4,
    "far_occ_sensitivity": 4,
    "near_detection": 0.1,
    "far_detection": 2.4,
    "occ_detection": 2.4
}`

odx, 2024-11-30:
@Wikibear thanks for testing. I noticed that LED, factory reset and ilumnation does not work quite correctly.
But presence and distance detection is fine for me. Your distance seems incorrect with exactly 1m and also presence is missing in your status completely?

i implemented this based on manufacturer specifications. maybe we can collect a few log entries to investigate?

Wikibear, 2024-11-30:
I test in debug mode and i get only this:
`[2024-11-30 16:52:28] debug: 	zhc:tuya: Datapoint 112 not defined for '_TZE204_eaulras5' with value 1
[2024-11-30 16:52:30] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 134
[2024-11-30 16:52:30] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 134
[2024-11-30 16:52:31] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 100
[2024-11-30 16:52:31] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 100
[2024-11-30 16:52:31] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 132
[2024-11-30 16:52:31] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 132
[2024-11-30 16:52:32] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 213
[2024-11-30 16:52:32] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 213
[2024-11-30 16:52:32] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 83
[2024-11-30 16:52:32] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 83
[2024-11-30 16:52:33] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 124
[2024-11-30 16:52:33] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 124
[2024-11-30 16:52:34] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 98
[2024-11-30 16:52:34] debug: 	zhc:tuya: Datapoint 9 not defined for '_TZE204_eaulras5' with value 98
[2024-11-30 16:52:40] debug: 	zhc:tuya: Datapoint 112 not defined for '_TZE204_eaulras5' with value 0
[2024-11-30 16:52:40] debug: 	zhc:tuya: Datapoint 112 not defined for '_TZE204_eaulras5' with value 0
`
No other datapoints.

HDHristov, 2024-12-01:
Hi.
I did some testing too.
What is referred to as "occupancy", is actually "motion". The motion timeout has no effect. Motion state changes immediately (which as far as I'm concerned is a good thing).
Presence is ok, as well as its timeout setting.
Lux value is null.
Led option doesn't work. Here is the error message:

`z2m: Publish 'set' 'led_indicator' to 'Living room presence sensor' failed: 'TypeError: Cannot read properties of undefined (reading 'to')'`

Hope this is useful.
Regards!

odx, 2024-12-01:
I think PIR (infrared motion) sensors commonly use "occupancy" for movement detection. That's why I mapped it this way.
The occupancy state means true -> movement, false -> present. The movement delay does not affect this state though but the present state. So if you turn down the presence detection range to 0 so only movement is detected, the delay affects after which time of no movement the present state flips back to 0.

I am getting illuminance values every now and then now. But this is no reliable source of information. Like every few hours or so.

Also, I played around with the factory and LED settings. And got rid of the "Cannot read properties" error.

I think it works this way:

- Flip factory reset to 0 will cause the device to load th factory settings.
- LED indicator: when on the LED will blink when no movement is detected anymore, otherwise LED is always ON.

This is my recent version:

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_eaulras5',
        },
    ],
    zigbeeModel: ['TS0601'],
    model: 'PJ3201A',
    vendor: 'Dongguan Pinjia Technology Co.,LTD.',
    description: 'Human Presence Sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.presence(),
        e.occupancy(),
        e.numeric('target_dis_closest', ea.STATE).withDescription('the target distance away from the sensor').withUnit('m'),
        e.numeric('illuminance_lux', ea.STATE).withDescription('illuminance').withUnit('lx'),
        e
            .numeric('mov_delay', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(43200)
            .withValueStep(1)
            .withDescription('setting the delay time to keep moving status  (the time from moving -> None)')
            .withUnit('s'),
        e
            .numeric('none_delay', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(43200)
            .withValueStep(1)
            .withDescription('setting the delay time to keep present status (the time from present -> None)')
            .withUnit('s'),
        e.illuminance_lux(),
        e
            .numeric('far_mov_sensitivity', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withDescription('the moving detecting sensitivity 1 meter away'),
        e
            .numeric('near_mov_sensitivity', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withDescription('the moving detecting sensitivity  within 1 meter'),
        e
            .numeric('near_occ_sensitivity', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withDescription('the presence detecting sensitivity  within 1 meter'),
        e
            .numeric('far_occ_sensitivity', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withDescription('the presence detecting sensitivity  1 meter away'),
        e.numeric('near_detection', ea.STATE_SET)
            .withValueMin(0.1)
            .withValueMax(7)
            .withValueStep(0.1)
            .withDescription('The closest distance that can be detected')
            .withUnit('m'),
        e.numeric('far_detection', ea.STATE_SET)
            .withValueMin(0.1)
            .withValueMax(7)
            .withValueStep(0.1)
            .withDescription('The farest distance that can be detected (moving)')
            .withUnit('m'),
        e.numeric('occ_detection', ea.STATE_SET)
            .withValueMin(0.1)
            .withValueMax(7)
            .withValueStep(0.1)
            .withDescription('The farest distance that can be detected (present)')
            .withUnit('m'),
        e.binary('restore_factory', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('restore_factory'),
        e.binary('led_indicator', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('turn on or off the led '),
    ],
    meta: {
        tuyaDatapoints: [
            [104, 'presence', tuya.valueConverter.trueFalse1],
            [112, 'occupancy', tuya.valueConverter.trueFalseInvert],
            [9, 'target_dis_closest', tuya.valueConverter.divideBy100], 
            [101, 'mov_delay', tuya.valueConverter.raw],
            [102, 'none_delay', tuya.valueConverter.raw],
            [103, 'illuminance_lux', tuya.valueConverter.raw],
            [105, 'far_mov_sensitivity', tuya.valueConverter.raw],
            [110, 'near_mov_sensitivity', tuya.valueConverter.raw],
            [109, 'near_occ_sensitivity', tuya.valueConverter.raw],
            [111, 'far_occ_sensitivity', tuya.valueConverter.raw],
            [3, 'near_detection', tuya.valueConverter.divideBy100],
            [4, 'far_detection', tuya.valueConverter.divideBy100],
            [108, 'occ_detection', tuya.valueConverter.divideBy100],
            [106, 'restore_factory', tuya.valueConverter.onOff],
            [107, 'led_indicator', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

HDHristov, 2024-12-01:
I have tested it in tuya. Luminance works, but the values are about 10x than expected.
Led off does really turn it off which is nice since it is bright and annoying.
Strange that it behaves differently in z2m.

odx, 2024-12-01:
You are right. Indeed it works. 

But on is off and off is on. Like with factory reset.  I will check if this can be fixed.

HDHristov, 2024-12-01:
Yep, just found out it works! Just inverted.
Only thing left is the luminance.
Thank you so much!

HDHristov, 2024-12-01:
`[107, 'led_indicator', tuya.valueConverterBasic.lookup({ON: false, OFF: true})],`

^ This seems to work, although I'm sure there is a more elegant solution. First time dipping my feet into converters.
Or you could call it "Disable led", then ON would true = disabled.

HDHristov, 2024-12-02:
So, no luck with the illuminance?

HDHristov, 2024-12-02:
I have not seen a single readout yet on two devices running. I played with the calibration value also, but it didn't make a difference.
I have a third device hoked to tuya and it reads out illuminance every 2~3 seconds.
Is it possible that they are polling it in some way? That would not be very nice.

odx, 2024-12-02:
I have no idea how to make the device send the data actually. The only way I could identify is doing a factory reset of the settings. :D

I added a devideBy10 to the illuminance (hoping this is meaningful, I have no device to compare values).

HDHristov, 2024-12-02:
Yeah, divide by 10 makes sense according to my other devices, but that needs more testing.
Now that you mentioned factory reset, it does indeed trigger a reading.
That's the first time I have seen a number for luminance.
I'm seriously thinking they might be polling it somehow instead of it reporting on its own.

HDHristov, 2024-12-02:
I monitored the behavior of the device in tuya. It doesn't send illuminance values regularly, rather on (significant) change.
Unfortunately that's not what happens in z2m. It is stuck at the value which was measured after factory reset, at least for me.
I guess that's what 10 eur gets you 🙂

Wikibear, 2024-12-04:
Illuminance works after factory reset. But the measured value is to high. Dunno if this is right or not.

closed 2025-02-10 by github-actions[bot]

JaxonTT, 2025-04-29:
I'm new in coding. I copied the code, added 'const fzLocal' & 'fromZigbee: [tuya.fz.datapoints, fzLocal.gateway_connection_status],' in external converter, illuminance worked. 

Not sure what's the code for. 

```
const fzLocal = {
    gateway_connection_status: {
        cluster: 'manuSpecificTuya',
        type: ['commandMcuGatewayConnectionStatus'],
        convert: async (model, msg, publish, options, meta) => {
            const payload = {payloadSize: 1,payload: 1}
            await msg.endpoint.command('manuSpecificTuya', 'mcuGatewayConnectionStatus', payload, {});
        },
    },
}

const definition = {
        fingerprint: [{
				modelID: 'TS0601',
				manufacturerName: '_TZE204_eaulras5',
		}],
        model: "PJ3201A",
        vendor: "Dongguan Pinjia Technology Co.,LTD.",
        description: "Custom Human Presence Sensor",
        fromZigbee: [tuya.fz.datapoints, fzLocal.gateway_connection_status],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
```