Koenkk/zigbee2mqtt #24780: Is external_temperature_input of Aqara SRTS-A01 device still working?

brunialti, 2024-11-15:
### What happened?

Using the code and the setting as the folloeing screenshot I cannot get working my TRV.
I was using them since last year correctly, by tnow the external temperature value does not update anymore.

[image]
[image]
[image]

### What did you expect to happen?

I expect the external_temperature_input be correctly updated 

### How to reproduce it (minimal and precise)

please see the config screenshots above

### Zigbee2MQTT version

Zigbee2MQTT version     1.41.0 

### Adapter firmware version

20240315

### Adapter

Sonoff Zigbee 3.0 Usb Dongle Plus

### Setup

plain

ste-fle, 2024-11-18:
I'm having the same problem - when I click into the textfield for "external temperature input" I'm getting the following error:

[image]

The value should be set via HomeAssistant MQTT publish "zigbee2mqtt/Büro Thermostat/set" and does not seem to work at all, but at least setting it in the HA device itself seems to set the correct value despite it not being shown properly in Z2M.

brunialti, 2024-11-18:
Same error occourred to me. I'm using the device in Nodered. 
I tried also to set the value directly via mqtt explorer, with the same result. 
The only effect is that the property local_temperature reported the external temperature correctly. 
The property external_temperature_input did'nt.

augard, 2025-01-20:
Same error can't set external temperature.

macroparts, 2025-01-29:
I would appreciate if this would get fixed

closed 2025-04-08 by github-actions[bot]