Koenkk/zigbee2mqtt #24781: [Feature request]: Support Device Based Discovery

corporategoth, 2024-11-15:
### Is your feature request related to a problem? Please describe

Home Assistant has FINALLY released Device Based Discovery in 2024.11

This should be able to cut down the massive number of messages being published for device discovery (in my case from 14,000 down to 180).

### Describe the solution you'd like

Support device-based discovery https://www.home-assistant.io/integrations/mqtt/#device-discovery-payload

### Describe alternatives you've considered

This issue has a long history: https://github.com/Koenkk/zigbee2mqtt/issues/20648

### Additional context

N/A

corporategoth, 2024-11-15:
@Koenkk It's finally here and an option :)

corporategoth, 2024-12-01:
@Koenkk is this planned for 2.0.0 (it seems like a major rev feature)?

Koenkk, 2024-12-01:
@corporategoth not planned for 2.0.0 (since it is not breaking)

corporategoth, 2024-12-01:
@Koenkk does everything in 2.0.0 have to be breaking? ;)

Koenkk, 2024-12-02:
@corporategoth for sure not but didnt have the time to work on this yet. Would be great if someone else could take a look at this.

corporategoth, 2025-02-01:
Bump, not stale

corporategoth, 2025-02-01:
@Koenkk could you remove the stale label? otherwise I'd have to open this anew.

craigwitter, 2025-03-03:
I'd vote for this.   I have around 145 devices in my environment and see a large load time with regards to the MQTT entries being processed. I'm told in an issue that I am following in the Home Assistant github that device based is supposed to improve this as it's able to load one device vs having to process each sensor for each device separately.

craigwitter, 2025-03-03:
> > large load time with regards to the MQTT entries being processed
> 
> Where exactly do you see this?

Here you go: https://github.com/home-assistant/core/issues/136495#issuecomment-2692499891

Thanks again for all your hard work.

Craig

ytimenkov, 2025-07-21:
Is there a reason to have this issue closed as not planned? I get that it may not be a priority, but "Would be great if someone else could take a look at this", so and having it open with "help wanted" label would increase chances.

I personally don't have many devices (still "yet" :) ) so the scalability is not a concern, but it feels very reverse that zigbee2mqtt first needs to explode devices into components for home assistant later recombine back with enormous amount of duplication. I expect code will be just cleaner.

The tricky part is supporting migration (so that entities' ids are preserved to keep automations / settings) rather than just making a breaking change.

And of course, compatibility: assistant has this feature for ~10 months - could be both very new or fairly stable feature, depending on auto-update settings :)