Koenkk/zigbee2mqtt #24782: Error while starting zigbee-herdsman: Not a tty, cannot set

Horstage, 2024-11-15:
### What happened?

Zigbee2MQTT fails to start due to an error "Not a tty, cannot set".

### How to reproduce it (minimal and precise)

docker-compose.yml
```yaml
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest-dev
    volumes:
      - ./zigbee2mqtt:/app/data
      - /run/udev:/run/udev:ro
    devices:
      # - /dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_012BCBFC-if01-port0:/dev/zigbee
      - /dev/ttyUSB1:/dev/zigbee
```

The serial device is available and writable
```
docker compose run --entrypoint=/bin/sh zigbee2mqtt

/app # test -w /dev/zigbee && echo success || echo failure
success                                                                                                                                              
```

Z-Station firmware is up-to-date afaik

```shell
sudo ./zme_zigbee info -d /dev/ttyUSB1
Openning port                            ..............................                            OK

				BOARD INFORMATION

	FIRMWARE:
		ZME CORE VERSION: 	0304
		BUILD SEQUENCE: 	00001641
		BUILD DATETIME: 	2024-04-15T19:28:38
		VERSION: 		00.00.23.00
		BOOTLOADER VERSION: 	02.04.00.01
		BOOTLOADER CRC32: 	2329e271
		SE VERSION:       	00.01.02.16
		IMAGE MIN:       	00044000
		IMAGE MAX:       	000f0000
		PAGE SIZE:       	00002000
		CRC32 KEY UPDATE:	c51431b2
		USER DATA:		VALID
		USER DATA PIN:	VALID

```

configuration.yaml
```yaml
homeassistant: false
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mosquitto
serial:
  port: /dev/zigbee
advanced:
  log_level: debug
```

### Zigbee2MQTT version

1.41.0

[... 54 lines cut ...]
2024-11-15T14:59:17.311702624Z [2024-11-15 15:59:17] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
2024-11-15T14:59:17.565033688Z [2024-11-15 15:59:17] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
2024-11-15T14:59:17.565786876Z [2024-11-15 15:59:17] debug: 	zh:zstack:unpi:writer: --> buffer [239]
2024-11-15T14:59:18.568004725Z [2024-11-15 15:59:18] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
2024-11-15T14:59:18.568483910Z [2024-11-15 15:59:18] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
2024-11-15T14:59:18.820383409Z [2024-11-15 15:59:18] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
2024-11-15T14:59:18.823634071Z [2024-11-15 15:59:18] error: 	z2m: Error while starting zigbee-herdsman
2024-11-15T14:59:18.824431480Z [2024-11-15 15:59:18] error: 	z2m: Failed to start zigbee
2024-11-15T14:59:18.824667561Z [2024-11-15 15:59:18] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
2024-11-15T14:59:18.824952236Z [2024-11-15 15:59:18] error: 	z2m: Exiting...
2024-11-15T14:59:18.825443751Z [2024-11-15 15:59:18] error: 	z2m: Error: Error: Not a tty, cannot set
2024-11-15T14:59:18.843821946Z 
2024-11-15T14:59:18.843850749Z /app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
2024-11-15T14:59:18.843856922Z   var er = new ERR_STREAM_WRITE_AFTER_END();
2024-11-15T14:59:18.843860708Z            ^
2024-11-15T14:59:18.843864100Z Error: write after end
2024-11-15T14:59:18.843867507Z     at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
2024-11-15T14:59:18.843870924Z     at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
2024-11-15T14:59:18.843874352Z     at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
2024-11-15T14:59:18.843877708Z     at Logger.log (/app/lib/util/logger.ts:198:25)
2024-11-15T14:59:18.843881004Z     at Logger.info (/app/lib/util/logger.ts:211:14)
2024-11-15T14:59:18.843884390Z     at Znp.onPortClose (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:96:16)
2024-11-15T14:59:18.843887794Z     at Object.onceWrapper (node:events:632:26)
2024-11-15T14:59:18.843891055Z     at SerialPort.emit (node:events:529:35)
2024-11-15T14:59:18.843894287Z     at port.close.then.closing (/app/node_modules/@serialport/stream/dist/index.js:242:18)

Ricc68, 2024-11-16:
You need to connect to a real serial port like /dev/ttyUSB1, you should import as-is into your docker.
I bet /dev/zigbee is a pseudo terminal (pty) which doesn't work because lacks many functions of a real serial port and it doesn't count that can echo characters back and forth, z2m requires a real serial not a pseudo tty.

Ricc68, 2024-11-16:
When we see an error like "Not a tty" means there is something that is opening a device expecting a tty device and finding something not a tty.

Horstage, 2024-11-18:
Setting the adapter to `ember` worked for me.

The new configuration.yml looks like this:
```yaml
homeassistant: false
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mosquitto
serial:
  port: /dev/zigbee
  disable_led: false
  adapter: ember
  baudrate: 115200
  rtscts: false
advanced:
  log_level: info
```

I created a PR [#3226](https://github.com/Koenkk/zigbee2mqtt.io/pull/3226) which would resolve the issue for me.

closed 2024-11-18 by Koenkk