Koenkk/zigbee2mqtt #24785: zigbee2mqtt log not showing and error

liamlain, 2024-11-15:
### What happened?

[image]

### How to reproduce it (minimal and precise)

click the log tab but no log showing and error message : Failed to get 9336c2b0_zigbee2mqtt logs, e.callApiRaw is not a function

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

Add on on Home Assistant pi4 8gb ram

Ricc68, 2024-11-15:
I had the issue of log not loading and I resolved it by updating Home Assistant.
I don't know if it was a true incompatibility or updating HA reset something that was stuck.

pilot1981, 2024-11-16:
> I had the issue of log not loading and I resolved it by updating Home Assistant. I don't know if it was a true incompatibility or updating HA reset something that was stuck.

so you solved updating HA?

Ricc68, 2024-11-16:
Yes, updating HA solved the issue for me.

Suffice, 2024-11-16:
Updating HA and doing a full restart fixed it for me as well.

Nerivec, 2024-11-16:
This is not limited to Z2M addon and is a problem on HA side. Should be fixed after updating supervisor to [2024.11.3](https://github.com/home-assistant/supervisor/releases/tag/2024.11.3) or later.

closed 2024-11-16 by Nerivec