Koenkk/zigbee2mqtt #24788: OXT dimmer module does not expose switch_type and generally is unstable

mikolajpiecuch, 2024-11-15:
### What happened?

Firstly, I came back to z2m after almost 2 years. Two years ago I added to the z2m my OXT dimmer, then it was added as a TS110E. I added even then an issue -> [16520](https://github.com/Koenkk/zigbee2mqtt/issues/16520). Back in the time, as I remember, it worked great. Unfortunately, now (2 years it was unconnected) I can't see for sure "switch_type" option and generally dimmer works very unstable - wall switch even has a problem with turning off/on this module, same problem from z2m UI.

My dimmer module is made by OXT. I see some devices here [OXT](https://www.zigbee2mqtt.io/supported-devices/#v=OXT) (despite fact z2m adds this device as TS110E), there is no OXT dimmer. There is wrong picture/name in the "[TS0001_switch_module](https://www.zigbee2mqtt.io/devices/TS0001_switch_module.html#tuya-ts0001-switch-module)" - in the title (and description) it's called switch but in the picture we can dimmer module (very similar to mine)

### What did you expect to happen?

My device is recognized and works properly

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0

### Setup

Add-On on Home Assistant OS on proxmox

wpak18, 2024-11-16:
I got the same problem, no switch_type option

`const {deviceEndpoints, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS110E'],
    model: 'TS110E',
    vendor: '_TZ3210_3mpwqzuu',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), light(), light()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;`

mikolajpiecuch, 2024-11-16:
I found in this commit switch type was remove -> https://github.com/Koenkk/zigbee-herdsman-converters/commit/401d1c4be1690c89c9d05dd69ce998af284484bc#diff-0157bb612c63e92c50f93116e1f50bf8e263030ed26d86caa28479159c2a9f4aR8052 
I prepared https://github.com/Koenkk/zigbee-herdsman-converters/pull/8326 but I'm not sure if not affects other devices

wpak18, 2024-11-19:
Stil dont have option switch_type for TS110E_2gang_2 OXT dimmer

FPSUsername, 2024-11-25:
This is quite problematic, I just installed this dimmer only to find out a core feature has been removed.
My vendor shows as `_TZ3210_3mpwqzuu`

mikolajpiecuch, 2024-11-25:
@wpak18 / @FPSUsername sorry, I completely forgot to add switch type also for 2 channel dimmer. Ready to merge but I don't have this device to check if it works correctly. I did this based on others devices. Are you able to check this on your side? https://github.com/Koenkk/zigbee-herdsman-converters/pull/8379

FPSUsername, 2024-11-25:
I just got the dimmer, so I have to figure out how it all works. I believe that I did see one false report where it showed that one of the two lights were reported as off, while it was on.

I run home assistant via the docker image (so I can make use of HACS and such) and use a Sonoff zigbee dongle.

Currently trying to figure out how to get this updated repository, I could use a helping hand.

mikolajpiecuch, 2024-11-25:
@FPSUsername to test only my changes you can add this two lines https://github.com/Koenkk/zigbee-herdsman-converters/pull/8379/files in your **tuya.js** file. I use HAOS but it should be similar in your case. Here is simple instruction (for HAOS addon but you can see general rule to test changes) https://github.com/zigbee2mqtt/hassio-zigbee2mqtt?tab=readme-ov-file#testing-changes-locally

FPSUsername, 2024-11-25:
I'm trying to edit `node_modules/zigbee-herdsman-converters/devices/tuya.js` but the whole TS110E entry doesn't seem to exist.
I am running the latest version:
Zigbee2MQTT
Current version: 1.41.0-1

FPSUsername, 2024-11-25:
Thank you, amazing work on the quick resolution!

It now works!

FPSUsername, 2024-11-26:
@mikolajpiecuch I too seem to have incorrect reporting of the on/off state.

The lights are turned off, updated in HA, but after a while one of them shows as turned on (even though the lights and the switches are off)

iroger, 2024-11-27:
Not sure if it is related but next to state being reported wrongly after a while I also have the issue with scenes. Turning off is no issue but turning on the dimmer appears to send only the new dim value and the lamp doesn't turn on. (It is on in HA dashboard) after switching off-on the new dimmed value is visible. If the device is already on switching scene immediately shows the new dimmed value.

iroger, 2024-11-28:
It looks like the wrong state reporting is somehow linked to the power on behavior, a fix was posted on another forum which seems to change the behavior:
1) turn off all the lights in home assistant
2) switch of the mains, wait 10s, turn on mains again
3) after limited testing it seems stable now

closed 2025-02-04 by github-actions[bot]

FPSUsername, 2025-03-17:
> It looks like the wrong state reporting is somehow linked to the power on behavior, a fix was posted on another forum which seems to change the behavior:
> 
> 1. turn off all the lights in home assistant
> 2. switch of the mains, wait 10s, turn on mains again
> 3. after limited testing it seems stable now

So I think I found out how to trigger the ghosting "light is off" for L1 after a while. If you put the power on behavior to off, then the light report will remain on until you actually turn off the lights. For some reason, L1 is reported incorrectly in the off state when the power on behavior is set to previous. Oddly enough it doesn't happen for L2, but there's another but because it's not configurable: https://github.com/Koenkk/zigbee2mqtt/issues/26759