Koenkk/zigbee2mqtt #24789: Yale 420W missing some data

diogofm7, 2024-11-16:
### What happened?

I installed a Yale 420w, and some data is not being captured.

Action source name: NULL
Action user: NULL
Battery: NULL 
Actions: NULL 

Firmware build date
220701
Firmware version
02.07.01

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20221226

### Adapter

zStack3x0 Sonoff-E

### Setup

Addons Home Assintant

LeandroIssa, 2024-12-05:
Same problem here!

diogofm7, 2025-01-05:
> Same problem here!

Any solution, or still having problem?

diogofm7, 2025-01-07:
> > Any solution, or still having problem?
> 
> Still having problem!

Será que é só nós dois?

closed 2025-03-16 by github-actions[bot]