Koenkk/zigbee2mqtt #24790: [Wrong device]: SNZB-05 recognized as a IH-K665 when it should be a TS0207_water_leak_detector_1

MilesTEG1, 2024-11-16:
### Link

https://fr.aliexpress.com/item/1005006512619030.html

### Model

SNZB-05

### Description

Water Leak Detector

### Vendor

Tuya

### Picture (link)

https://i.imgur.com/0hTgLZm.png

### Database entry

{"id":81,"type":"EndDevice","ieeeAddr":"0xa4c1382c55337701","nwkAddr":59608,"manufId":4742,"manufName":"_TZ3000_upgcbody","powerSource":"Battery","modelId":"SNZB-05","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SNZB-05","manufacturerName":"_TZ3000_upgcbody","powerSource":3,"zclVersion":3,"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x847127fffedfe3cd","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":112,"batteryVoltage":26}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x847127fffedfe3cd","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x847127fffedfe3cd","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":543112114},"lastSeen":1731749633703,"checkinInterval":3600}

### Notes

Hello,
My Tuya Water Leak Detector isn't recognized as the right on. For Z2M it's a IH-K665.
But here the model I have.
[image]
It should be a TS0207_water_leak_detector_1.

Can you fix the detection please?

Thanks in advance.

Koenkk, 2024-11-18:
Fixed in https://github.com/Koenkk/zigbee-herdsman-converters/commit/16efcd232b9167285ab55e60e5011c3831ffafd4

closed 2024-11-18 by Koenkk

MilesTEG1, 2024-11-19:
Thanks you very much! 😇