Koenkk/zigbee2mqtt #24792: Group with one device, two endpoints not working with latest `dev` branch

mundschenk-at, 2024-11-16:
### What happened?

The Hue Twilight has to light endpoints (`11` and `12`). Since adding support for it it in zhc, I've had a group for it to switch both lights on and off simultaneously. With `dev` from immediately after the release of `1.41.0`, this worked fine. However, two days ago I updated to the latest `dev` (after the merge of my `event` PRs). Now that group does not work anymore. Groups containing several different devices seem to be fine.

With groups only containing endpoints from a single device, the result when trying to toggle them on looks like this:
```
[2024-11-16 14:01:27] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/my_group/set' with data '{"state":"ON","color":{"x":0.409,"y":0.362},"transition":0.4,"brightness":253}'
[2024-11-16 14:01:27] error: 	z2m: No converter available for 'brightness' (253)
[2024-11-16 14:01:27] error: 	z2m: No converter available for 'state' ("ON")
[2024-11-16 14:01:27] debug: 	z2m: Publishing 'set' 'color' to 'zigbee2mqtt/my_group'
[2024-11-16 14:01:27] debug: 	zh:controller:group: Command 14 lightingColorCtrl.moveToColor({"transtime":4,"colorx":26804,"colory":23724})
```

### What did you expect to happen?

The group should work normally.

### Zigbee2MQTT version

1.41.0-dev

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

Plain

mundschenk-at, 2024-11-16:
@Nerivec @Koenkk As soon as I add a different (single-endpoint) device to the group, it starts to work fine. Any idea where I should look?

mundschenk-at, 2024-11-16:
A bit more experimenting: Groups that only contain multi-endpoint devices don't work either, even if there's two of them (i.e. `devA/ep11`, `devA/ep12`, `devB/ep11`, `devB/ep12`). Trying to remove one of them results in an error:

```
[2024-11-16 14:10:04] error: 	z2m: Failed to remove from group (ZCL command 0x001788010dff71ff/1 genGroups.remove({"groupid":8}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 35656 - 1 - 108 - 4 - 3 after 10000ms))
```

mundschenk-at, 2024-11-16:
Trying to remove one of them fails, possibly because the command seems to target endpoint one instead of the actual group member endpoint, which might be another multi-endpoint frontend issue:

```
[2024-11-16 14:10:04] error: 	z2m: Failed to remove from group (ZCL command 0x00XXXXXXXXX/1 genGroups.remove({"groupid":8}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 35656 - 1 - 108 - 4 - 3 after 10000ms))
```

Nerivec, 2024-11-17:
I've tested it with 2.0.0 branch (including frontend), with a TS0002, I can't reproduce. Removing works as well.

```
zh:controller:endpoint: ZCL command 0xa4c1387d3b6e41cc/2 genGroups.remove({"groupid":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
```

https://github.com/Koenkk/zigbee2mqtt/pull/24432 changes the logic quite a bit though, so, I guess it could have removed the problem?

Koenkk, 2024-11-26:
@mundschenk-at could you provide the debug log when controlling works with 1.40.2 and not with 1.41.0?

Nerivec, 2024-11-26:
https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/groups.ts
https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/group.ts

mundschenk-at, 2024-11-30:
@Koenkk The difference is between `1.41.0` and `dev` (soon to be `1.42.0`). Here are the debug logs for the same command:

# 1.41.0
```
[2024-11-30 22:23:34] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/my_twilight_group/set' with data '{"state":"ON","color":{"x":0.453,"y":0.374},"transition":0.4,"brightness":203}'
[2024-11-30 22:23:34] debug: 	z2m: Publishing 'set' 'brightness' to 'my_twilight_group'
[2024-11-30 22:23:34] debug: 	zh:controller:group: Command 13 genLevelCtrl.moveToLevelWithOnOff({"level":203,"transtime":4})
[2024-11-30 22:23:34] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x000000000000000d","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":8,"transid":26,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[17,22,4,203,4,0]}}
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:writer: --> frame [254,26,36,2,1,13,0,0,0,0,0,0,0,255,0,0,1,8,0,26,0,30,6,0,17,22,4,203,4,0,8]
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,2,0,103]
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,2,0,103]
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 2 - [0] - 103
[2024-11-30 22:23:34] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequestExt - {"status":0}
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,26,220]
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,26,220]
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,26] - 220
[2024-11-30 22:23:34] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":26}
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-11-30 22:23:34] debug: 	z2m: Publishing 'set' 'color' to 'my_twilight_group'
[2024-11-30 22:23:34] debug: 	zh:controller:group: Command 13 lightingColorCtrl.moveToColor({"transtime":4,"colorx":29687,"colory":24510})
[2024-11-30 22:23:34] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x000000000000000d","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":768,"transid":27,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[17,23,7,247,115,190,95,4,0]}}
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:writer: --> frame [254,29,36,2,1,13,0,0,0,0,0,0,0,255,0,0,1,0,3,27,0,30,9,0,17,23,7,247,115,190,95,4,0,166]
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,2,0,103]
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,2,0,103]
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 2 - [0] - 103
[2024-11-30 22:23:34] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequestExt - {"status":0}
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,27,221]
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,27,221]
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,27] - 221
[2024-11-30 22:23:34] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":27}
[2024-11-30 22:23:34] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-11-30 22:23:34] debug: 	z2m: Publishing 'set' 'transition' to 'my_twilight_group'
[2024-11-30 22:23:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight', payload '{"brightness_back":203,"brightness_front":203,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_back":{"hue":22,"saturation":73,"x":0.453,"y":0.374},"color_front":{"hue":22,"saturation":73,"x":0.453,"y":0.374},"color_mode":"color_temp","color_mode_back":"xy","color_mode_front":"xy","color_temp":366,"color_temp_back":404,"color_temp_front":404,"color_temp_startup":65535,"color_temp_startup_back":65535,"color_temp_startup_front":65535,"device":{"applicationVersion":2,"dateCode":"20240902","friendlyName":"my_twilight","hardwareVersion":0,"ieeeAddr":"0x00123456789abcde","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003711201","networkAddress":35656,"powerSource":"Mains (single phase)","softwareBuildID":"1.122.2","stackVersion":1,"type":"Router","zclVersion":8},"gradient_back":null,"linkquality":109,"power_on_behavior_back":"on","power_on_behavior_front":"on","state_back":"OFF","state_front":"ON","update":{"installed_version":16780546,"latest_version":16780546,"state":"idle"},"update_available":null}'
[2024-11-30 22:23:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight_group', payload '{"brightness":203,"color":{"h":22,"hue":22,"s":73,"saturation":73,"x":0.453,"y":0.374},"color_mode":"xy","color_temp":404,"state":"ON","state_front":null}'
[2024-11-30 22:23:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight/front', payload '{"brightness":203,"color":{"hue":22,"saturation":73,"x":0.453,"y":0.374},"color_mode":"xy","color_temp":404,"color_temp_startup":65535,"power_on_behavior":"on","state":"ON"}'
[2024-11-30 22:23:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight', payload '{"brightness_back":203,"brightness_front":203,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_back":{"hue":22,"saturation":73,"x":0.453,"y":0.374},"color_front":{"hue":22,"saturation":73,"x":0.453,"y":0.374},"color [... cut ...]
[2024-11-30 22:18:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight/front', payload '{"brightness":254,"color":{"hue":22,"saturation":73,"x":0.453,"y":0.374},"color_mode":"xy","color_temp":404,"color_temp_startup":65535,"power_on_behavior":"on","state":"OFF"}'
[2024-11-30 22:18:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight/back', payload '{"brightness":254,"color":{"hue":22,"saturation":73,"x":0.453,"y":0.374},"color_mode":"xy","color_temp":404,"color_temp_startup":65535,"gradient":null,"power_on_behavior":"on","state":"OFF"}'
```

mundschenk-at, 2024-12-01:
@Koenkk, @Nerivec I've looked over the changes between `1.41.0` and `dev` for Z2M itself, ZH and ZHC and I don't see anything that should impact group behaviour. There are some `z-stack` changes, but to my untrained I, they appear non-related (logging). Also, there is a [change to the device converter](https://github.com/Koenkk/zigbee-herdsman-converters/commit/8e3ed7c41eaa4d008d7c459ee98c8353b96423bc), but that also appears to be a noop (especially since I own an LGT001).

I can set up more detailed debugging if you tell me where to look - I am completely stumped at this point (it does not help that the frontend is buggy with multi-endpoint devices, excluding/including worked around those issues for me).

Koenkk, 2024-12-02:
@mundschenk-at could you check with the current `dev`? (which is z2m 2.0.0)

mundschenk-at, 2024-12-03:
@Koenkk: Unfortunately, it looks pretty much the same:

```
[2024-12-03 09:11:49] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/my_twilight_group/set' with data '{"state":"ON","color":{"x":0.379,"y":0.352},"transition":0.4,"brightness":254}'
[2024-12-03 09:11:49] error: 	z2m: No converter available for 'brightness' (254)
[2024-12-03 09:11:49] error: 	z2m: No converter available for 'state' ("ON")
[2024-12-03 09:11:49] debug: 	z2m: Publishing 'set' 'color' to 'my_twilight_group'
[2024-12-03 09:11:49] debug: 	zh:controller:group: Command 14 lightingColorCtrl.moveToColor({"transtime":4,"colorx":24838,"colory":23068})
[2024-12-03 09:11:49] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x000000000000000e","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":768,"transid":88,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[17,76,7,6,97,28,90,4,0]}}
[2024-12-03 09:11:49] debug: 	zh:zstack:unpi:writer: --> frame [254,29,36,2,1,14,0,0,0,0,0,0,0,255,0,0,1,0,3,88,0,30,9,0,17,76,7,6,97,28,90,4,0,249]
[2024-12-03 09:11:49] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,2,0,103]
[2024-12-03 09:11:49] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,2,0,103]
[2024-12-03 09:11:49] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 2 - [0] - 103
[2024-12-03 09:11:49] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequestExt - {"status":0}
[2024-12-03 09:11:49] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-12-03 09:11:50] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,88,158]
[2024-12-03 09:11:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,88,158]
[2024-12-03 09:11:50] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,88] - 158
[2024-12-03 09:11:50] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":88}
[2024-12-03 09:11:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-12-03 09:11:50] debug: 	z2m: Publishing 'set' 'transition' to 'my_twilight_group'
[2024-12-03 09:11:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight', payload '{"action":null,"brightness_back":254,"brightness_front":254,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_back":{"hue":24,"saturation":43,"x":0.379,"y":0.352},"color_front":{"hue":24,"saturation":43,"x":0.379,"y":0.352},"color_mode":"color_temp","color_mode_back":"xy","color_mode_front":"xy","color_temp":366,"color_temp_back":259,"color_temp_front":259,"color_temp_startup":65535,"color_temp_startup_back":65535,"color_temp_startup_front":65535,"device":{"applicationVersion":2,"dateCode":"20231211","friendlyName":"my_twilight","hardwareVersion":0,"ieeeAddr":"0x00123456789abcde","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003711201","networkAddress":65143,"powerSource":"Mains (single phase)","softwareBuildID":"1.113.2","stackVersion":1,"type":"Router","zclVersion":8},"gradient_back":null,"linkquality":72,"power_on_behavior_back":"on","power_on_behavior_front":"on","state_back":"OFF","state_front":"OFF","update":{"installed_version":16779522,"latest_version":16779522,"state":"idle"}}'
[2024-12-03 09:11:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight_group', payload '{"brightness":254,"color":{"h":24,"hue":24,"s":43,"saturation":43,"x":0.379,"y":0.352},"color_mode":"xy","color_temp":259,"state":"OFF","state_front":null}'
[2024-12-03 09:11:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight/front', payload '{"brightness":254,"color":{"hue":24,"saturation":43,"x":0.379,"y":0.352},"color_mode":"xy","color_temp":259,"color_temp_startup":65535,"power_on_behavior":"on","state":"OFF"}'
[2024-12-03 09:11:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight', payload '{"action":null,"brightness_back":254,"brightness_front":254,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_back":{"hue":24,"saturation":43,"x":0.379,"y":0.352},"color_front":{"hue":24,"saturation":43,"x":0.379,"y":0.352},"color_mode":"color_temp","color_mode_back":"xy","color_mode_front":"xy","color_temp":366,"color_temp_back":259,"color_temp_front":259,"color_temp_startup":65535,"color_temp_startup_back":65535,"color_temp_startup_front":65535,"device":{"applicationVersion":2,"dateCode":"20231211","friendlyName":"my_twilight","hardwareVersion":0,"ieeeAddr":"0x00123456789abcde","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003711201","networkAddress":65143,"powerSource":"Mains (single phase)","softwareBuildID":"1.113.2","stackVersion":1,"type":"Router","zclVersion":8},"gradient_back":null,"linkquality":72,"power_on_behavior_back":"on","power_on_behavior_front":"on","state_back":"OFF","state_front":"OFF","update":{"installed_version":16779522,"latest_version":16779522,"state":"idle"}}'
[2024-12-03 09:11:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight/back', payload '{"brightness":254,"color":{"hue":24,"saturation":43,"x":0.379,"y":0.352},"color_mode":"xy","color_temp":259,"color_temp_startup":65535,"gradient":null,"power_on_behavior":"on","state":"OFF"}'
[2024-12-03 09:11:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight/front', payload '{"brightness":254,"color":{"hue":24,"saturation":43,"x":0.379,"y":0.352},"color_mode":"xy","color_temp":259,"color_temp_startup":65535,"power_on_behavior":"on","state":"OFF"}'
[... cut ...] ration":43,"x":0.379,"y":0.352},"color_mode":"xy","color_temp":259,"state":"OFF"}'
[2024-12-03 09:11:50] error: 	z2m: EventBus error 'HomeAssistant/publishEntityState': Cannot read properties of null (reading 'toString')
[2024-12-03 09:11:50] debug: 	z2m: TypeError: Cannot read properties of null (reading 'toString')
    at HomeAssistant.onPublishEntityState (/opt/zigbee2mqtt/lib/extension/homeassistant.ts:1287:50)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:204:17)
```

Koenkk, 2024-12-03:
For groups the list of toZigbee converter is determined here: https://github.com/Koenkk/zigbee2mqtt/blob/acba533397dbfaf54f26a9bc5c5b199e46f2a0f5/lib/extension/publish.ts#L307 (in this case definitions is an array of all the definitions of devices in the group). Could you add some `console.log()` here to debug this?

mundschenk-at, 2024-12-05:
With additional `console.log()` calls (definition and converter array):

```
[2024-12-05 16:00:03] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/my_twilight_group/set' with data '{"state":"ON","color":{"x":0.453,"y":0.374},"transition":0.4,"brightness":203}'
getDefinitionConverters [
  {
    toZigbee: [
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object]
    ],
    fromZigbee: [
      [Object], [Object],
      [Object], [Object],
      [Object], [Object],
      [Object], [Object],
      [Object], [Object],
      [Object], [Object],
      [Object], [Object],
      [Object], [Object]
    ],
    exposes: [
      [Light], [Enum],
      [Enum],  [Light],
      [Enum],  [Enum],
      [List],  [Enum],
      [Enum],  [Numeric]
    ],
    meta: {
      supportsHueAndSaturation: true,
      turnsOffAtBrightness1: true,
      multiEndpoint: true
    },
    configure: [AsyncFunction: configure],
    endpoint: [Function: endpoint],
    onEvent: undefined,
    ota: true,
    zigbeeModel: [ 'LGT001', 'LGT002', 'LGT003' ],
    model: '929003711201',
    vendor: 'Philips',
    description: 'Hue Twilight sleep and wake-up light white',
    whiteLabel: [ [Object], [Object] ],
    options: [ [Numeric], [Binary], [Binary] ]
  },
  {
    toZigbee: [
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object], [Object], [Object], [Object],
      [Object]
    ],
[... 175 lines cut ...]
[2024-12-05 16:00:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight/back', payload '{"brightness":254,"color":{"hue":25,"saturation":39,"x":0.372,"y":0.35},"color_mode":"xy","color_temp":247,"color_temp_startup":65535,"gradient":null,"power_on_behavior":"on","state":"OFF"}'
[2024-12-05 16:00:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight/front', payload '{"brightness":254,"color":{"hue":22,"saturation":73,"x":0.453,"y":0.374},"color_mode":"xy","color_temp":404,"color_temp_startup":65535,"power_on_behavior":"on","state":"OFF"}'
[2024-12-05 16:00:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/other_group_containing_twilight', payload '{"brightness":254,"color":{"h":22,"hue":22,"s":73,"saturation":73,"x":0.453,"y":0.374},"color_mode":"xy","color_temp":404,"state":"OFF"}'
[2024-12-05 16:00:03] error: 	z2m: EventBus error 'HomeAssistant/publishEntityState': Cannot read properties of null (reading 'toString')
[2024-12-05 16:00:03] debug: 	z2m: TypeError: Cannot read properties of null (reading 'toString')
    at HomeAssistant.onPublishEntityState (/opt/zigbee2mqtt/lib/extension/homeassistant.ts:1287:50)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:204:17)
[2024-12-05 16:00:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/my_twilight/back', payload '{"brightness":254,"color":{"hue":22,"saturation":73,"x":0.453,"y":0.374},"color_mode":"xy","color_temp":404,"color_temp_startup":65535,"gradient":null,"power_on_behavior":"on","state":"OFF"}'
```

@Koenkk What looks weird to me is 
```
{
    key: [
      'state',
      'brightness',
      'brightness_percent',
      'on_time',
      'off_wait_time'
    ],
    ...
    endpoints: [ 'front' ]
  },
``` 
as `front` is only one of the endpoints of the device (the other one being `back`, both are part of the group but not distinguishable in the `console.log(definition)` output. The two light endpoints have similar, but not identical, capabilities (`back` is a gradient light, `front` is not.

mundschenk-at, 2024-12-08:
I have linked so e additional reports.

ciberkids, 2024-12-10:
hi i have the same problem
i have 2 groups
 1 called kitchen
 1 called Toilet Lights
 
 in both groups there are 2 devices of the same nature and type: https://www.zigbee2mqtt.io/devices/UP-SA-9127D.html#sunricher-up-sa-9127d

the conf is like following:
```
'5':
    friendly_name: Toilet Lights
    devices:
      - 0xe0798dfffe797c4d/1 # sunricher
      - 0xe0798dfffe797c4d/2 # sunricher
  '7':
    friendly_name: Kitchen
    devices:
      - 0xe0798dfffe79bfde/1 # sunricher
      - 0x0017880108593d6b/11
      - 0x0017880108593d81/11
      - 0xe0798dfffe79bfde/2 # sunricher
```
interesting enough if i send the command

```
{
  "state": "ON"
}

```

to the topic `zigbee2mqtt/Kitchen/set` it works

instead if i send it to the topic `zigbee2mqtt/Toilet Lights/set`

i receive the error:
```
2024-12-10 16:06:43z2m: No converter available for 'state' ("ON")
```

ciberkids, 2024-12-10:
Hi, I have the same problem maybe i can add some more info to the matter :)

this is my version [1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0) commit: [861cba6](https://github.com/Koenkk/zigbee2mqtt/commit/861cba6)

and it is running on a container

basically i have many groups with different devices in them. but the problem is manifested only on one

this is the group definition in the conf:
```
 '2':
    friendly_name: Living Room
    devices:
      - 0x001788010b4086e7/11
      - 0x0017880109213d75/11
      - 0x001788010856df34/11
      - 0x00158d00025843eb/1
  '3':
    friendly_name: Test Lights
    devices: []
  '4':
    friendly_name: Kitchen Table
    devices:
      - 0x0017880108593d6b/11
      - 0x0017880108593d81/11
  '5':
    friendly_name: Toilet Lights
    devices:
      - 0xe0798dfffe797c4d/1  #Sunricher UP-SA-9127D
      - 0xe0798dfffe797c4d/2  #Sunricher UP-SA-9127D
  '6':
    friendly_name: All Lights
    devices: []
  '7':
    friendly_name: Kitchen
    devices:
      - 0xe0798dfffe79bfde/1 #Sunricher UP-SA-9127D
      - 0x0017880108593d6b/11
      - 0x0017880108593d81/11
      - 0xe0798dfffe79bfde/2  #Sunricher UP-SA-9127D
```
The problem manifest with the group: `Toilet Lights`
The device in it is [Sunricher UP-SA-9127D](https://www.zigbee2mqtt.io/devices/UP-SA-9127D.html#sunricher-up-sa-9127d)

if i send the command:

```
{
  "state": "ON"
}
```
to the topic:
`zigbee2mqtt/Toilet Lights/set`

i got this error
`2024-12-10 16:25:17z2m: No converter available for 'state' ("ON")`

interesting if i send the same command to the group `kitchen` it works
for the record: in the kitchen group there is also a philips hue light [Philips 4090331P9](https://www.zigbee2mqtt.io/devices/4090331P9.html#philips-4090331p9)

Koenkk, 2024-12-10:
@mundschenk-at could you try if the following change fixes your issue? https://github.com/Koenkk/zigbee2mqtt/pull/25156/files

mundschenk-at, 2024-12-10:
@Koenkk Unfortunately, no change in behavior.

Koenkk, 2024-12-11:
Added extra debug logging to https://github.com/Koenkk/zigbee2mqtt/pull/25156/files, can you provide the debug log with this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

mundschenk-at, 2024-12-13:
@Koenkk Sorry for the delay, here's the log:

```
[2024-12-13 19:18:00] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/my_twilight_group/set' with data '{"state":"ON","color":{"x":0.453,"y":0.374},"transition":0.4,"brightness":203}'
[2024-12-13 19:18:00] debug: 	z2m: == ALL CONVERTERS: state,brightness,brightness_percent,on_time,off_wait_time,transition,level_config,rate,brightness_move,brightness_move_onoff,brightness_step,brightness_step_onoff,color,color_temp,color_temp_percent,color_mode,color_options,colortemp_move,color_temp_move,color_temp_step,color_temp_startup,hue_move,saturation_move,hue_step,saturation_step,power_on_behavior,hue_power_on_behavior,hue_power_on_brightness,hue_power_on_color_temperature,hue_power_on_color,effect,state,brightness,brightness_percent,on_time,off_wait_time,gradient_scene,gradient,scene_store,scene_recall,scene_add,scene_remove,scene_remove_all,scene_rename,read,write,command,reset,zclcommand
[2024-12-13 19:18:00] debug: 	z2m: == FIND CONVERTER: brightness
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='state,brightness,brightness_percent,on_time,off_wait_time', endpoints='front', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='transition', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='level_config', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='rate', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='brightness_move,brightness_move_onoff', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='brightness_step,brightness_step_onoff', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='color,color_temp,color_temp_percent', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='color_mode', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='color_options', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='colortemp_move,color_temp_move', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='color_temp_step', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='color_temp_startup', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='hue_move,saturation_move', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='hue_step,saturation_step', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='power_on_behavior', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='hue_power_on_behavior', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='hue_power_on_brightness,hue_power_on_color_temperature,hue_power_on_color', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='effect', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='state,brightness,brightness_percent,on_time,off_wait_time', endpoints='back', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='gradient_scene', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='gradient', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='scene_store', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='scene_recall', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='scene_add', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='scene_remove', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='scene_remove_all', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='scene_rename', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='read', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='write', endpoints='undefined', isGroup=false, endpointName=undefined
[2024-12-13 19:18:00] debug: 	z2m: == MATCHING CONVERTER: key='command', endpoints='undefined', isGroup=false, endpointName=undefined
[... cut ...] 3,"y":0.374},"color_mode":"xy","color_temp":404,"state":"OFF","state_front":null}'
[2024-12-13 19:18:00] error: 	z2m: EventBus error 'HomeAssistant/publishEntityState': Cannot read properties of null (reading 'toString')
[2024-12-13 19:18:00] debug: 	z2m: TypeError: Cannot read properties of null (reading 'toString')
    at HomeAssistant.onPublishEntityState (/opt/zigbee2mqtt/lib/extension/homeassistant.ts:1284:50)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:204:17)
```

Koenkk, 2024-12-14:
Found & fixed the issue!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2024-12-14 by Koenkk