Koenkk/zigbee2mqtt #24793: [New device support]: iGET HOME TS10 Thermostatic Radiator Valve - thermostatic head, Zigbee 3.0, LED LCD

Jozefik, 2024-11-16:
### Link

https://gear-up.me/usd/iget-home-ts10-thermostatic-radiator-valve-thermostatic-head-zigbee-3-0-led-lcd.html

### Database entry

{"id":35,"type":"EndDevice","ieeeAddr":"0x3c2ef5fffe9ecad0","nwkAddr":28075,"manufId":4098,"manufName":"_TZE200_ch3t1xbl","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ch3t1xbl","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1731762467694}

### Zigbee2MQTT version

1.41.0

### Comments

Tried to model it after other similar Tuya Presence detection models but no luck

I have tried to model it using other similar thermostatic valves but it does not work.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_ch3t1xbl',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Nothing is working. It says "device not supported" message.

Jozefik, 2024-11-17:
Okay i have managed to make it work using this External Definition:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_ch3t1xbl',
        },
    ],
    model: 'iGET Home TS10',
    vendor: 'iGET Home',
    description: 'Thermostat radiator valve',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetLocalTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery_low(),
        e.child_lock(),
        e.open_window(),
        e.open_window_temperature().withValueMin(5).withValueMax(30),
        e.comfort_temperature().withValueMin(5).withValueMax(30),
        e.eco_temperature().withValueMin(5).withValueMax(30),
        e
            .climate()
            .withPreset(['auto', 'manual', 'holiday'])
            .withLocalTemperatureCalibration(-5, 5, 0.1, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET)
            .withSystemMode(['off', 'heat'], ea.STATE_SET, 'Only for Homeassistant'),
        e
            .binary('heating_stop', ea.STATE_SET, 'ON', 'OFF')
            .withDescription(
                'Battery life can be prolonged' +
                    ' by switching the heating off. To achieve this, the valve is closed fully. To activate the ' +
                    'heating stop, the device display "HS", press the pair button to cancel.',
            ),
        tuya.exposes.frostProtection(
            'When Anti-Freezing function is activated, the temperature in the house is kept ' +
                'at 8 °C, the device display "AF".press the pair button to cancel.',
        ),
        e
            .numeric('boost_timeset_countdown', ea.STATE_SET)
            .withUnit('s')
            .withDescription(
                'Setting ' +
                    'minimum 0 - maximum 465 seconds boost time. The boost (â¨) function is activated. The remaining ' +
                    'time for the function will be counted down in seconds ( 465 to 0 ).',
            )
            .withValueMin(0)
            .withValueMax(465),
        e.holiday_temperature().withValueMin(5).withValueMax(30),
        e
            .text('holiday_start_stop', ea.STATE_SET)
            .withDescription(
[... 55 lines cut ...]
            [104, 'comfort_temperature', tuya.valueConverter.divideBy10],
            [105, 'eco_temperature', tuya.valueConverter.divideBy10],
            [106, 'schedule', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [107, null, tuya.valueConverter.TV02SystemMode],
            [107, 'system_mode', tuya.valueConverter.TV02SystemMode],
            [107, 'heating_stop', tuya.valueConverter.TV02SystemMode],
            [115, 'online', tuya.valueConverter.onOffNotStrict],
            [108, 'schedule_monday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [112, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [109, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [113, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [110, 'schedule_friday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [114, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [111, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDaySingleDP],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

Which is the same as model: **'TV02-Zigbee',** in tuya.ts definiton and everything seems to be working. So can we add support for it ?

alesoharek, 2024-11-17:
Dobrý den, dnes jsem nainstaloval tuto hlavici na radiátory ale bohužel se mi jí nedaří přidat rozumně do HomeAssistant . Máte někdo toto rozchozené v HA? Děkuji za radu jak to zprovoznit.

closed 2025-01-24 by github-actions[bot]

cucpo, 2025-04-16:
> TV02-Zigbee

> Okay i have managed to make it work using this External Definition:
>
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

> Okay i have managed to make it work using this External Definition:
>
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

Hi, how i use this external configuration ?