Koenkk/zigbee2mqtt #24795: [Wrong device]: The TRV601 identifies itself as TRV602 or TRV602Z

ZVlog, 2024-11-16:
### Link

https://pl.aliexpress.com/item/1005007553996616.html?spm=a2g0o.order_list.order_list_main.28.19911c24Ba3tdg&gatewayAdapt=glo2pol

### Model

TRV601

### Description

TRV601 thermostats are reported as TRV602Z. Zigbee manufacturer: _TZE204_ltwbm23f and  _TZE200_rtrmfadk

### Vendor

Tuya

### Picture (link)

https://a.allegroimg.com/original/116a6b/7515749a4c3dbcc2261e8e3b51ed/Glowica-Grzejnikowa-Inteligentny-Termostat-4szt-TRV-601-Smart-Tuya-Zigbee-Stan-opakowania-oryginalne

### Database entry

{"id":85,"type":"EndDevice","ieeeAddr":"0xa4c13810c5b43a35","nwkAddr":49655,"manufId":4417,"manufName":"_TZE204_ltwbm23f","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"y��.i|��.i���.i*��.i.��.i\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE204_ltwbm23f","zclVersion":3,"appVersion":74,"modelId":"TS0601","powerSource":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1731772868630} {"id":86,"type":"EndDevice","ieeeAddr":"0xa4c138df75486d9b","nwkAddr":24927,"manufId":4417,"manufName":"_TZE204_ltwbm23f","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"\u0006\u0000\u0000\u0000e\u0007\u0000\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_ltwbm23f","powerSource":3,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1731772451587}

waclaw66, 2024-12-01:
Same problem

lacko-hub, 2024-12-03:
My is identified as TS0601 and manufacturer by _TZE204_ltwbm23f, however even if all functions work correctly in zigbee2mqtt edge HA addon, device wont work in Homeassistant. Climate entities are visible, but they dont react on any actions.

waclaw66, 2024-12-08:
Mine valve is recognized as...
[image]

It works fine, however Z2M doesn't present some features during discovery: `modes` and `preset_modes`. Therefore HA shows it always as OFF and offers all climate modes...

|actual ([Tuya TRV601](https://www.zigbee2mqtt.io/devices/TRV601.html#tuya-trv601))|expected ([TS0601_thermostat](https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#tuya-ts0601-thermostat))|
|-|-|
|[image]|[image]|
|[image]|[image]|

lacko-hub, 2024-12-08:
Correct. Its still possible to control this valve via mqqt with topic zigbee2mqtt/climate.name/set and payloads 'on' 'standby' etc., which are not available in HA features. That is the reason why we can not operate these valves directly.
In theory its useable but needs to have more programming inputs.

[image]

keisezrg9, 2024-12-09:
I have the same issue, also it seems that it won't react to setpoint schedule set via Z2M
[image]
[image]

Smandurlo, 2024-12-13:
I have 3 of them, there are some bugs, but nothing special, they work as I want.

Wrong identification as TRV602Z, when I reboot HA the valves are marked as OFF buy they are actually working, there are a lof of wrong settings (fan, cooling, and so on as per pics someone posted before) and there is the possibility to setup the power of the motor, but it is not allowed in this model, so it reverts to medium.

shajek, 2024-12-16:
Hello,
i have same problems.  Modes are broken in HA and day schedule does not work

waclaw66, 2024-12-20:
I've tried to act `_TZE204_ltwbm23f` as TRV601 within tuya.js file. TRV601 profile lacks e.g. `eco_temperature`, `comfort_temperature`, `mode` behaves differently. I think `_TZE204_ltwbm23f` is closer to TRV602Z than TRV601.

kubiajir, 2025-01-07:
Hi, 
I have the same issue with mine TRV601. I tried to play with it a bit and so far I have observed following:

- Position - OK
-  Mode 
   valve => z2m
   auto => auto (OK)
   antifrost => antifrost (OK)
   off => standby (fail) 
   night => eco (OK)
   day + manual => comfort (OK)
- Local temperature (OK)
- Temperature setpoint (OK)
 - Comfort/eco temperature (OK)
 - Schedule (FAIL - it messed up the schedule)
 - Display brightness (OK)
 - I didn't test the remining options.

Other notes:
- I tried to set it as different valve but it dint work. 
- HA integration shows bad properties - not usable at all
- I will ordered Tuya router to see the zigbee options.

closed 2025-03-16 by github-actions[bot]