Koenkk/zigbee2mqtt #24798: [New device support]: Adding Namron device

SanderThorsen, 2024-11-17:
### Link

https://www.namron.com/products/namron-zigbee-edge-termostat-hvit/

### Database entry

{"id":81,"type":"Router","ieeeAddr":"0x385cfbfffecbce02","nwkAddr":52543,"manufId":4714,"manufName":"Namron AS","powerSource":"Unknown","modelId":"4512783","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,6,513,516,1029,1794,2820,4096],"outClusterList":[3,25,1030],"clusters":{"hvacUserInterfaceCfg":{"attributes":{"tempDisplayMode":0}},"hvacThermostat":{"attributes":{"32774":0,"32778":1,"pICoolingDemand":0,"pIHeatingDemand":0,"systemMode":4,"programingOperMode":1,"occupiedHeatingSetpoint":2000,"localTemp":2040,"runningState":0}},"genOnOff":{"attributes":{"onOff":1}},"msRelativeHumidity":{"attributes":{"measuredValue":2900}},"genBasic":{"attributes":{"modelId":"4512783","manufacturerName":"Namron AS","powerSource":0,"zclVersion":8,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20240117","swBuildId":"1.11"}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":1,"acPowerMultiplier":1,"acCurrentDivisor":1,"acCurrentMultiplier":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xbc026efffef2cea8","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0xbc026efffef2cea8","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1},{"cluster":1029,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20240117","swBuildId":"1.11","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1731786271063}

### Zigbee2MQTT version

1.41.0

### Comments

I tried to add a pull request with this device, but got this

fatal: unable to access 'https://github.com/Koenkk/zigbee2mqtt.io.git/': The requested URL returned error: 403

Can someone please help me?

### External definition

```shell
const {onOff, humidity, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['4512783'],
    model: '4512783',
    vendor: 'Namron AS',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), humidity(), electricityMeter()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Not sure what to write here, but the device is not supported. It's a termostat from Namron. Used it on a zigbee network. The Values I got to use from it was humidity, power on, measured temperature and set temperature

Teeseeone, 2024-11-19:
Please add this to Z2M

andebor, 2024-11-21:
@SanderThorsen Based on your error message, it seems you've tried to commit directly to this repo. You need to fork the repo and commit the change there. After doing that, you can create a pull request against this repo. Check out this link for more info https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/fork-a-repo

Teeseeone, 2025-01-08:
Shouldn't this device be added soon as supported?

seime, 2025-02-17:
Duplicate of #24047

closed 2025-04-27 by github-actions[bot]