Koenkk/zigbee2mqtt #24800: Need help with percentage control in my Tuya day/night curtain converter！！

lccox, 2024-11-17:
### What happened?

**I wrote an external converter for the Tuya day/night curtains. The curtain device has the following DP points:**
{
  "1": "Day Curtain Control",
  "2": "Day Curtain Percentage Control",
  "3": "Day Curtain Percentage Status",
  "101": "Night Curtain Percentage Control",
  "102": "Night Curtain Percentage Status",
  "109": "Night Curtain Control",
  "120": "Night Curtain Motor Direction",
  "121": "Day Curtain Motor Direction"
}
Using these DP points, I edited an external converter. It allows me to control the day and night curtains' open/close using DP 1 and 109, while DP 2 and 3 enable percentage control and feedback for the day curtain. However, I cannot achieve percentage control for the night curtain using DP 101, although I seem to be able to read the percentage using DP 102.

Here is the converter code I wrote:

### const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_luexzx4j',
    description: 'Zhongling-DayNightCover-GM24TF/S-TYZ with dual cover support',
    fromZigbee: [tuya.fz.datapoints, legacy.fromZigbee.tuya_cover, fz.ignore_basic_report],
    toZigbee: [tuya.tz.datapoints, legacy.toZigbee.tuya_cover_control, legacy.toZigbee.tuya_cover_options],
    exposes: [
        e.cover_position().withEndpoint('cover_a').setAccess('position', ea.STATE_SET),
        e.cover_position().withEndpoint('cover_b').setAccess('position', ea.STATE_SET),
    ],
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            // Cover A
            [1, 'state_cover_a', tuya.valueConverterBasic.lookup({'CONTINUE': tuya.enum(3), 'CLOSE': tuya.enum(1), 'STOP': tuya.enum(2), 'OPEN': tuya.enum(0)})],
            [2, 'position_cover_a', tuya.valueConverter.coverPositionInverted],
            [3, 'position_cover_a_raw', tuya.valueConverter.raw],
            // Cover B
            [109, 'state_cover_b', tuya.valueConverterBasic.lookup({'CONTINUE': tuya.enum(3), 'CLOSE': tuya.enum(1), 'STOP': tuya.enum(2), 'OPEN': tuya.enum(0)})],
            [101, 'position_cover_b', tuya.valueConverter.coverPositionInverted], // Fix for percentage control logic
            [102, 'position_cover_b_raw', tuya.valueConverter.raw],
        ],
    },
    endpoint: (device) => {
        return {
            cover_a: 1,
            cover_b: 1, // DP is used for distinguishing features, no need for separate endpoints
        };
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, logger);
    },
};

module.exports = definition;
**I would appreciate any insights or solutions to enable percentage control for the night curtain using DP 101.**

### What did you expect to happen?

**I need assistance in completing the percentage control for the night curtain to improve the external converter, allowing the device to function correctly. Thank you!**

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211210

### Adapter

ember

### Setup

Addon on HAOS

### Debug log

[2024-11-17 15:34:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385ce13eda64', payload '{"linkquality":58,"position":0,"position_b":74,"position_cover_a":37,"position_cover_a_raw":0,"position_cover_b":44,"position_cover_b_raw":47,"position_d":61,"position_n":100,"running":false,"state":"OPEN","state_cover_a":"CLOSE","state_cover_b":"STOP"}'
[2024-11-17 15:34:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385ce13eda64/cover_a', payload '{"position":0,"state":"CLOSE"}'
[2024-11-17 15:34:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385ce13eda64/cover_b', payload '{"position":44,"state":"STOP"}'
[2024-11-17 15:34:53] debug: 	zh:zstack:unpi:parser: <-- [254,33,68,129,0,0,0,239,168,224,1,1,0,58,0,213,169,186,0,0,13,9,39,2,0,220,101,2,0,4,0,0,0,56,168,224,29,76]
[2024-11-17 15:34:53] debug: 	zh:zstack:unpi:parser: --- parseNext [254,33,68,129,0,0,0,239,168,224,1,1,0,58,0,213,169,186,0,0,13,9,39,2,0,220,101,2,0,4,0,0,0,56,168,224,29,76]
[2024-11-17 15:34:53] debug: 	zh:zstack:unpi:parser: --> parsed 33 - 2 - 4 - 129 - [0,0,0,239,168,224,1,1,0,58,0,213,169,186,0,0,13,9,39,2,0,220,101,2,0,4,0,0,0,56,168,224,29] - 76
[2024-11-17 15:34:53] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":57512,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":58,"securityuse":0,"timestamp":12233173,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,39,2,0,220,101,2,0,4,0,0,0,56]}}
[2024-11-17 15:34:53] debug: 	zh:controller: Received payload: clusterID=61184, address=57512, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=58, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":39,"commandIdentifier":2},"payload":{"seq":56320,"dpValues":[{"dp":101,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,56]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}
[2024-11-17 15:34:53] debug: 	zh:controller:endpoint: ZCL command 0xa4c1385ce13eda64/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":39,"writeUndiv":false})
[2024-11-17 15:34:53] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c1385ce13eda64:57512/1 (0,0,1)
[2024-11-17 15:34:53] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":57512,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":154,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,39,11,2,0]}}
[2024-11-17 15:34:53] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,168,224,1,1,0,239,154,0,30,5,16,39,11,2,0,50]
[2024-11-17 15:34:53] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-11-17 15:34:53] debug: 	z2m: Received Zigbee message from '0xa4c1385ce13eda64', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,56],"type":"Buffer"},"datatype":2,"dp":101}],"seq":56320}' from endpoint 1 with groupID 0
[2024-11-17 15:34:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385ce13eda64', payload '{"linkquality":58,"position":0,"position_b":74,"position_cover_a":37,"position_cover_a_raw":0,"position_cover_b":44,"position_cover_b_raw":47,"position_d":61,"position_n":100,"running":false,"state":"OPEN","state_cover_a":"CLOSE","state_cover_b":"STOP"}'
[2024-11-17 15:34:53] debug: 	zhc:legacy:fz:tuya_cover: Unrecognized DP #101: {"dp":101,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,56]}}
[2024-11-17 15:34:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385ce13eda64/cover_a', payload '{"position":0,"state":"CLOSE"}'
[2024-11-17 15:34:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385ce13eda64', payload '{"linkquality":58,"position":0,"position_b":74,"position_cover_a":37,"position_cover_a_raw":0,"position_cover_b":44,"position_cover_b_raw":47,"position_d":61,"position_n":100,"running":false,"state":"OPEN","state_cover_a":"CLOSE","state_cover_b":"STOP"}'
[2024-11-17 15:34:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385ce13eda64/cover_b', payload '{"position":44,"state":"STOP"}'
[2024-11-17 15:34:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385ce13eda64/cover_a', payload '{"position":0,"state":"CLOSE"}'
[2024-11-17 15:34:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385ce13eda64/cover_b', payload '{"position":44,"state":"STOP"}'

n71154plus, 2024-11-29:
I have the same issue.
I forced it to return DP values and got the following:

debug 2024-11-29 15:32:25zhc:legacy:fz:tuya_cover: commandDataResponse DP #3: {"dp":3,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
debug 2024-11-29 15:32:25zhc:legacy:fz:tuya_cover: commandDataResponse DP #102: {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
debug 2024-11-29 15:32:25zhc:legacy:fz:tuya_cover: commandDataResponse DP #104: {"dp":104,"datatype":1,"data":{"type":"Buffer","data":[0]}}
debug 2024-11-29 15:32:25zhc:legacy:fz:tuya_cover: commandDataResponse DP #105: {"dp":105,"datatype":1,"data":{"type":"Buffer","data":[0]}}
debug 2024-11-29 15:32:25zhc:legacy:fz:tuya_cover: commandDataResponse DP #120: {"dp":120,"datatype":1,"data":{"type":"Buffer","data":[0]}}
debug 2024-11-29 15:32:25zhc:legacy:fz:tuya_cover: commandDataResponse DP #121: {"dp":121,"datatype":1,"data":{"type":"Buffer","data":[0]}}
debug 2024-11-29 15:32:25zhc:legacy:fz:tuya_cover: commandDataResponse DP #122: {"dp":122,"datatype":1,"data":{"type":"Buffer","data":[0]}}
debug 2024-11-29 15:32:25zhc:legacy:fz:tuya_cover: commandDataResponse DP #123: {"dp":123,"datatype":1,"data":{"type":"Buffer","data":[0]}}

When sending DP 101, data is returned, but the curtain does not respond.
Additionally, DP 120 and DP 121 cannot be set and show no reaction.

Based on the returned data, DP 104, DP 105, DP 122, and DP 123 all have values equal to 0.

Do you have a TUYA Zigbee gateway? Is there any way to identify what these DP points represent?

lccox, 2024-11-29:
> I have the same issue. I forced it to return DP values and got the following:
>
> debug 2024-11-29 15:32:25zhc:legacy:fz:tuya_cover: commandDataResponse DP #3: [...]

Hey Bro,What model of **[day and night curtains]** are you using? 
Is it GM24TF/S-TYZ-0.8??

This is the dp point I collected through the tuya gateway. 
 " 104 ":" day curtain travel limit switch "
 "105" : "the night curtain travel limit switch" 
 "109" : "the night curtain control", "120" : "the night curtain motor", "121", "curtain motor direction", "122" : "the night curtain inching switch", "123" : "day curtain inching switch"}

**By the way，We can also communicate through Wechat hahah**

n71154plus, 2024-12-03:
I'm not sure about the model of my motor; it's already installed on the window, and disassembling it is a bit tricky. However, I compared the remote control instructions with the GM24TF manual, and they match

By the way , during the Zigbee pairing process for this set of curtains, I didn't set up the TUYA MagicPacket. I'm not sure if that makes a difference, but re-pairing seems to require disassembling the curtains to press the pairing button. I'll have to wait until I have time to try that. As for the TUYA DP Points, the feedback I got matches yours.

n71154plus, 2024-12-03:
I am temporarily using a timer-based approach to control the position of the night curtain. You can refer to my code. Make sure to set your const cover_travel_time beforehand.
```
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const { logger } = require('zigbee-herdsman-converters/lib/logger');

const e = exposes.presets;
const ea = exposes.access;

const sleep = (ms) => new Promise(resolve => setTimeout(resolve, ms));
const cover_travel_time = 47000 //你的窗簾由全開至全關所需花費的時間 ms表示

// 定义常量
const DP = {
    POSITION_SET: 101,
    POSITION: 102,
    TILT_SET: 2,
    TILT: 3,
    STATE: 109,
    TILT_CONTROL: 1,
    REVERSE_NIGHT: 120,
    REVERSE_DAY: 121,
    INCHING_NIGHT: 122,
    INCHING_DAY: 123,
};

// 处理DP值的映射
const mapDpToResult = (dp, value, result) => {
    switch (dp) {
        case DP.POSITION_SET:
            result.position_set = 100 - (value & 0xff);
            break;
        case DP.POSITION:
            result.position = 100 - (value & 0xff);
            break;
        case DP.TILT_SET:
            result.tilt_set = 100 - (value & 0xff);
            break;
        case DP.TILT:
            result.tilt = 100 - (value & 0xff);
            break;
        case DP.STATE:
            result.raw_state = ['OPEN', 'CLOSE', 'STOP', 'continue'][value] || 'unknown';
            break;
        case DP.TILT_CONTROL:
            result.tilt_control_raw = ['OPEN', 'CLOSE', 'STOP', 'continue'][value] || 'unknown';
            break;
        case DP.REVERSE_NIGHT:
            result.reverse_night = ['forward', 'back'][value] || 'unknown';
            break;
        case DP.REVERSE_DAY:
            result.reverse_day = ['forward', 'back'][value] || 'unknown';
            break;
        case DP.INCHING_NIGHT:
            result.inching_night = ['disable', 'enable'][value] || 'unknown';
            break;
        case DP.INCHING_DAY:
            result.inching_day = ['disable', 'enable'][value] || 'unknown';
            break;
    }
};

// FromZigbee 转换器
const fzLocal = {
    tuya_cover_response: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport','commandDataResponse'],
[... 92 lines cut ...]
};

// 定义设备
const definition = {
    fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_luexzx4j' }],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Tuya Cover',
    fromZigbee: [fzLocal.tuya_cover_response, fzLocal.tuya_cover_report],
    toZigbee: [tz.command, tzLocal.refresh, tzLocal.tuya_cover_control],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.cover_position_tilt().setAccess('position', ea.STATE_SET).setAccess('tilt', ea.STATE_SET),
        e.enum('refresh', ea.STATE_SET, ['refresh', '']),
        e.enum('tilt_control', ea.STATE_SET, ['OPEN', 'STOP', 'CLOSE']),
        e.enum('reverse_night', ea.STATE_SET, ['forward', 'back', 'unknown']),
        e.enum('reverse_day', ea.STATE_SET, ['forward', 'back', 'unknown']),
        e.enum('inching_night', ea.STATE_SET, ['disable', 'enable', 'unknown']),
        e.enum('inching_day', ea.STATE_SET, ['disable', 'enable', 'unknown']),
    ],
};

module.exports = definition;

```

lccox, 2024-12-16:
> I am temporarily using a timer-based approach to control the position of the night curtain. You can refer to my code. Make sure to set your const cover_travel_time beforehand.我暂时使用基于定时器的方法来控制夜帘的位置。您可以参考我的代码。确保您事先设置好 const cover_travel_time。
> 
> ```
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const legacy = require('zigbee-herdsman-converters/lib/legacy');
> const { logger } = require('zigbee-herdsman-converters/lib/logger');
> 
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const sleep = (ms) => new Promise(resolve => setTimeout(resolve, ms));
> const cover_travel_time = 47000 //你的窗簾由全開至全關所需花費的時間 ms表示
> 
> 
> // 定义常量
> const DP = {
>     POSITION_SET: 101,
>     POSITION: 102,
>     TILT_SET: 2,
>     TILT: 3,
>     STATE: 109,
>     TILT_CONTROL: 1,
>     REVERSE_NIGHT: 120,
>     REVERSE_DAY: 121,
>     INCHING_NIGHT: 122,
>     INCHING_DAY: 123,
> };
> 
> // 处理DP值的映射
> const mapDpToResult = (dp, value, result) => {
>     switch (dp) {
>         case DP.POSITION_SET:
>             result.position_set = 100 - (value & 0xff);
>             break;
>         case DP.POSITION:
>             result.position = 100 - (value & 0xff);
>             break;
>         case DP.TILT_SET:
>             result.tilt_set = 100 - (value & 0xff);
>             break;
>         case DP.TILT:
>             result.tilt = 100 - (value & 0xff);
>             break;
>         case DP.STATE:
>             result.raw_state = ['OPEN', 'CLOSE', 'STOP', 'continue'][value] || 'unknown';
>             break;
>         case DP.TILT_CONTROL:
>             result.tilt_control_raw = ['OPEN', 'CLOSE', 'STOP', 'continue'][value] || 'unknown';
>             break;
>         case DP.REVERSE_NIGHT:
>             result.reverse_night = ['forward', 'back'][value] || 'unknown';
>             break;
>         case DP.REVERSE_DAY:
>             result.reverse_day = ['forward', 'back'][value] || 'unknown';
>             break;
>         case DP.INCHING_NIGHT:
>             result.inching_night = ['disable', 'enable'][value] || 'unknown';
>             break;
>         case DP.INCHING_DAY:
>             result.inching_day = ['disable', 'enable'][value] || 'unknown';
>             break;
>     }
> };
> 
> // FromZigbee 转换器
> const fzLocal = {
>     tuya_cover_response: {
[... 99 lines cut ...]
>     model: 'TS0601',
>     vendor: 'TuYa',
>     description: 'Tuya Cover',
>     fromZigbee: [fzLocal.tuya_cover_response, fzLocal.tuya_cover_report],
>     toZigbee: [tz.command, tzLocal.refresh, tzLocal.tuya_cover_control],
>     configure: tuya.configureMagicPacket,
>     exposes: [
>         e.cover_position_tilt().setAccess('position', ea.STATE_SET).setAccess('tilt', ea.STATE_SET),
>         e.enum('refresh', ea.STATE_SET, ['refresh', '']),
>         e.enum('tilt_control', ea.STATE_SET, ['OPEN', 'STOP', 'CLOSE']),
>         e.enum('reverse_night', ea.STATE_SET, ['forward', 'back', 'unknown']),
>         e.enum('reverse_day', ea.STATE_SET, ['forward', 'back', 'unknown']),
>         e.enum('inching_night', ea.STATE_SET, ['disable', 'enable', 'unknown']),
>         e.enum('inching_day', ea.STATE_SET, ['disable', 'enable', 'unknown']),
>     ],
> };
> 
> module.exports = definition;
> ```

Hello! After using your files, I have been able to control my device, Thanks a lot！ 
but, it seems I haven't understood its control logic. 
Sometimes, when I control it, the motor keeps running continuously and doesn't stop. 
Is there an issue with the way I'm operating it, or is it that I don't understand the device's control logic? 
I have an RF remote controller that smoothly opens and closes the day and night curtains.

closed 2025-03-03 by github-actions[bot]