Koenkk/zigbee2mqtt #24802: [Wrong device]: Moes ERS-10TZBVB-AA identified as Loginovo ZG-101ZL

ronnyadam, 2024-11-17:
### Link

https://www.aliexpress.com/item/1005005805357128.html?spm=a2g0o.productlist.main.13.2f0f34e7FK9nqT&algo_pvid=b376ddee-0e2e-4aa7-af9f-096a35fc0eca&algo_exp_id=b376ddee-0e2e-4aa7-af9f-096a35fc0eca-6&pdp_npi=4%40dis%21EUR%2120.91%2111.29%21%21%21155.81%2184.13%21%40211b61a417318463877681721ecd99%2112000034615134708%21sea%21DE%212266807991%21XZ&curPageLogUid=UsBuav1Cnias&utparam-url=scene%3Asearch%7Cquery_from%3A#nav-specification

### Model

Moes ERS-10TZBVB-AA

### Description

Smart Button

### Vendor

Moes

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/ERS-10TZBVB-AA.png

### Database entry

{"id":22,"type":"EndDevice","ieeeAddr":"0x70ac08fffec5cd30","nwkAddr":18655,"manufId":4098,"manufName":"_TZ3000_ja5osu5g","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096,57345],"outClusterList":[25,10,3,4,6,8,4096],"clusters":{"manuSpecificTuya_3":{"attributes":{"53265":"�\u0000\u0003�\u0003�\u0000Z\u0003�\u0003�\u0000\u0003�\u0000d\u0000�\u0003�\u0003�\u0000\u0001�\u0003 \u0001\u000e\u0003�\u0003�\u0001\u0001�\u0000\u0000\u0000�\u0000�\u0000�"}},"genBasic":{"attributes":{"65506":31,"65508":1,"65534":0,"modelId":"TS004F","manufacturerName":"_TZ3000_ja5osu5g","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":65,"powerSource":3}},"genOnOff":{"attributes":{"tuyaOperationMode":1}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0618","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0618","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1684586747},"lastSeen":1731845682150}

### Notes

I have this device -> https://www.zigbee2mqtt.io/devices/ERS-10TZBVB-AA.html

z2m identifies as https://www.zigbee2mqtt.io/devices/ZG-101ZL.html

jan-stemberg-ui, 2024-11-21:
I am facing the same problem.

roy81f, 2024-11-24:
> I am facing the same problem.

+1

ronnyadam, 2025-01-26:
exist an expectation of a solution for this issue?

closed 2025-06-09 by github-actions[bot]