Koenkk/zigbee2mqtt #24807: z2m restarts, when I am trying to do  OTA update

antst, 2024-11-17:
### What happened?

With 1.41 and Sonoff ember dongle, z2m always crashes on the course of OTA update.

I can't compare with earlier versions, as before  I had 1.40 and Sonoff zstack dongle.

I can exclude that this problem appears not only OTA update, it just I noticed it during one, as it is kind of reproducible.

### What did you expect to happen?

To finish OTA all the way, not crash somewhere is the middle (always at different percentage though)

### How to reproduce it (minimal and precise)

Z2M 1.41, sonoff-E dongle, do OTA update. Possibly, you need 100+ devices in your network :)

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

Sonoff ZBDongle-E (ember)

### Setup

z2m as HA add-on , supervised HA vm on proxmox. Serial port (USB device) directly exposed too VM.

### Debug log

```
[2024-11-17 22:43:20] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-11-17 22:43:20] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-11-17 22:43:20] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-11-17 22:43:20] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Total frames: RX=4981, TX=6482
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   DATA frames : RX=4969, TX=1499
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   DATA bytes  : RX=116328, TX=107333
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Retry frames: RX=10, TX=3
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=4979
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Retry dupes     : RX=10
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-11-17 22:43:20] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=3
[2024-11-17 22:43:29] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=217 Len=7].
[2024-11-17 22:43:29] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_NCP_FATAL_ERROR. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=217 Len=7].
[2024-11-17 22:43:29] info: 	z2m: Update of 'Living_light_center' at 67.06%, ≈ 2 minutes remaining
[2024-11-17 22:43:29] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living_light_center', payload '{"brightness":84,"color_mode":"color_temp","color_temp":500,"last_seen":"2024-11-17T22:43:14+01:00","linkquality":240,"power_on_behavior":null,"state":"ON","update":{"installed_version":16786946,"latest_version":16787456,"progress":67.06,"remaining":122,"state":"updating"},"update_available":null}'
[2024-11-17 22:43:29] error: 	zh:controller:device: Read response to 0x00158d00030223d7 failed (ZCL command 0x00158d00030223d7/1 genTime.readRsp({"time":785194996}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":107,"writeUndiv":false}) failed (~x~> [ZCL to=0x00158d00030223d7:48745 apsFrame={"profileId":260,"clusterId":10,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0}] Failed to send request with status=FAIL.))
```

Koenkk, 2024-11-18:
Try the 8.0.2 firmware (https://www.zigbee2mqtt.io/guide/adapters/emberznet.html)

antst, 2024-11-18:
It was very very bad idea ))))
whit 8.02 it restarts, in similar way, non-stop. I don't even need an OTA for that anymore )))

antst, 2024-11-18:
Actually, flashed 7.4.4 back. then flashed updated version of 8.0.2, which (according to the commit log) touches nothing in relevant part of the code, and got at least some stability back. Not as good as 7.4.4 was (which was restarting only during OTA), but not as bad as first attempt with 8.0.2 was.  It is slightly more unstable than 7.4.4. But none of them is stable enoug to do OTA updates in my network :(
But! To be honest, aside of this issue, my network seems to be more stable than was with zstack. At least this is  impression of first two days though. So, after a month I might see it differently)))

antst, 2024-11-19:
I looked around, it seems to be not unusual problem with ember sticks.
there must be something common between those who experience it.
It can be just dense network, but it could be some particular device in the network, which we all have )

cedricbarbecue, 2024-11-22:
hello
same issue for me on OTA update for a SONOFF device
I have the same z2m release 1.41.0 and sonof ember stick 7.4.4

kusk, 2024-11-28:
Same problem with 7.4.5

antst, 2024-11-29:
I can confirm, that in "empty network", where only ember coordinator and couple devices, OTA works, even if I have 5 devices in this network and simultaneously update them.

Problem appears in busy network. Potentially, there is some spammy devices or whatever, which all of us who has problem, have.

antst, 2024-11-29:
Dare to share list of devices you have?
you can get list of devices types by 
```
jq -c '[.modelId,.manufName]' database.db | sort -u
```

Nerivec, 2024-12-03:
Regrouping in https://github.com/Koenkk/zigbee2mqtt/issues/23861