Koenkk/zigbee2mqtt #24822: [New device support]: TS0601 TZE204_nqqylykc

gedas07, 2024-11-18:
### Link

https://es.aliexpress.com/item/1005006832261783.html?spm=a2g0o.productlist.main.3.6293MoKcMoKcNR&algo_pvid=9854aa72-31ba-4c32-a548-3b444f90e431&algo_exp_id=9854aa72-31ba-4c32-a548-3b444f90e431-1&pdp_npi=4%40dis%21EUR%218.77%212.81%21%21%219.03%212.90%21%40210318ec17319652606493769e7f73%2112000038447232552%21sea%21ES%210%21ABX&curPageLogUid=bQHXQHDBkDA1&utparam-url=scene%3Asearch%7Cquery_from%3A

### Database entry

{"id":67,"type":"Router","ieeeAddr":"0xa4c1387afaab2279","nwkAddr":45740,"manufId":4417,"manufName":"_TZE204_nqqylykc","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":";u�.f<u�.\u0012<u�.e","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1731965080532}

### Zigbee2MQTT version

1.41.0-dev commit: 152ad12

### Comments

NO se como hacerlo compatible

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_nqqylykc',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

No he sabido crearla

gedas07, 2024-12-07:
Thanks a lot, but i don't know how can i add . 
The last week i put in the configuration.yaml this:
external_converters:
    - ts0601.js

an add this :`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_nqqylykc',
        },
    ],

    model: 'ZDMS16-1',
    vendor: 'Avatto',
    description: 'Zigbee 1 channel Dimmer',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
            tuya.exposes.lightBrightnessWithMinMax(),
            tuya.exposes.countdown(),
            tuya.exposes.lightType(),
            e.power_on_behavior().withAccess(ea.STATE_SET),
            tuya.exposes.switchType(),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
            [3, 'min_brightness', tuya.valueConverter.scale0_254to0_1000],
            [4, 'switch_type', tuya.valueConverterBasic.lookup({'toggle': tuya.enum(0), 'state': tuya.enum(1), 'momentary': tuya.enum(2)})],
            [5, 'max_brightness', tuya.valueConverter.scale0_254to0_1000],
            [6, 'countdown', tuya.valueConverter.countdown],
            [14, 'power_on_behavior', tuya.valueConverter.powerOnBehaviorEnum],
            [13, 'light_type', tuya.valueConverter.lightType],
        ],
    },
};

module.exports = definition;`

And works perfectly , but today I updated to a new version , an doesn´t work anymore .
How can i resolve?

bernardesarthur, 2024-12-28:
I have some device here but different ManufacturerName: _TZE204_nqqylykc

Someone can help me?