Koenkk/zigbee2mqtt #24823: [New device support]: TS0505B (TS0505B_1) / _TZ3210_bfwvfyx1

ClusterM, 2024-11-19:
### Link

https://www.aliexpress.com/item/1005006724896566.html

### Database entry

{"id":18,"type":"Router","ieeeAddr":"0xa4c138d49643100f","nwkAddr":40220,"manufId":4417,"manufName":"_TZ3210_bfwvfyx1","powerSource":"DC Source","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,57346,61184,60928,57344,4096],"outClusterList":[3,10,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_bfwvfyx1","powerSource":4,"zclVersion":3,"appVersion":162,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017"}},"genOnOff":{"attributes":{"onOff":1}},"lightingColorCtrl":{"attributes":{"colorCapabilities":29}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":162,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1732013350107}

### Zigbee2MQTT version

1.41.0 commit: 69f72829

### Comments

This is RGB+CW+WW lamp, Tuya based. ZigBee2MQTT shows it as supported
[image]

But:
* RGB control doesn't work at all. I can select "Color (X/Y)" but nothing changes. Seems like Z2M can't change mode from white to RGB. 
* Color temperature range is much smaller. I can set much warmer white light using official Tuya up.
* Seems like icon is wrong too.

I tried to add `_TZ3210_bfwvfyx1` to the `node_modules/zigbee-herdsman-converters/devices/tuya.js` file in the `tuya.whitelabel('Tuya', 'TS0505B_1_1', 'Zigbee 3.0 18W led light bulb E27 RGBCW'` but without any success, only icon changed.

### External definition

```shell
const {light, commandsOnOff, commandsLevelCtrl, commandsColorCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0505B'],
    model: 'TS0505B',
    vendor: '_TZ3210_bfwvfyx1',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[0,1000]},"color":{"modes":["xy","hs"]}}), commandsOnOff(), commandsLevelCtrl(), commandsColorCtrl()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

* RGB control still doesn't work
* Color temperature works fine (full range) but HASS can't understand such color temperature range: `math.floor(1000000 / mired_temperature)` -> `division by zero`

ClusterM, 2024-11-19:
upd:
The RGB control begins to work after the following lines are added (copy-pasted from `TS0505B_2`):
```
        toZigbee: [toZigbee_1.default.on_off, tzLocal.led_control, tuya.tz.do_not_disturb],
        fromZigbee: [fromZigbee_1.default.on_off, fromZigbee_1.default.tuya_led_controller, fromZigbee_1.default.brightness, fromZigbee_1.default.ignore_basic_report],
        exposes: [e.light_brightness_colortemp_colorhs([143, 500]).removeFeature('color_temp_startup'), tuya.exposes.doNotDisturb()],
```

ClusterM, 2024-11-19:
upd: [143, 1000] color temperature allows to set really warm white color.

will-tm, 2024-11-24:
@ClusterM just seeing this, it seems you and I went through the same invertigation with this device, I just got a [pull request](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8361) merged that addressed this issue in a similar way

ESP4Ever, 2024-12-02:
Hi!

I recently bought a 5x pack of TS0505B (_TZ3210_bfwvfyx1) bulbs. They came on a different FW versions and only one of them is working normally with Z2M(((

Zigbee Model: TS0505B
Zigbee Manufacturer: _TZ3210_bfwvfyx1
Model: TS0505B_1
Compatible lamp shows FW v1.2.5 via Tuya gateway and no FW version at all via ZigBee2MQTT "About"
Not compatible ones - FW v2.2.2 and v0122052017 accordingly

The problem I see here is that Z2M can't distinguish them and apply a correct converter automatically

will-bdh, 2024-12-02:
@ESP4Ever use Z2m version 1.42.0 that got released yesterday and you should be good

ESP4Ever, 2024-12-02:
> @ESP4Ever use Z2m version 1.42.0 that got released yesterday and you should be good

I'm away this week so I checked only GET functionality and it still fails. On top of it I suspect that the situation can be the opposite now: new FW works, old is not because as I mentioned earlier, the product fingerprint doesn't take FW version into account

closed 2025-02-09 by github-actions[bot]