Koenkk/zigbee2mqtt #24825: [New device support]: TS0601 TZE204_3regm3h6 - Smart thermostat - Pilot wire

sb18-63, 2024-11-19:
### Link

https://a.aliexpress.com/_EGs0JO5 

### Database entry

{"id":14,"type":"Router","ieeeAddr":"0xa4c138eb21e67ed4","nwkAddr":43857,"manufId":4417,"manufName":"_TZE204_3regm3h6","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":" m .\u0013\u001dn .e\u001en .\u0012\u001en .e#n .e$n .e%n .\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}},"genGroups":{"attributes":{"nameSupport":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1732022586433}

### Zigbee2MQTT version

1.41.0-1

### Comments

Hello,

I recently bought at AliExpress Zigbee devices to manage heating radiators through pilot wire.
It able management using 6 pilot wire modes (Comfort, Comfort -1°C, Comfort -2°C, Eco, No Frost, Stop) and include temperature probe as well as power consumption monitoring.

It works pretty well with SmartLife app, but I would like to add them in my HA.
It was discovered as "unsupported device" but after adding custom TS0601.js add-on file, it appears now as "supported"

However, it seems that it is not sufficient to use this device with HA / Zigbee2MQTT.
I have no idea how I can setup additional definition to get information from this device and manage it properly in HA

Many thanks in advance for any useful information.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_3regm3h6',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I get no information from _TZE204_3regm3h6 devices in HA / Zigbee2MQTT

Zelox69, 2024-12-14:
Hello,

I just managed to integrate my file

external_converters:
- tze204_3regm3h6.js

but it is still very buggy…

here is the code:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_3regm3h6']),
    model: 'TS0601_3regm3h6',
    vendor: 'Tuya',
    description: 'Appareil de gestion de chauffage par fil pilote',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        exposes.enum('mode', ea.ALL, [
            'comfort_minus_2',
            'comfort_minus_1',
            'comfort',
            'eco',
            'anti_frost',
            'off',
            'program'
        ]).withDescription('Mode de fonctionnement du chauffage'),
        e.temperature().withUnit('°C').withDescription('Température mesurée'),
        e.power().withUnit('W').withDescription('Puissance électrique consommée'),
        e.energy().withUnit('kWh').withDescription('Énergie consommée totale'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'mode', {
                from: (value) => {
                    if (value === true) return 'comfort'; // Par défaut, 'true' correspond à 'comfort'
                    if (value === false) return 'off';   // Par défaut, 'false' correspond à 'off'
                    const mapping = {
                        5: 'comfort_minus_2',
                        4: 'comfort_minus_1',
                        0: 'comfort',
                        1: 'eco',
                        2: 'anti_frost',
                        3: 'off',
                        6: 'program',
                    };
                    return mapping[value] || `unknown_${value}`;
                },
                to: (value) => {
                    const reverseMapping = {
                        'comfort_minus_2': 5,
                        'comfort_minus_1': 4,
                        'comfort': 0,
                        'eco': 1,
                        'anti_frost': 2,
                        'off': 3,
                        'program': 6,
                    };
                    return reverseMapping[value] || 0; // Par défaut, retourner 'comfort' (0)
                },
            }],
            [101, 'temperature', tuya.valueConverter.divideBy10],
            [102, 'power', tuya.valueConverter.raw],
            [103, 'energy', tuya.valueConverter.divideBy100],
        ],
        supportsGet: true, // Permet d'utiliser la commande GET pour récupérer l'état
    },
    endpoint: (device) => {
        return {
            l1: 1,
        };
    },
};

module.exports = definition;
```

Gexwh, 2024-12-29:
Hello,
I am in the game too. Thank you.
Anyone did progress? I am not that good here, I will check if we can improve this

icarius, 2024-12-30:
I've tested [Zelox69](https://github.com/Zelox69)'s version, which recognizes the equipment, but it's impossible to remotely control the state change.
I do get the values back:
 - Temperature
 - Power 
 - Energy
I've made a modification to the temperatures to get, for example, 32.5°C
[101, 'temperature', tuya.valueConverter.divideBy100], // Correction of temperature conversion
> [101, 'temperature', tuya.valueConverter.divideBy10],

Thanks [Zelox69](https://github.com/Zelox69)

Gexwh, 2024-12-30:
Well, it looks like complicated to me.

The temperature is the setpoint or the measured temperature?

Zelox69, 2024-12-30:
helllo,

voici les datapoints:

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/lambe/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/lambe/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#467886" vlink="#96607D">

1 | Switch
-- | --
2 | mode
11 | current power
16 | current Temp.
17 | Windows State
19 | Temp. Correction
20 | Fault Alarm
29 | Windows Check
101 | Current Voltage
102 | Cur Current
39 | Child Lock
103 | Temp. Sensitivy
104 | Today's Energy
105 | Yesterday's Energy
106 | Device mode Type
107 | Total Energie
108 | Weekly timing data_1
109 | Weekly timing data_2
110 | Weekly timing data_3
111 | Weekly timing data_4
112 | Weekly timing data_5
113 | Weekly timing data_6
50 | Set Target Temp.
114 | Weekly timing data_7
115 | Set Temp. Switch
116 | Antifrost Switch
117 | ECO Temp. Set
118 | Comfort Temp. Set
119 | HORS Temp. Set
120 | Lighting control

</body>

</html>

Gexwh, 2024-12-30:
Where did you get this table?

I just modified the code and got it working, I will test it live and share

Zelox69, 2024-12-30:
> Where did you get this table?
> 
> I just modified the code and got it working, I will test it live and share

I got it back thanks to this tutorial and a Zigbee gateway

https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#_6-find-your-devices

Gexwh, 2024-12-30:
What do you think of:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_3regm3h6']),
    model: 'TS0601_3regm3h6',
    vendor: 'Tuya',
    description: 'Appareil de gestion de chauffage par fil pilote',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [
        {
            key: ['mode'],
            convertSet: async (entity, key, value, meta) => {
                const reverseMapping = {
                    'comfort_minus_2': 5,
                    'comfort_minus_1': 4,
                    'comfort': 0,
                    'eco': 1,
                    'anti_frost': 2,
                    'off': 3,
                    'program': 6,
                };
                const dpValue = reverseMapping[value];
                if (dpValue === undefined) {
                    throw new Error(`Mode non valide : ${value}`);
                }
                await tuya.sendDataPointEnum(entity, 2, dpValue);
            },
            convertGet: async (entity, key, meta) => {
                await tuya.sendDataPointValue(entity, 2);
            },
        },
        {
            key: ['target_temperature'],
            convertGet: async (entity, key, meta) => {
                await tuya.sendDataPointValue(entity, 50);
            },
            convertSet: async (entity, key, value, meta) => {
                // On convertit la température en divisant par 10 avant d'envoyer
                const dpValue = Math.round(value * 10); // convertir en dixièmes
                await tuya.sendDataPointValue(entity, 50, dpValue);
            },
        },
        {
            key: ['current_temperature'],
            convertGet: async (entity, key, meta) => {
                await tuya.sendDataPointValue(entity, 16);
            },
        },
        {
            key: ['anti_frost'],
            convertSet: async (entity, key, value, meta) => {
                const dpValue = value === 'ON' ? 1 : 0;
                await tuya.sendDataPointBool(entity, 116, dpValue);
            },
            convertGet: async (entity, key, meta) => {
                await tuya.sendDataPointValue(entity, 116);
            },
        },
        {
            key: ['child_lock'],
            convertSet: async (entity, key, value, meta) => {
                const dpValue = value === 'ON' ? 1 : 0;
                await tuya.sendDataPointBool(entity, 39, dpValue);
            },
            convertGet: async (entity, key, meta) => {
                await tuya.sendDataPointValue(entity, 39);
            },
        },
    ],
    exposes: [
        exposes.enum('mode', ea.ALL, [
            'comfort_minus_2',
            'comfort_minus_1',
            'comfort',
            'eco',
            'anti_frost',
            'off',
            'program',
        ]).withDescription('Mode de fonctionnement du chauffage'),
        e.numeric('current_temperature', ea.STATE)
            .withUnit('°C')
            .withDescription('Température mesurée'),
        e.numeric('target_temperature', ea.ALL)
            .withUnit('°C')
            .withDescription('Température cible pour le chauffage'),
        e.binary('anti_frost', ea.ALL, 'ON', 'OFF')
            .withDescription('Mode anti-gel activé/désactivé'),
        e.binary('child_lock', ea.ALL, 'ON', 'OFF')
            .withDescription('Verrouillage enfant activé/désactivé'),
        e.numeric('current_power', ea.STATE)
            .withUnit('W')
            .withDescription('Puissance électrique consommée'),
    ],
    meta: {
        tuyaDatapoints: [
            [2, 'mode', {
                from: (value) => {
                    const mapping = {
                        5: 'comfort_minus_2',
                        4: 'comfort_minus_1',
                        0: 'comfort',
                        1: 'eco',
                        2: 'anti_frost',
                        3: 'off',
                        6: 'program',
                    };
                    return mapping[value] || `unknown_${value}`;
                },
                to: (value) => {
                    const reverseMapping = {
                        'comfort_minus_2': 5,
                        'comfort_minus_1': 4,
                        'comfort': 0,
                        'eco': 1,
                        'anti_frost': 2,
                        'off': 3,
                        'program': 6,
                    };
                    return reverseMapping[value];
                },
            }],
            // Correction de la conversion de la température cible
            [50, 'target_temperature', {
                from: (value) => value / 10, // Diviser la valeur par 10
                to: (value) => Math.round(value * 10), // Multiplier par 10 lors de l'écriture
            }],
            // Correction de la conversion de la température actuelle
            [16, 'current_temperature', {
                from: (value) => value / 10, // Diviser la valeur par 10
                to: (value) => Math.round(value * 10), // Multiplier par 10 lors de l'écriture
            }],
            [116, 'anti_frost', tuya.valueConverter.raw],
            [39, 'child_lock', tuya.valueConverter.raw],
        ],
        supportsGet: true,
    },
    endpoint: (device) => {
        return {
            l1: 1,
        };
    },
};

module.exports = definition;

Zelox69, 2024-12-31:
Hello,

The version you gave me reports an unrecognized device.

Here is my version but I still have 2-3 bugs.

```typescript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

let currentVoltage = 230; // Valeur par défaut pour la tension

const definition = {
    fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE204_3regm3h6' }],
    model: 'TS0601_3regm3h6',
    vendor: 'Tuya',
    description: 'Gestion de chauffage par fil pilote avec synchronisation de l\'interface',

    fromZigbee: [
        tuya.fz.datapoints,
        {
            cluster: 'manuSpecificTuya',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                const dp = msg.data.dp; // Datapoint
                const value = msg.data.value; // Valeur brute

                switch (dp) {
                    case 1: // Datapoint pour l'état ON/OFF
                        return { state: value ? 'ON' : 'OFF' };
                    case 2: // Datapoint pour le mode de fonctionnement
                        const modes = {
                            5: 'comfort_minus_2',
                            4: 'comfort_minus_1',
                            0: 'comfort',
                            1: 'eco',
                            2: 'anti_frost',
                            3: 'off',
                            6: 'program',
                        };
                        return { mode: modes[value] || `unknown_${value}` };
                    case 101: // Datapoint pour la tension actuelle
                        currentVoltage = value / 10;
                        return { voltage: currentVoltage };
                    case 102: // Datapoint pour le courant
                        const power = Math.round((value * currentVoltage) / 1000);
                        return { current_power: power };
                    case 16: // Datapoint pour la température actuelle
                        return { current_temperature: value / 10 };
                    case 119: // Datapoint pour la température hors gel
                        return { antifrost_temperature: value / 10 };
                    case 117: // Datapoint pour la température ECO
                        return { eco_temperature: value / 10 };
                    case 118: // Datapoint pour la température confort
                        return { comfort_temperature: value / 10 };
                    case 50: // Datapoint pour la température cible
                        return { target_temperature: value / 10 };
                    default:
                        meta.logger.warn(`Datapoint non géré : ${dp}, valeur : ${value}`);
                        return null;
                }
            },
        },
    ],

    toZigbee: [
[... 152 lines cut ...]
                const modes = {
                    'comfort_minus_2': 5,
                    'comfort_minus_1': 4,
                    'comfort': 0,
                    'eco': 1,
                    'anti_frost': 2,
                    'off': 3,
                    'program': 6,
                };
                return modes[value];
            }}],
            [101, 'voltage', { from: (value) => value / 10 }],
            [102, 'current_power', { from: (value) => Math.round((value * currentVoltage) / 1000) }],
            [16, 'current_temperature', { from: (value) => value / 10 }],
            [119, 'antifrost_temperature', { from: (value) => value / 10, to: (value) => Math.round(value * 10) }],
            [117, 'eco_temperature', { from: (value) => value / 10, to: (value) => Math.round(value * 10) }],
            [118, 'comfort_temperature', { from: (value) => value / 10, to: (value) => Math.round(value * 10) }],
            [50, 'target_temperature', { from: (value) => value / 10, to: (value) => Math.round(value * 10) }],
        ],
    },
};

module.exports = definition;

```

yoFra, 2025-02-14:
Hi all,

I try to add the code above in an added Z2M Extention file named 'TS0601_3regm3h6.js'.
When I save it, i get the following error : 

> TS0601_3regm3h6.js contains invalid code: moduler is not a constructor.

Could you explain me why ?

Is there a change in the way to code an Z2M extension ?

    .YO.

Fyngah, 2025-03-22:
Hi all,

Did you all get your Thermostat working?
I just got one from AliExpress and if I'm here it's because I can't get it working properly.
I'm using Home Assistant with a Skyconnect.
I want to get it working with zha but I think I don't have all the clusters.
Here is my Signature
```
{

  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_3regm3h6",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

And here is the data I get when I run a scan device command with zha_toolkit.
```
zha_toolkit_version: v1.1.27
zigpy_version: 0.77.1
zigpy_rf_version: 0.43.0
ieee_org:
  - 64
  - 35
  - 92
  - 209
  - 51
  - 56
  - 193
  - 164
ieee: a4:c1:38:33:d1:5c:23:40
[... 373 lines cut ...]
                - Analog
              access: READ
              access_acl: 1
              attribute_value: 0
            "0xfffd":
              attribute_id: "0xfffd"
              attribute_name: cluster_revision
              value_type:
                - "0x21"
                - uint16_t
                - Analog
              access: READ
              access_acl: 1
              attribute_value: 3
          commands_received: {}
          commands_generated: {}
    - id: 242
      device_type: "0x0061"
      profile: "0xa1e0"
success: true
```

What do you think about this data?
I'm pretty sure I don't have all the clusters but how can I find them?
Any help would be much appreciated.

closed 2025-05-29 by github-actions[bot]

dbn-b4e, 2025-08-13:
Hello, I managesd to get full description of DP from Tuya and finished a full converter if required

dbn-b4e, 2025-08-13:
[TS0601_3regm3h6_V2.txt](https://github.com/user-attachments/files/21749795/TS0601_3regm3h6_V2.txt)

dbn-b4e, 2025-08-13:
Also did external converter for better device from Tuya TS0601_x9usygq1
Can only source through factory, waiting from price info should be in the 15€ range, someone is interested?

FYI I am stuck with f**** javascript to encode / decode the simple payload for each day. Not very useful to me but I wanted to be complete

mmorelon, 2025-09-13:
Hello, I made an external converter for this device and the request to add it to converter has been done : https://github.com/Koenkk/zigbee-herdsman-converters/issues/10000

mmorelon, 2025-09-13:
Waiting integration, this is the external converter :

[TS0601_3regm3h6.js](https://github.com/user-attachments/files/22309004/TS0601_3regm3h6.js)

dbn-b4e, 2025-09-13:
Also device trademark and reference are not correct.

@Koen wacht een beejte daarvoor ik zal de modificatie doen.

Mvg 
David

⁣Télécharger BlueMail pour Android ​

mmorelon, 2025-09-13:
Hi,
I did the external converter for model TS0601_3regm3h6 by myself and it is working. All functions are working except the scheduling feature which is a tuya's specific feature and isn't needed if you use Z2M with HA so i didn't try to espose this feature. Just put the file in your data/external_converters/ folder on your Z2M installation then restart Z2M. You will see that the device is recognized and you can pilot it.

666dsa666, 2025-12-23:
Would anyone with that device be able to check if your version is the same as mine ?

While recognized, I can't get readings for power/voltage etc and pilot wire seems to be not working as the heater is stuck in confort mode.
Changing the state On/Off makes no difference, output relay seems to be stuck closed (which may be linked to the option that set if the heater has a memory or not but unsure)
Thermostat mode change seems to be working visually but the system mode is always N/A and output voltage is not what expected (5v, 73v etc...)
I confirmed the issue is not on the heater side or it's configuration by connecting it's pilot wire to P and N and got confort / eco working.

I wonder if I got a non-working module or if I somehow got a V2/V1.2

[image]

[image]