Koenkk/zigbee2mqtt #24826: Building error in lib/util/logger.ts on 1.41.0 build

alesf, 2024-11-19:
### What happened?

When building the package an error is thrown.

 ```bash
% npm run build

> zigbee2mqtt@1.41.0 build
> tsc && node index.js writehash

lib/util/logger.ts:117:57 - error TS2322: Type 'unknown' is not assignable to type 'string'.

117                 format: winston.format.printf((info) => info.message),
                                                            ~~~~~~~~~~~~

  node_modules/logform/index.d.ts:53:37
    53   function printf(templateFunction: (info: TransformableInfo) => string): Format;
                                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    The expected type comes from the return type of this signature.

Found 1 error in lib/util/logger.ts:117
```

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Zi-Stick

### Setup

plain, MacOS (node 20 and 22)

r4ptor, 2024-11-20:
Quick workaround is to cast info.message to a string

edit lib/util/logger.ts 

``
const options: KeyValue = {
                app_name: 'Zigbee2MQTT',                                                                                                format: winston.format.printf((info) => String(info.message)),                                                          ...settings.get().advanced.log_syslog,
            };``

So just add String() around info.message.

Build succeeds after that change.

alesf, 2024-11-20:
Sure, just putting it out there since I didn't see it reported.

Koenkk, 2024-11-20:
After upgrading, did you execute a `npm ci`?

closed 2025-01-28 by github-actions[bot]