Koenkk/zigbee2mqtt #24832: SONOFF TRVZB can push data but doesn't receive any

benjR, 2024-11-19:
### What happened?

Hello,

Using:
zigbe2mqtt 1.41.0
Version de l'interface 0.7.4
Version Zigbee-herdsman-converters 20.38.0
Version Zigbee-herdsman 2.1.7
Type de coordinateur ConBee2/RaspBee2
Révision du coordinateur 0x26720700

I use multiple SONOFF TRVZB to control my temperature and I have centralized the temperature management / scheduling onto HA. It didn't cause any particular issue for a while (and worked well last year). 

Since a few days, I can see that the temperature doesn't change anymore on one or two TRV at a time (not in sync), they receive all information from the TRV (current temp, target temp) and they update very quick when I change something on the TRV, the problem is that they can't receive anything anymore whatever the info sent (temperature, schedule, mode, etc...).

It's the same via Z2M (so not HA issue), restarting Z2M doesn't do anything, to have them back up I have to pair them again, sometimes it doesn't work and I have to do it again, then it will work for some time and stop working again.

I also tried updating the TRVs to the latest firmware available, and this doesn't change anything.

I always get some kind of timeout such as in logs attached.

Sometimes the timeout takes a few seconds and sometimes the timeout is almost instant, if the timeout occured once, there is nothing to do to get the device back.

It happened on several TRVs but it seems that I have the issue more on some of them.
Will now try to hard reset one of them to see if it's better and report

If you have any idea?
Thanks for your help!

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

Running via Docker

### Debug log

```
error 2024-11-19 19:57:22z2m: Publish 'set' 'occupied_heating_setpoint' to 'TRV Bureau S' failed: 'Error: ZCL command 0x0ceff6fffedc1b76/1 hvacThermostat.write({"occupiedHeatingSetpoint":1900}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (39) Error: waiting for response TIMEOUT)'

error 2024-11-19 20:00:38z2m: Publish 'set' 'occupied_heating_setpoint' to 'TRV Bureau S' failed: 'Error: ZCL command 0x0ceff6fffedc1b76/1 hvacThermostat.write({"occupiedHeatingSetpoint":750}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (ZCL command 0x0ceff6fffedc1b76/1 hvacThermostat.write({"occupiedHeatingSetpoint":750}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (51) Error: waiting for response TIMEOUT))'

error 2024-11-19 20:00:38z2m: Publish 'set' 'occupied_heating_setpoint' to 'TRV Bureau S' failed: 'Error: ZCL command 0x0ceff6fffedc1b76/1 hvacThermostat.write({"occupiedHeatingSetpoint":850}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (52) Error: waiting for response TIMEOUT)'

error 2024-11-19 20:10:33z2m: Publish 'set' 'occupied_heating_setpoint' to 'TRV Bureau S' failed: 'Error: ZCL command 0x0ceff6fffedc1b76/1 hvacThermostat.write({"occupiedHeatingSetpoint":1400}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (54) Error: waiting for response TIMEOUT)'
info 2024-11-19 20:10:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'occupied_heating_setpoint' to 'TRV Bureau S' failed: 'Error: ZCL command 0x0ceff6fffedc1b76/1 hvacThermostat.write({\"occupiedHeatingSetpoint\":1400}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (no response received (54) Error: waiting for response TIMEOUT)'","meta":{"friendly_name":"TRV Bureau S"},"type":"zigbee_publish_error"}'
```

jj-uk, 2024-11-20:
I'm seeing an issue where I can publish a mqtt message to the /set/ topic of the TRVZB device, but Z2M doesn't pass the json formatted data to the device, and there's no error in the logs.

I'm wondering if this is the method (mqtt) you're using to send data to the device, or are you using the Frontend GUI ?

Otherwise, i'll open a separate issue.

benjR, 2024-11-20:
Couldn't delete so "Force-"deleted one of them to be able to pair it again, will update accordingly

leperroy, 2024-11-24:
I have exactly the same issue. 

Coordinator is SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

Zigbee2MQTT version : 1.41.0
Coordinator type :zStack3x0
Coordinator revision :20230507
Coordinator IEEE Address : 0x00124b002c3fdf6d
Frontend version : 0.7.4
zigbee-herdsman-converters version 20.38.0
zigbee-herdsman version : 2.1.7

 Version of firmware on TRVZB is 2.2.1.

alexcf, 2024-11-28:
Same error for me too. Although, mine has now graduated to "No network route" - although seems to be pushing updates.

alexcf, 2024-11-28:
@benjR @leperroy what version are your TRVZBs? I'm on 1.1.5, there is a 1.2.1 available, but since I can't connect, no OTA updates! 😒

benjR, 2024-11-28:
@alexcf all on 1.2.1 but one in 1.1.1 (I have the issue on both versions)

alexcf, 2024-11-29:
Random thought - what "parent" device is the TRV connected to? I can't quite see what mine's connected to, but my hunch is, it'll be a Hue light bulb. I'm running a firmware update now to see if that makes any difference.

alexcf, 2024-11-29:
OK, so I updated my Hue lamp, and I re-paired (turned the temp to "OFF" then held the top button for 3 seconds) and seems to be back in action.. I'll see how it goes and report back.

alexcf, 2024-12-01:
> OK, so I updated my Hue lamp, and I re-paired (turned the temp to "OFF" then held the top button for 3 seconds) and seems to be back in action.. I'll see how it goes and report back.

Still working for me. Might be an issue between the TRV and one of the repeaters.

benjR, 2024-12-01:
Thanks for the update @alexcf, mine are going through Legrand devices mostly, I'll check there is a difference between the ones that are working and the others.

alexcf, 2024-12-07:
Problem is back again! Will have a look, but it's basically read only. I'm going to force it to connect to a specific repeater rather than any to see if that helps.

alexcf, 2024-12-07:
> Problem is back again! Will have a look, but it's basically read only. I'm going to force it to connect to a specific repeater rather than any to see if that helps. 

Forced it to use a Hue bulb. Seems to be back again.

benjR, 2024-12-09:
It's been working on and off lately for me and not a lot of time to test, I can't find the reason why it's not working sometimes (some are going through a repeater, others not), it seems to be more stable after a while though (even if I'm not doing anything they seem to end up acting correctly after a while).

cartercloutm, 2025-01-03:
I have same issue.

sticilface, 2025-02-16:
Me to. I'm running about 8 of these sonoff TRVs but one of them stubbornly refuses to receive information. The moment I change the temp on the valve it works and updates. It appears to push updates to Z2M no bother.

I'll try repairing direct to the router. (Which I think it is anyway)

alexcf, 2025-02-16:
I ended up getting fed up with it getting into the weird unidirectional communication, so I’ve just bought a MOES one with the same/similar design to the SONOFF one - this one runs Tuya, hoping it’s a software issue with SONOFF!

closed 2025-04-26 by github-actions[bot]