Koenkk/zigbee2mqtt #24835: Missing "Local temperature" sensor for _TZE200_hue3yfsn

Feriman22, 2024-11-20:
### What happened?

Z2M and HA can see the local temperature, but HA has no sensor for this.

Inside Z2M, under exposes:
[image]

Inside HA, under Devices / MQTT / this device:
[image]

This is exactly what this device is all about:

```
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_hue3yfsn
Description: Thermostat radiator valve
Support status: Supported
IEEE Address: 0x94b216fffe45f2da
Network address: 0xF79D / 63389
Manufacturer: Tuya
Model: TV02-Zigbee
```

Could you please add this sensor as well to HA? Existing entities for this device:

```
Floor_Hallway_TRV
climate.floor_hallway_trv

Floor_Hallway_TRV
update.floor_hallway_trv

Floor_Hallway_TRV Battery
binary_sensor.floor_hallway_trv_battery_low

Floor_Hallway_TRV Boost timeset countdown
number.floor_hallway_trv_boost_timeset_countdown

Floor_Hallway_TRV Child lock
lock.floor_hallway_trv_child_lock

Floor_Hallway_TRV Comfort temperature
number.floor_hallway_trv_comfort_temperature

Floor_Hallway_TRV Eco temperature
number.floor_hallway_trv_eco_temperature

Floor_Hallway_TRV Error status
sensor.floor_hallway_trv_error_status

Floor_Hallway_TRV Frost protection
switch.floor_hallway_trv_frost_protection

Floor_Hallway_TRV Heating stop
switch.floor_hallway_trv_heating_stop

Floor_Hallway_TRV Holiday start stop
text.floor_hallway_trv_holiday_start_stop

Floor_Hallway_TRV Holiday temperature
number.floor_hallway_trv_holiday_temperature

Floor_Hallway_TRV Linkquality
sensor.floor_hallway_trv_linkquality

Floor_Hallway_TRV Online
switch.floor_hallway_trv_online

Floor_Hallway_TRV Open window
switch.floor_hallway_trv_open_window

Floor_Hallway_TRV Open window temperature
number.floor_hallway_trv_open_window_temperature

Floor_Hallway_TRV Schedule friday
sensor.floor_hallway_trv_schedule_friday

Floor_Hallway_TRV Schedule monday
sensor.floor_hallway_trv_schedule_monday

Floor_Hallway_TRV Schedule saturday
sensor.floor_hallway_trv_schedule_saturday

Floor_Hallway_TRV Schedule sunday
sensor.floor_hallway_trv_schedule_sunday

Floor_Hallway_TRV Schedule thursday
sensor.floor_hallway_trv_schedule_thursday

Floor_Hallway_TRV Schedule tuesday
sensor.floor_hallway_trv_schedule_tuesday

Floor_Hallway_TRV Schedule wednesday
sensor.floor_hallway_trv_schedule_wednesday

Floor_Hallway_TRV Temperature
number.floor_hallway_trv_local_temperature_calibration

Floor_Hallway_TRV Working day
select.floor_hallway_trv_working_day
```

Thanks you! (:

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20221226

### Adapter

SLZB-06 (CC2652P)

### Setup

Docker container

Feriman22, 2025-01-20:
Hi @Koenkk 
Is there any chance of implement this sensor by default?

Thanks!

Koenkk, 2025-01-20:
If it is not supported in z2m 2.0.0, it will be supported in the 1 February release (https://github.com/Koenkk/zigbee-herdsman-converters/blob/15f15aa151239c1d5af5d776449a6f8077d59da6/src/devices/tuya.ts#L4515)

Feriman22, 2025-02-04:
Temperature does not update at all after updating to v2.1.0

closed 2025-04-13 by github-actions[bot]