Koenkk/zigbee2mqtt #24845: Aqara Light Sensor T1 no data after successful interview

Chartreusito, 2024-11-20:
### What happened?

After a successful interview, the device never updates its attributes.

I don't even get some initial data.

### What did you expect to happen?

I expected the device to update its status and attributes. Values stays null.
A click on the device button does not wake it to update the values.
I tried keeping clicking the button during the interview, at least three different  times to no avail.
Tested with 2 devices manufactured in 2024/05 and 2024/10

### How to reproduce it (minimal and precise)

Add a [GZCGQ11LM](https://www.zigbee2mqtt.io/devices/GZCGQ11LM.html#aqara-gzcgq11lm) to z2m, and check if its values updates

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

SMlight SLZB06

### Setup

Add-on on HAOS on Promox node

### Debug log

[log.log](https://github.com/user-attachments/files/17837098/log.log)

vit5041, 2024-11-21:
> > Mine works well..
> 
> Well, i'd be happy to have some input. What is your setup/adapter and z2m and firmware versions ?

HA in Docker, latest version

Z2M:
Zigbee2MQTT version
1.41.0 commit: cc31cea
Coordinator type
zStack3x0
Coordinator revision
20240710

Sensor: 
Zigbee Model
lumi.sen_ill.agl01
Zigbee Manufacturer
LUMI
Description
Light sensor T1
Firmware build date
20200312
Firmware version
2019www.

Chartreusito, 2024-11-21:
What coordinator are you using ?
Apart from your docker type install, everything version/firmware is identical on my side

Chartreusito, 2024-12-02:
Well, it was my zigbee network choking on its 160+ devices. After a split of about 30 devices and a repair on buggy devices all is now ok.
I'd love to have more clear signs of when a network is full but i guess that does not happen to a lot of people.

Thanks @vit5041 for your answers

closed 2024-12-02 by Chartreusito