Koenkk/zigbee2mqtt #24846: Aqara H1M ZNQBKG26LM - Power outage memory not available on 3-gang but is available on 2-gang

andeekaii, 2024-11-20:
### What happened?

'Power outage memory' doesn't work on Aqara H1M triple-rocker switch ZNQBKG26LM
The Zigbee2MQTT page states that the power_outage_memory should be exposed.

I also own the double-rocker versions ZNQBKG25LM and the Power outage memory' option works as expected.

[image]

### Zigbee2MQTT version

Zigbee2MQTT version 1.41.0 commit: unknown

### Adapter firmware version

20230507

### Adapter

SONOFF Universal Zigbee 3.0 USB Dongle Plus Gateway Based on TI CC2652P + CP2102N

### Setup

Home Assistant Add-On

### Debug log

error 2024-11-20 21:54:38z2m: Publish 'set' 'power_outage_memory' to 'Light Switch - Kitchen' failed: 'Error: Not supported'

ad-cdavila, 2024-12-07:
Same for me. Did you find any workaround?

andeekaii, 2024-12-07:
> Same for me. Did you find any workaround?

Unfortunately not, which is really frustrating as the two-gang switches work perfectly. I ended up creating a script that I can manually run after a power cut that sets all the light switches to the correct operation mode depending on their usage.

ad-cdavila, 2024-12-15:
I finally found a fix! I created the Pull Request for the library and it is already merged. So we need to wait for the next version to include these changes.

https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v21.4.0

andeekaii, 2024-12-15:
> I finally found a fix! I created the Pull Request for the library and it is already merged. So we need to wait for the next version to include these changes.
> 
> https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v21.4.0

Great news, thank you!

closed 2025-02-23 by github-actions[bot]