Koenkk/zigbee2mqtt #24853: Z2M UI refresh is blocking user input while OTA is being downloaded

Soukyuu, 2024-11-21:
### What happened?

When an OTA update is running, the UI displays the progress bar and the rotating arrows symbol on the device. This animation seems to be rapidly (<1s interval) restarting from the beginning and any user input has to be done between those restarts -> mashing the left click button to i.e. enter a device page.

### What did you expect to happen?

The animation should run smoothly, not restart all the time and to not block user input like the interview animation does

### How to reproduce it (minimal and precise)

- Start an OTA and wait for the download to begin
- Observe animation stuttering and restarting
- Attempt to enter a device page on the OTA or Devices page

### Zigbee2MQTT version

 1.41.0

### Adapter firmware version

26580700

### Adapter

deconz Conbee II

### Setup

Add-on on Home Assistant OS

Koenkk, 2024-11-22:
Please create issues related to the frontend in this repository: https://github.com/nurikk/zigbee2mqtt-frontend

closed 2024-11-22 by Koenkk

Soukyuu, 2024-11-22:
Oh, didn't know. Will do!