Koenkk/zigbee2mqtt #24858: [New device support]: Tuya TS0601 _TZE200_ogkdpgy2 CO2 sensor

VivantSenior, 2024-11-21:
### Link

https://www.aliexpress.com/item/1005004058860966.html

### Database entry

{"id":171,"type":"Router","ieeeAddr":"0xa4d458095989ca67","nwkAddr":5729,"manufId":4417,"manufName":"_TZE204_ogkdpgy2","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\b\u0000\u0000\u0000i\r\u0000\u0000\u0000i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1732206995568}

### Zigbee2MQTT version

1.41.0

### Comments

Hi. I bought the same NDIR CO2 sensor as in [#18594](https://github.com/Koenkk/zigbee2mqtt/issues/18594) but mine introduces itself as `_TZE204_ogkdpgy2` instead of `_TZE200_ogkdpgy2`. The sensor is powered by mains.

Some findings:
1. According to the [#23205](https://github.com/Koenkk/zigbee2mqtt/issues/23205) the round one has the same ID: `_TZE204_ogkdpgy2`.
2. I found [#23569](https://github.com/Koenkk/zigbee2mqtt/issues/23569) where square sensor has the same ID: `_TZE204_ogkdpgy2`.
3. On @blakadder's [page](https://zigbee.blakadder.com/Tuya_DCR-LCD.html) this CO2 is called DCR-LCD and its ID: `TS0601 | _TZE200_ogkdpgy2`
4. Found `_TZE204_ogkdpgy2` in the [Tuya.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/df5717b482f26bf106688f068f4fe76fd763dcb1/src/devices/tuya.ts#L1351) source but z2m didn't recognized the sensor before adding external converter.

Sensor looks like:
[image]

I created an external converter for CO2 but it doesn't use modernExtend and I'm not sure if there are some other metrics that this sensor exposes like CO2 calibration, internal temperature.

### External definition

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_ogkdpgy2',
        },
    ],
    model: 'TS0601_TZE204_ogkdpgy2',
    vendor: 'Tuya',
    description: 'NDIR CO2 sensor',
    extend: [],
    meta: {},
    fromZigbee: [legacy.fromZigbee.tuya_air_quality],
    toZigbee: [],
    exposes: [e.co2()],

	configure: async (device, coordinatorEndpoint, logger) => {
		device.powerSource = 'Mains (single phase)';
		device.type = 'Router';
		device.save();
	}
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Works: CO2 metrics

Doesn't work:
- The external converter doesn't use modernExtend
- I'm not sure if there are some other metrics that this sensor exposes like CO2 calibration, internal temperature.

When using [this external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_2-adding-your-device) to check TuYa data I can only see such logs:

> Received Zigbee message from '0xa4d458095989ca67', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,7,252],"type":"Buffer"},"datatype":2,"dp":2}],"seq":52736}' from endpoint 1 with groupID 0
>
> Received Zigbee message from '0xa4d458095989ca67', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":0,"appVersion":74}' from endpoint 1 with groupID 0

Koenkk, 2024-11-23:
This device has already been added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/8354

Will be supported in the 1 December release.

closed 2024-11-23 by Koenkk

VivantSenior, 2024-11-23:
@Koenkk thank you for the information but `TZE204_ogkdpgy2` doesn't expose temperature nor humidity. When I added temperature and humidity to the external converter their values are not being detected:
[image]