Koenkk/zigbee2mqtt #24861: Chacon ZB-ERSM-01 issue at 100%

cachavilla, 2024-11-21:
### What happened?

When the cover is at position 100, the state is closed, and also I have to enable the cover invert option.
[image]

### What did you expect to happen?

it must be ''open'' and ''close'' when it was 0

### How to reproduce it (minimal and precise)

always happen

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240716

### Adapter

zstack

### Setup

Add-on on home assistant OS

solideus, 2024-11-26:
Me too.

If I pair with the bluetooth and official app it works fine, but with zigbee2mqtt not

solideus, 2024-11-26:
Solved with:

[https://github.com/Koenkk/zigbee-herdsman-converters/commit/d0b843fe97f79c6cd883ca7d4b1a449d0cd2a653](url)

cachavilla, 2024-11-26:
Great news! how yo apply this change?

solideus, 2024-11-26:
Edit chacon.js from node_modules/zigbee-herdsman-converters/devices/ folder
Add " coverInverted: true," like the file of the link above
Or wait until the next update :-|

cachavilla, 2024-11-26:
Thank you very much, I'll wait for next update because I can't find this folder, I'm under Homeassistant OS.

closed 2025-02-02 by github-actions[bot]