Koenkk/zigbee2mqtt #24862: Issue with Zigbee2MQTT after restoring Home Assistant backup on new PC with Proxmox 8.3

Peres1992, 2024-11-22:
### What happened?

Hello,

I recently upgraded to a new PC and installed Proxmox 8.3 (I was using 7.0 before). After setting up Proxmox, I restored my Home Assistant backup, but now I’m facing an issue with Zigbee.

The logs show the following error:

yaml
Skopiuj kod
[01:20:22] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-11-22 01:20:24] info: 	z2m: Logging to console, file (filename: log.log)
[2024-11-22 01:20:24] info: 	z2m: Starting Zigbee2MQTT version 1.41.0 (commit #unknown)
[2024-11-22 01:20:24] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2024-11-22 01:20:25] error: 	z2m: Error while starting zigbee-herdsman
[2024-11-22 01:20:25] error: 	z2m: Failed to start zigbee
[2024-11-22 01:20:25] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-11-22 01:20:25] error: 	z2m: Exiting...
[2024-11-22 01:20:25] error: 	z2m: Error: Error while opening serialport 'Error: Error: No such device, cannot open /dev/ttyACM0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:225:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/@serialport/stream/dist/index.js:118:18
I tried changing the port to /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2416826-if00, but it didn’t help. I’m still seeing the same error.

I also enabled DECONZ for testing, and it worked, but I disabled it again. I’m unsure where to look for the problem now.

Any help would be greatly appreciated!

Thanks in advance!

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

i dont know right now.

### Adapter

Conbee 2

### Setup

i95 intel 

### Debug log

[01:09:22] INFO: Preparing to start...
[01:09:22] INFO: Socat not enabled
[01:09:22] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-11-22 01:09:24] info: 	z2m: Logging to console, file (filename: log.log)
[2024-11-22 01:09:24] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2024-11-22 01:09:24] info: 	z2m: Starting Zigbee2MQTT version 1.41.0 (commit #unknown)
[2024-11-22 01:09:24] debug: 	z2m: sd-notify loaded
[2024-11-22 01:09:24] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2024-11-22 01:09:24] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[20],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2416826-if00\",\"adapter\":\"deconz\"},\"adapter\":{\"disableLED\":false}}"'
[2024-11-22 01:09:25] debug: 	zh:deconz:driver: Set delay to 0
[2024-11-22 01:09:25] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[20]},"serialPort":{"path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2416826-if00","adapter":"deconz"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2024-11-22 01:09:25] debug: 	zh:deconz:driver: Opening with /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2416826-if00
[2024-11-22 01:09:25] debug: 	zh:deconz:driver: Serialport opened
[2024-11-22 01:09:25] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 1 paramId: 5
[2024-11-22 01:09:25] debug: 	zh:deconz:frameparser: PANID: 1a62
[2024-11-22 01:09:25] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 2 paramId: 11
[2024-11-22 01:09:25] debug: 	zh:deconz:frameparser: APS_EXT_PANID: 0xdddddddddddddddd
[2024-11-22 01:09:25] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 3 paramId: 28
[2024-11-22 01:09:35] debug: 	zh:deconz:driver: Timeout for request - CMD: 0xa seqNr: 3
[2024-11-22 01:09:35] error: 	z2m: Error while starting zigbee-herdsman
[2024-11-22 01:09:35] error: 	z2m: Failed to start zigbee
[2024-11-22 01:09:35] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-11-22 01:09:35] error: 	z2m: Exiting...
[2024-11-22 01:09:35] error: 	z2m: Error: TIMEOUT
    at Driver.processBusyQueue (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:515:28)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:168:40)
    at listOnTimeout (node:internal/timers:573:17)
    at processTimers (node:internal/timers:514:7)
/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();

chris-1243, 2024-11-22:
> I also enabled DECONZ for testing, and it worked, but I disabled it again. I’m unsure where to look for the problem now.

You have a Conbee II as coordinator. To specify the driver used in z2m `configuration.yamk` would make sense and you said it worked... You have your solution.

Peres1992, 2024-11-22:
[image]
deconz is ok but in zigbee is still problem. now i unstall deconz.

chris-1243, 2024-11-24:
As you updated Proxmox, are you completely sure you have set the `usb pass through` correctly for Z2M? I would check there first. Ensure Deconz is not working nor enabled. This may lead to conflict with Z2M.

> error: z2m: Error: Error while opening serialport 'Error: Error: No such device, cannot open /dev/ttyACM0'
at SerialPort. (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:225:28)
at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
at /app/node_modules/@serialport/stream/dist/index.js:118:18
I tried changing the port to /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2416826-if00, but it didn’t help. I’m still seeing the same error.

jeriks31, 2025-01-06:
Did you figure it out @Peres1992 ?

I just moved my HA from VMWare on a windows PC to a VM in unraid, and i'm having the same issue with my ConBee II

closed 2025-03-15 by github-actions[bot]