Koenkk/zigbee2mqtt #24868: [Wrong device]: Motion sensor detected as water leak sensor

robovanbasten, 2024-11-22:
### Link

https://www.zigbee2mqtt.io/devices/ZM-35H-Q.html

### Model

ZM-35H-Q

### Description

Motion Senor

### Vendor

Tuya

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/ZM-35H-Q.png

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0xa4c138f8b34d8d9f","nwkAddr":33829,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"SNZB-03","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002e1dd526","zoneState":1,"zoneType":13}},"genPollCtrl":{"attributes":{"checkinInterval":13200}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b002e1dd526","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002e1dd526","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1732278041386,"checkinInterval":3300}

### Notes

[image]
[image]

robovanbasten, 2024-11-22:
I was expecting this motion sensor to appear as Tuya (ZM-35H-Q), however zigbee2mqtt has detected it as a SONOFF (SNZB-03)? I have removed and repaired several time, same issue.

lyneld, 2024-11-22:
I have the same issue, after updating to 1.41.0-1. One of my motion sensors which was previously working fine now shows up as a water leak sensor. Tried removing and adding it again, but shows the same.

[image]

joaolrc, 2024-11-23:
Same issue here! 
I can add that on the box, it is written Model: EZ-P1
But seems to be the same, by the looks

lyneld, 2024-11-23:
Did some digging earlier today and it seems like there was an update to the zigbee-herdsman-converters a few days ago to fix this issue. Which hopefully means the next zigbee2mqtt update should solve this problem.

Koenkk, 2024-11-24:
Will be fixed in the 1 December release!

closed 2024-11-24 by Koenkk

SigiBaes, 2024-12-25:
Well my eWeLink leak sensor detects as a motion sensor...

OWYP87, 2025-01-15:
i`ve got the same issue as @SigiBaes , bought couple of water leak sensors and now all of them are added as motion sensor SNZB-03. Are there any way to solve it locally? Maybe edit some config file?

rhambus, 2025-09-22:
> same problem here: Tuya water leak sensor is detected as SNZB-03_ewelink motion sensor... Z2M-version: 2.6.0

I am having this very same issue.