Koenkk/zigbee2mqtt #24870: Avato TRV07 not reporting position, alarm and window

mwasowski, 2024-11-22:
### What happened?

After pairing Avatto TRV07 I was expecting to get position, alarm state and window properties according to [this](https://www.zigbee2mqtt.io/devices/TRV07.html#avatto-trv07). This is not the case. Window is actually reported, but it is a binary_switch for some reason and for that reason it shows `Unknown`. Other show this status because they are not reported correctly in Zigbee2mqtt:
[image]

### What did you expect to happen?

All properties that the TRV supports to be properly reported

### How to reproduce it (minimal and precise)

Pair Avatto TRV07 and after setup is complete, look at exposed entities.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

SMLIGHT CC2652P Zigbee USB Adapter SLZB-02

### Setup

Z2M on Rpi

### Debug log

```
[2024-11-22 14:52:19] info:     z2m: Zigbee: allowing new devices to join.
[2024-11-22 14:52:20] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"z39zh-4"}'
[2024-11-22 14:53:13] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138e9671a6646', payload '{"last_seen":"2024-11-22T13:53:13.178Z","linkquality":69}'
[2024-11-22 14:53:13] info:     zh:controller: Interview for '0xa4c138e9671a6646' started
[2024-11-22 14:53:13] info:     z2m: Device '0xa4c138e9671a6646' joined
[2024-11-22 14:53:13] info:     z2m: Starting interview of '0xa4c138e9671a6646'
[2024-11-22 14:53:13] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138e9671a6646","ieee_address":"0xa4c138e9671a6646"},"type":"device_joined"}'
[2024-11-22 14:53:13] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138e9671a6646","ieee_address":"0xa4c138e9671a6646","status":"started"},"type":"device_interview"}'
[2024-11-22 14:53:13] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138e9671a6646', payload '{"last_seen":"2024-11-22T13:53:13.305Z","linkquality":69}'
[2024-11-22 14:53:13] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138e9671a6646","ieee_address":"0xa4c138e9671a6646"},"type":"device_announce"}'
[2024-11-22 14:53:13] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138e9671a6646', payload '{"last_seen":"2024-11-22T13:53:13.965Z","linkquality":76}'
[2024-11-22 14:53:14] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138e9671a6646', payload '{"last_seen":"2024-11-22T13:53:14.248Z","linkquality":83}'
[2024-11-22 14:53:14] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138e9671a6646', payload '{"last_seen":"2024-11-22T13:53:14.561Z","linkquality":76}'
[2024-11-22 14:53:14] info:     zh:controller: Succesfully interviewed '0xa4c138e9671a6646'
[2024-11-22 14:53:14] info:     z2m: Successfully interviewed '0xa4c138e9671a6646', device has successfully been paired
[2024-11-22 14:53:14] info:     z2m: Device '0xa4c138e9671a6646' is supported, identified as: Avatto Thermostatic radiator valve (TRV07)
[2024-11-22 14:53:14] info:     z2m: Configuring '0xa4c138e9671a6646'
[2024-11-22 14:53:14] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c138e9671a6646/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c138e9671a6646/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138e9671a6646"],"manufacturer":"Avatto","model":"Thermostatic radiator valve (TRV07)","name":"0xa4c138e9671a6646","via_device":"zigbee2mqtt_bridge_0x00124b0024c0e92d"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"0xa4c138e9671a6646_battery","origin":{"name":"Zigbee2MQTT","sw":"1.41.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138e9671a6646","unique_id":"0xa4c138e9671a6646_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
[2024-11-22 14:53:14] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Thermostatic radiator valve","exposes":[{"access":1,"category":"diagnostic","description":"Remaining battery in %, can take up to 24 hours before reported","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"features":[{"access":3,"description":"Enables/disables physical input on the device","label":"State","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"label":"Child lock","type":"lock"},{"access":3,"description":"Maximum temperature","label":"Max temperature","name":"max_temperature","property":"max_temperature","type":"numeric","unit":"°C","value_max":35,"value_min":15},{"access":3,"description":"Minimum temperature","label":"Min temperature","name":"min_temperature","property":"min_temperature","type":"numeric","unit":"°C","value_max":15,"value_min":1},{"access":1,"description":"Position","label":"Position","name":"position","property":"position","type":"numeric","unit":"%"},{"features":[{"access":3,"description":"Enables/disables window detection on the device","label":"State","name":"state","property":"window_detection","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"label":"Window detection","type":"switch"},{"access":1,"description":"Window status closed or open ","label":"Window","name":"window","property":"window","type":"binary","value_off":"OPEN","value_on":"CLOSED"},{"access":1,"description":"Thermostat in error state","label":"Alarm switch","name":"alarm_switch","property":"alarm_switch","type":"binary","value_off":"OFF","value_on":"ON"},{"features":[{"access":1,"description":"Current temperature measured on the device","label":"Local temperature","name":"local_temperature","property":"local_temperature","type":"numeric","unit":"°C"},{"access":3,"description":"Temperature setpoint","label":"Current heating setpoint","name":"current_heating_setpoint","property":"current_heating_setpoint","type":"numeric","unit":"°C","value_max":35,"value_min":5,"value_step":0.5},{"access":3,"description":"Offset to add/subtract to the local temperature","label":"Local temperature calibration","name":"local_temperature_calibration","property":"local_temperature_calibration","type":"numeric","unit":"°C","value_max":30,"value_min":-30,"value_step":0.1},{"access":3,"description":"MANUAL MODE ☝ - In this mode, the device executes manual temperature setting. When the set temperature is lower than the \"minimum temperature\", the valve is closed (forced closed). AUTO MODE ⏱ - In this mode, the device executes a preset week programming temperature time and temperature. ON - In this mode, the thermostat stays open OFF - In this mode, the thermostat stays closed","label":"Preset","name":"preset","property":"preset","type":"enum","values":["auto","manual","off","on"]},{"access":1,"description":"Mode of this device","label":"System mode","name":"system_mode","property":"system_mode","type":"enum","values":["auto","heat","off"]},{"access":1,"description":"The current running state","label":"Running state","name":"running_state","property":"running_state","type":"enum","values":["idle","heat"]}],"type":"climate"},{"access":3,"description":"Schedule for monday, format: \"HH:MM/C HH:MM/C HH:MM/C HH:MM/C\"","label":"Schedule monday","name":"schedule_monday","property":"schedule_monday","type":"text"},{"access":3,"description":"Schedule for tuesday, format: \"HH:MM/C HH:MM/C HH:MM/C HH:MM/C\"","label":"Schedule tuesday","name":"schedule_tuesday","property":"schedule_tuesday","type":"text"},{"access":3,"description":"Schedule for wednesday, format: \"HH:MM/C HH:MM/C HH:MM/C HH:MM/C\"","label":"Schedule wednesday","name":"schedule_wednesday","property":"schedule_wednesday","type":"text"},{"access":3,"description":"Schedule for thursday, format: \"HH:MM/C HH:MM/C HH:MM/C HH:MM/C\"","label":"Schedule thursday","name":"schedule_thursday","property":"schedule_thursday","type":"text"},{"access":3,"description":"Schedule for friday, format: \"HH:MM/C HH:MM/C HH:MM/C HH:MM/C\"","label":"Schedule friday","name":"schedule_friday","property":"schedule_friday","type":"text"},{"access":3,"description":"Schedule for saturday, format: \"HH:MM/C HH:MM/C HH:MM/C HH:MM/C\"","label":"Schedule saturday","name [... cut ...] /20.0 12:00/15.0 18:00/22.0 22:00/15.0","schedule_monday":"06:00/20.0 12:00/15.0 18:00/22.0 22:00/15.0","schedule_saturday":"06:00/20.0 12:00/15.0 18:00/22.0 22:00/15.0","schedule_sunday":"06:00/20.0 12:00/15.0 18:00/22.0 22:00/15.0","schedule_thursday":"06:00/20.0 12:00/15.0 18:00/22.0 22:00/15.0","schedule_tuesday":"06:00/20.0 12:00/15.0 18:00/22.0 22:00/15.0","schedule_wednesday":"06:00/20.0 12:00/15.0 18:00/22.0 22:00/15.0","system_mode":"auto","window":"CLOSE","window_detection":"OFF"}'
```

symeonn, 2024-12-12:
Were you able to solve it? I just got mine TRV and have the same issue.

mwasowski, 2024-12-13:
> Were you able to solve it? I just got mine TRV and have the same issue.

Hey. That's not something you solve :) Converter is not working correctly for this TRV, despite it marked as supported. Sadly I didn't have time to look at it and fix it on my own :/ Maybe I'll have more time around Christmas/New Year.

nkznkz, 2024-12-18:
https://github.com/eteodun/Avatto-TRV07-TS0601-_TZE200_bvrlmajk

IgorSimic, 2024-12-24:
> https://github.com/eteodun/Avatto-TRV07-TS0601-_TZE200_bvrlmajk

Will be this convertor added to mainline?

mwasowski, 2025-01-05:
> https://github.com/eteodun/Avatto-TRV07-TS0601-_TZE200_bvrlmajk

I tried this during holidays. Removed the device, added converter, repaired and..it didn't work :/ I got `nulls` on every property 🤔

symeonn, 2025-01-05:
> > https://github.com/eteodun/Avatto-TRV07-TS0601-_TZE200_bvrlmajk
> 
> I tried this during holidays. Removed the device, added converter, repaired and..it didn't work :/ I got `nulls` on every property 🤔

if you already have Zigbee2MQTT 2.0.0 try this one, worked for me:
https://github.com/Koenkk/zigbee2mqtt/issues/25447#issuecomment-2571265075

mwasowski, 2025-01-05:
> if you already have Zigbee2MQTT 2.0.0 try this one, worked for me:
> https://github.com/Koenkk/zigbee2mqtt/issues/25447#issuecomment-2571265075

Thanks, I'm on 1.42.0 still ;) Will look at it when I upgrade. Still..i believe this should be in the build in converter..

IgorSimic, 2025-01-06:
@symeonn 
Strange, it works for me iin both 1.4x and 2.0.0
Did you restart zigbee2mqtt after adding converter?
Apparently, external converters are loaded during startup.

mwasowski, 2025-01-06:
> @symeonn 
> Strange, it works for me iin both 1.4x and 2.0.0
> Did you restart zigbee2mqtt after adding converter?
> Apparently, external converters are loaded during startup.

Yeah, did that. And also removed and paired the TRV again. Oh well ;)

IgorSimic, 2025-01-06:
Ha, I have another idea. Try delete it in zigbee2mqtt and pair it via another router, preferably tuya one if you have it. I had some situations with compatibility routers and end devices.

mwasowski, 2025-02-11:
Is there a chance this converter will end up in main branch?

mwasowski, 2025-03-11:
Ok, Since I migrated to HomeAssistant Yellow and it's kinda easier for me to tinker with Z2M now I gave it a shot again. Converter was loaded as I see the three additional settings at the bottom (Display brightness, Screen orientation and Mode). But for Position I still see `N/A%` and for Fault alarm `N/A`.

Any ideas?

symeonn, 2025-03-11:
> Ok, Since I migrated to HomeAssistant Yellow and it's kinda easier for me to tinker with Z2M now I gave it a shot again. Converter was loaded as I see the three additional settings at the bottom [...]

Try to re-pair the device.

mwasowski, 2025-03-11:
> Try to re-pair the device.

Did that, without removing, but I did. Same. Maybe it's stupid, but do I need to remove it?

mwasowski, 2025-05-11:
Not stale, just forgotten 😢

mwasowski, 2025-07-12:
Not stale. Will that converter ever make it to main branch?