Koenkk/zigbee2mqtt #24875: Aqara Vibration Sensor DJT11LM not working/exposing entities

lx05, 2024-11-22:
### What happened?

My Aqara Vibration Sensor DJT11LM does not report the values for Angle and Acceleration. Also the Cluster is completely empty. According to Github (#24185 ) it should be fixed in the newest version. I'm running version 1.41.0 which seems to be the latest. Im using SLZB-06 with FW-version 20240710.

[image]
[image]
[image]
[image]

### What did you expect to happen?

expose all values

### How to reproduce it (minimal and precise)

connect DJT11LM to Z2M

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

SLZB-06 with CC2652P chip

### Setup

HA on old Thinclient (x86) with Z2M addon

che1404, 2024-12-21:
Same issue here. Please let us know if there is anything we can do.

lx05, 2025-01-04:
For me, it started working when I changed the battery and waited for a little bit. It seems that the provided battery isn't full, and the sensor is very picky about that.

closed 2025-01-04 by lx05