Koenkk/zigbee2mqtt #24876: Sinope LM4110ZB Calculation

richterdc, 2024-11-22:
### What happened?

Forgive me if ive not posted this correctly, i didnt find a section that specifically called out a modification to a supported device.

For the Sinope LM4110ZB https://www.zigbee2mqtt.io/devices/LM4110ZB.html the "Tank Level" is based upon one of two gauge types that can be used. A "5-95" or "10-80", currently it is defaulted to "10-80" when it onboards and there is no way to change this. (in my case i have the 5-95 gauge).

It appears the current fzlocal of "tank_level" value is a calculated value based on the 10-80 gauge calculation. Happy to help test, work on the calculation etc. with some guidance. 

"10-80" Values.
xMin = 110;
xMax = 406;
delta = 46;
lower_limit = 10;
upper_limit = 80;

"5-95" Values.
x_Min = 110;
x_Max = 415;
delta = 55;
lower_Limit = 5;
upper_Limit = 95;

I would think having a selection to choose which values.

localfz when values are based on the 5-95 values:

`    tank_level: {
        cluster: 'genAnalogInput',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result: KeyValue = {};
            if (msg.data.presentValue !== undefined) {
                let x = msg.data['presentValue'];
                if (x == -1) {
                    result.tank_level = 0;
                } else {
                    const xMin = 110;
                    const xMax = 415;
                    const delta = 55;
                    if (delta <= x && x <= 70) {
                        x = delta;
                    }
                    if (0 <= x && x <= delta) {
                        x = x + 360;
                    }
                    const y = (x - xMin) / (xMax - xMin);
                    const lowerLimit = 5;
                    const upperLimit = 95;
                    const valueRange = upperLimit - lowerLimit;
                    const pct = y * valueRange + lowerLimit;

                    result.tank_level = utils.precisionRound(pct, 2);
                }
            }
            return result;
        },
    } satisfies Fz.Converter,`

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20240710

### Adapter

SLZB-06 - CC2652P

### Setup

Pi4 Containerized

cwr5033, 2024-12-05:
I have this device setup with mqtt amd did not realize this issue existed until today when my propane tank ran empty while reporting 27% tank_level value. I had the tank filled and now says 80% on the physical gauge, but reporting 67% to tank_level. Happy to test as well, would love an accurate reading :)

closed 2025-02-12 by github-actions[bot]

richterdc, 2025-02-18:
Can we re-open this?