Koenkk/zigbee2mqtt #24877: Bosch BTH-RA heats to full load all time after update

lllluke, 2024-11-22:
### What happened?

I was offered an update (906499348) for the Bosch thermostats via Zigbee2Mqtt. I installed it. Since then, the thermostats have been heating permanently and no longer regulate after the target temperature has been reached. Is this a bug in the firmware? Does anyone else have this problem related to Zigbee2Mqtt?

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

7.4.4

### Adapter

ember

### Setup

add-on, vm

salopette, 2024-11-23:
You have to repeat the valve calibration!

FrankSchoene, 2024-11-23:
> You have to repeat the valve calibration!

Yep. That helps :-)
But what hast changed in this Release? Could Not find any information.

dasEtwasmitPunkten, 2024-11-23:
> > You have to repeat the valve calibration!
> 
> Yep. That helps :-) But what hast changed in this Release? Could Not find any information.

There are release notes in German: https://www.bosch-smarthome.com/be/de/release-notes/bosch-smart-home-system/2024-11-20.html

FrankSchoene, 2024-11-24:
Thanks a lot. Not too much information there :-) Anyhow, I give it a try.

closed 2025-02-01 by github-actions[bot]