Koenkk/zigbee2mqtt #24880: Frontend does not start

McMornan, 2024-11-23:
### What happened?

I am running zigbee2mqtt using the official image from koenk from docker.io on a synology. (for 4 years now ...) Recently I updated the docker container (latest) and the frontend failed to start. Here is the log message:

[2024-11-23 10:14:15] error: 	z2m: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:114:18)
    at Controller.callExtensions (/app/lib/controller.ts:399:17)
    at Controller.stop (/app/lib/controller.ts:265:9)
    at stop (/app/index.js:165:5)
    at process.handleQuit (/app/index.js:173:13))

I checked with netstat and the frontend ist indeed not available on my designated port 8888. Here is the config;

homeassistant: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mqtt.zuhause
serial:
  port: /dev/ttyUSB0
frontend:
    port: 8888
    host: 0.0.0.0
advanced:
  log_level: warn
  last_seen: ISO_8601_local
  pan_id: 4711
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  report: true
  ikea_ota_use_test_url: true
  cache_state_send_on_startup: false
  cache_state_persistent: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
devices:
  '0x90fd9ffffe5a698c':
    friendly_name: IKEADimmer1
  '0x90fd9ffffed58e9a':
    friendly_name: IKEALampe1

.... and many more lines, I haev over 200 devices.

I checked the documentation and I am not aware that I have changed anything in my configuration,

The system itself starts and zigbee2mqtt itself is working fine. 

### What did you expect to happen?

I expected the frontend to start as it has in the past years

### How to reproduce it (minimal and precise)

In my configuration it fails to start reliably ...

### Zigbee2MQTT version

latest (1.41.0)

### Adapter firmware version

znp_CC2652RB_20200715.bin

### Adapter

Slaesh'S CR2652RB Stick

### Setup

official docker image from docker.io on  a Synology 415 (20 GB ram)

vdbrink, 2024-11-24:
I  had the same error. With the same stick and the same Z2M version 1.41.0

I see on that mentioned file /app/lib/extension/frontend.ts:114 on the line it's related to websocket config.
The websocket use the `base_url` property from the frontend to init.
If I set this in my configuration.yaml to this:
```
frontend:
    base_url: /
``` 
The error is gone from the logs and after three minutes I see in the logs again the line `z2m: Started frontend on port ..` again. And indeed it works again and also my toggles works now also again in Home Assistant!

The documentation state the `base_url` has default this value already but probably not (anymore?).

McMornan, 2024-11-28:
Hi, 

many thanks for your reply, but that did not fix it.

Lars

McMornan, 2024-11-28:
Hi, I have found exactly my issue in the official frontend thread: https://github.com/Koenkk/zigbee2mqtt/issues/4266#issuecomment-1435523061

There the poster said it's not connected to MQTT ... but mit container has a connection and is happily sending and receiving MQTT messages - still no frontend.

ldeutsch@Horizon:~$ sudo docker exec -it zigbee2mqtt /bin/netstat -an
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 172.17.0.2:34120        192.168.2.5:1883        ESTABLISHED
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node Path

closed 2025-02-04 by github-actions[bot]

McMornan, 2025-05-03:
Hi, I did update the docker container again today and suddenly the frontend was available again. I did not change my configuration sind opening this thread. Whatever it was ...