Koenkk/zigbee2mqtt #24882: Support negative numbers being set as strings

Crazyphil, 2024-11-23:
### What happened?

I own an __TZE200_c88teujp_ ([Saswell SEA802](https://www.zigbee2mqtt.io/devices/SEA801-Zigbee_SEA802-Zigbee.html)) TRV and recently started with temperature offset calibration. When tinkering, I noticed that setting the calibration value (`local_temperature_calibration`) only worked for positive values. Whenever I published a negative value via MQTT, the value was always reported back both to the z2m UI and MQTT as _4,294,967,285_ (or _0xFFFFFFF5_ in hex).
Setting the value via the z2m UI worked perfectly well, the value was set correctly (the measured temperature value was corrected by the configured amount) and also reflected to the UI and MQTT as it should.

After further investigation, I noticed that I was sending the MQTT value as a string value, for example for an offset of -2°C, as follows:

Topic: `zigbee2mqtt/SaswellThermostatOffice/set`
Payload: `{ "local_temperature_calibration" : "-2" }`

After noticing that the number is sent as a string, I tried removing the parantheses and simply sending the JSON value as a number, which immediately alleviated the problem and resulted in correct negative values to be set:

Payload: `{ "local_temperature_calibration" : -2 }`

### What did you expect to happen?

Setting positive temperature calibration values works both when sending them as JSON string values as well as JSON number values. This should also be the case for negative numbers (or not allowing strings for numeric values at all), treating all numbers equally and not cause confusion for users.

### How to reproduce it (minimal and precise)

Send negative numbers to a numeric topic (I assume this might work for other values as well and not only Saswell temperature offsets) and see the actual value be corrected to a very high number.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Manual install in /opt/zigbee2mqtt on Armbian on CubieTruck

### Debug log

[log.log](https://github.com/user-attachments/files/17880006/log.log)

MaxRower, 2024-12-13:
It seems, some upgrade of z2m in the past removed all my calibrations from all my sensors!
And now I can set only positive whole values, unusable!
Please fix that.

majkers, 2025-02-12:
Stilo exists

JonathanJorstad, 2025-02-19:
I am sorry if this is not really the same as your issue, but I am able to set negative temperature offsets in zigbee2mqtt UI under "Settings (specific)". I can see in the log afterwards that the value is sent as a number and not a string (i.e. without "")
It is an Aqara WSDCGQ12LM sensor (https://www.zigbee2mqtt.io/devices/WSDCGQ12LM.html#aqara-wsdcgq12lm)
zigbee2mqtt version 2.1.1, add-on to home assistant on a rpi4

majkers, 2025-04-22:
still exists

closed 2025-06-29 by github-actions[bot]