Koenkk/zigbee2mqtt #24886: [New device support]: Aeotoec Router Zigbee range extender ZI WG001 not supported

Matmat34, 2024-11-23:
### Link

https://aeotec.com/products/aeotec-range-extender-zi/

### Database entry

{     "device": {         "friendlyName": "0x2c1165fffe81abff",         "ieeeAddr": "0x2c1165fffe81abff",         "manufacturerID": 4619,         "networkAddress": 60815,         "type": "EndDevice"     },     "last_seen": "2024-11-23T15:50:16.554Z",     "linkquality": 255

### Zigbee2MQTT version

2024-09-27 

### Comments

Hello
I tried to pair a Zigbee router aeotoc ZI WG001 to extend my trafic. But it seems this Exeter is not supported by zigbee2mqtt while it was the case with the Zigbee plugin. Can you help me ?
Thank you so much.
Matthieu

### External definition

```shell
I don't know how proceed for a router.
```

### What does/doesn't work with the external definition?

I don't made an external definition.

chris-1243, 2024-11-23:
I would suggest to check #23916

This device is well supported. Unfortunatelly, it is hard to join. You will need multiple attempts to get it paired.

Check as well the [documentation](https://www.zigbee2mqtt.io/devices/WG001.html). According to it, you must be as close as possible of your coordinator

Matmat34, 2024-11-24:
Thank you for your answer.

This what I get when I try to pair.

￼

Can you tell me if for you it is ok, because Z2M said this material is not recognized.

Thanks a lot for your support.

chris-1243, 2024-11-24:
I do not use this device. I helped some users to have it working on their setup. What I would recommend for you:
 - Factory reset your device;
 - Bring it as close as possible of your coordinator (less than 50cm)
 - Enable `permit join` through your coordinator only;
 - Be **patient** as you may need several tries until it is paired correctly
 
When paired, you should be able to move your device where you wish to have it. Read carefully the user manual. Some of my tips come from it.

closed 2025-02-01 by github-actions[bot]

topa-LE, 2025-02-27:
No works with Vers. 2.x.