Koenkk/zigbee2mqtt #24887: Failed To configure SONOFF SNZB-06P - EMBER

ortofan, 2024-11-23:
### What happened?

Latest everything using EMBER, Failed fo configure presence sensors errors.

### What did you expect to happen?

No errors and relaible operations

### How to reproduce it (minimal and precise)

Simply pair the deives, once it has a few end-nodes hanging off of it the log is filling up with the error and endnodes connected to it it are unrelaible in operation.

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

v2.5.8

### Adapter

SMLIGHT SLZB-06M + darkxst 7.4.4 

### Setup

HAOS VM

### Debug log

error 2024-11-23 18:16:53z2m: Failed to configure 'Jelenlét - FoE : Iroda -', attempt 3 (Error: ZCL command 0x0ceff6fffe6bf09f/1 msOccupancySensing.configReport([{"minimumReportInterval":10,"maximumReportInterval":60,"reportableChange":0,"attribute":"occupancy"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":54552,"apsFrame":{"profileId":260,"clusterId":1030,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":122},"zclSequence":7,"commandIdentifier":7} timed out after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:195:23) at EmberOneWaitress.startWaitingFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:219:50) at /app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2131:55 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at EmberAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2070:16) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17))

tamet83, 2024-11-28:
``z2m: Failed to configure 'Presenza sala', attempt 5 (Error: ZCL command 0x0ceff6fffe4b831a/1 msOccupancySensing.configReport([{"minimumReportInterval":10,"maximumReportInterval":60,"reportableChange":0,"attribute":"occupancy"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":3522,"apsFrame":{"profileId":260,"clusterId":1030,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":66},"zclSequence":32,"commandIdentifier":7} timed out after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:195:23) at EmberOneWaitress.startWaitingFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:219:50) at /app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2131:55 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at EmberAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2070:16) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17))``

tsiapkinis, 2024-11-30:
I have the same error message:

# What happened?

I assded the presence sensor SONOFF SNZB-06P via zigbee2mqtt. It seems to work so far, but I got this confusing error message

# What did you expect to happen?

No error.

# How to reproduce it (minimal and precise)

Pair the device in zigbee2Mqtt. Error message appears after a couple of seconds

# Zigbee2MQTT version

1.41.0-1
# Adapter firmware version
- Sonoff-E (Zigbee 3.0 USB Dongle Plus)
- Driver: 6.7.0.0
- Firmware: newest, need to check

# Setup
HAOS over RPi 4 model B

# Debug log

```
2024-11-30 13:19:57z2m: Failed to configure 'presenceSensor', attempt 1 (Error: ZCL command 0x0ceff6fffe597033/1 msOccupancySensing.configReport([{"minimumReportInterval":10,"maximumReportInterval":60,"reportableChange":0,"attribute":"occupancy"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":22111,"apsFrame":{"profileId":260,"clusterId":1030,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":73},"zclSequence":3,"commandIdentifier":7} timed out after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:195:23) at EmberOneWaitress.startWaitingFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:219:50) at /app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2131:55 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at EmberAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2070:16) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17))
```

daubman, 2024-12-01:
I also have the same error with a 3rd reality garage door sensor using SLZB-06M in ember mode:

```
[2024-12-01 13:06:13] info: 	z2m: Zigbee2MQTT started!
[2024-12-01 13:07:56] info: 	z2m: Configuring 'garage-door-3rd-1'
[2024-12-01 13:08:07] error: 	z2m: Failed to configure 'garage-door-3rd-1', attempt 1 (Error: ZCL command 0x282c02bfffec9720/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":26562,"apsFrame":{"profileId":260,"clusterId":1,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":8},"zclSequence":7,"commandIdentifier":7} timed out after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:195:23)
    at EmberOneWaitress.startWaitingFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:219:50)
    at /app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2131:55
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at EmberAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2070:16)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17))
```

uffevedenbrant, 2024-12-03:
Similar error message and SNZB-06P goes "offline" and all entities stops existing. ( with 2 different sensors running 1.06 )
This with Z2M / Ember / SkyConnect with a ember compatible release , 7.4.2.0
Testing with ZHA in a different HA instance and the sensors comes alive and works perfect.
This with a different Aeotec stick running whatever release.
I can possibly test with Z2M and the Aeotec stick on this HA instance.

uffevedenbrant, 2024-12-03:
Tested now with Aeotec stick , EZSP , latest HA ( updated )  / z2m etc with the sensor as the only node in this Zigbee network
Seems like the sensor just stops answering.. Getting a number of lines like below
z2m: Failed to configure ... attempt 3

Note then that ZHA worked immediately.
To large message for the sensor to handle?

uffevedenbrant, 2024-12-04:
Upgraded SkyConnect to 7.4.4 , Still same issue. The sensor is discovered but refuses to respond and after a while is considered to be dead,

fernbrun, 2024-12-09:
Same issue for me using SMLIGHT SLZB-06M.

Rising-Sun, 2024-12-09:
Me too with a SMLIGHT SLZB-07p7 dongle.

huh123, 2024-12-12:
Added two new Sonoff SNZB-06P recently and both are giving timeouts for msOccupancySensing.configReport ZCL command. 

SNZB-06P firmware 1.0.6 
zStack3x0 with 20240710 firmware
1.42.0 zigbee2mqtt version running on LCX container. 

Tried to restart everything. No luck.

`2024-12-12 13:52:33z2m: Failed to configure 'z2m_presence_office_1', attempt 1 (Error: ZCL command 0x0ceff6fffe16bd61/1 msOccupancySensing.configReport([{"minimumReportInterval":10,"maximumReportInterval":60,"reportableChange":0,"attribute":"occupancy"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 58820 - 1 - 16 - 1030 - 7 after 10000ms) at Object.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:640:47) at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:672:28) at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:422:20 at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:420:16) at Request.send (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20) at Endpoint.zclCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9))`

uffevedenbrant, 2024-12-12:
Me testing with Zigbe2MQTT Edge now and and its seems to behave well without any strange errors.
I'm happy to have both my prod / staging / test HA devices. 
I.e. any changes and I do not need to care about a big ZigBee network..
Playing around it seems to work with ZHA as well as with Z2M Edge.
I.e. occupancy shows when I'm around and when not being around for the "timeout" it shows no occupancy.

tamet83, 2024-12-18:
This is the answer Sonoff gave me on Amazon market place:

The error message means that the connection is not stable, but you can ignore it unless the device is offline.

In my case all devices are within 5 meters from the coordinator so I think is not a stability problem.
Maybe someone form z2m can provide any further hypothesis.

Rising-Sun, 2024-12-26:
Disabling occupancy reporting fixes the error. I have created a pull request.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/8521

Koenkk, 2024-12-26:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/8521, thanks @Rising-Sun !

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2024-12-26 by Koenkk