Koenkk/zigbee2mqtt #24888: herdsman didn't start - "No Path"

docugraf, 2024-11-23:
### What happened?

I'm starting a fresh Home assistant build from scratch with a SLZB06M Zigbee IP Stick.

The SLZB is connected via switch to the HA pc and found by the HA add-on, latest updates are made via HA.
But Z2M won't start the Herdman process, I get this error:
----
z2m: Error: No path provided and failed to auto detect path
at Function.create (/app/node_modules/zigbee-herdsman/src/adapter/adapter.ts:107:23)
----
What path is this?

It makes no difference if using ezsp or ember, same error

### What did you expect to happen?

starting the herdsman

### How to reproduce it (minimal and precise)

just start the service

### Zigbee2MQTT version

 1.41.0-1 

### Adapter firmware version

2.1.7

### Adapter

eszp & ember

### Setup

HA add-on 

### Debug log

z2m: Error: No path provided and failed to auto detect path
at Function.create (/app/node_modules/zigbee-herdsman/src/adapter/adapter.ts:107:23)

andriikut, 2024-12-02:
@docugraf  did you installed Z2m update 1.42.0-1. ?  I did restore and now everything is working

tomknevitt, 2024-12-02:
I too have the same issue that has just started today. Not sure what has caused it, I have a uzg-01 and have come home to no zigbee.

[2024-12-02 20:39:57] info: 	z2m: Starting Zigbee2MQTT version 1.42.0 (commit #unknown)
[2024-12-02 20:39:57] info: 	z2m: Starting zigbee-herdsman (2.1.9)
[2024-12-02 20:39:57] error: 	z2m: Error while starting zigbee-herdsman
[2024-12-02 20:39:57] error: 	z2m: Failed to start zigbee
[2024-12-02 20:39:57] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-12-02 20:39:57] error: 	z2m: Exiting...
[2024-12-02 20:39:57] error: 	z2m: Error: No path provided and failed to auto detect path
    at Function.create (/app/node_modules/zigbee-herdsman/src/adapter/adapter.ts:107:23)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:133:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)

inomdzhon, 2024-12-07:
> I too have the same issue that has just started today. I have a uzg-01 and have come home to no zigbee. It was automatically updated to v1.42.0-2 and stopped working. I have now downgraded to v1.42.0-1 and is working again. Similar to OP I also have a network coordinator.
> 
> [2024-12-02 20:39:57] info: z2m: Starting Zigbee2MQTT version 1.42.0 (commit #unknown) [2024-12-02 20:39:57] info: z2m: Starting zigbee-herdsman (2.1.9) [2024-12-02 20:39:57] error: z2m: Error while starting zigbee-herdsman [2024-12-02 20:39:57] error: z2m: Failed to start zigbee [2024-12-02 20:39:57] error: z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions [2024-12-02 20:39:57] error: z2m: Exiting... [2024-12-02 20:39:57] error: z2m: Error: No path provided and failed to auto detect path at Function.create (/app/node_modules/zigbee-herdsman/src/adapter/adapter.ts:107:23) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:133:24) at Zigbee.start (/app/lib/zigbee.ts:69:27) at Controller.start (/app/lib/controller.ts:161:27) at start (/app/index.js:154:5)

+

same problems(

Nevga, 2024-12-29:
Same here, using a TI CC2652P7 via a local IP adress (version 1.42.0-2). I tried both connecting directly by

serial:
 port:tcp://fixed-ip-address:port

and via socat. Did anyone find a solution aside from downgrading? I'm on Home Assistant OS and downgrading is  not as easy there.

closed 2025-03-07 by github-actions[bot]