Koenkk/zigbee2mqtt #24889: (Error: Bind 0xa4c138b11d6cbb49/1  failed (Status 'INVALID_EP

h0m3d0m0, 2024-11-23:
### What happened?

Error on all my tuya wall switches.
Switches working in HA but is see this in the z2m log 

`Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2024-11-23 20:43:40] error: 	z2m: Failed to configure 'Schakelaar_Overloop', attempt 1 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2856:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[2024-11-23 20:43:42] error: 	z2m: Failed to configure 'Schakelaar_Overloop', attempt 2 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2856:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[2024-11-23 20:43:42] error: 	z2m: Failed to configure 'Schakelaar_Overloop', attempt 3 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2856:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[2024-11-23 20:43:47] error: 	z2m: Publish 'set' 'state' to 'Schakelaar_Eetplaats' failed: 'Error: ZCL command 0x540f57fffe2fcac0/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (15) Error: waiting for response TIMEOUT)'
[2024-11-23 20:43:48] error: 	z2m: Publish 'set' 'state' to 'Schakelaar_Eetplaats' failed: 'Error: ZCL command 0x540f57fffe2fcac0/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (18) Error: waiting for response TIMEOUT)'
[2024-11-23 20:43:49] error: 	z2m: Failed to configure 'Schakelaar__Badkamer', attempt 1[image]')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2856:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[2024-11-23 20:43:49] error: 	z2m: Failed to configure 'Schakelaar__Badkamer', attempt 2 (Error: Bind 0xa4c138b11d6cbb49/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2856:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))`
[image]

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Setup

Docker

### Debug log

[log.log](https://github.com/user-attachments/files/17882438/log.log)

fapgomes, 2024-12-21:
Maybe related with: https://github.com/Koenkk/zigbee2mqtt/issues/24585 ?

gianniacke, 2024-12-22:
No i have it on all my  wall switches, and one powerplugs 

`error 2024-12-22 15:48:15z2m: Failed to configure 'Schakelaar_Overloop', attempt 1 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
error 2024-12-22 15:48:23z2m: Publish 'set' 'state' to 'Schakelaar_Eetplaats' failed: 'Error: ZCL command 0x540f57fffe2fcac0/3 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (194) Error: waiting for response TIMEOUT)'
error 2024-12-22 15:48:23z2m: Publish 'set' 'state' to 'Schakelaar_Eetplaats' failed: 'Error: ZCL command 0x540f57fffe2fcac0/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (190) Error: waiting for response TIMEOUT)'
error 2024-12-22 15:48:23z2m: Publish 'set' 'state' to 'Powerplug_Tv_Boven' failed: 'Error: ZCL command 0x70ac08fffe5f8cb1/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (181) Error: waiting for response TIMEOUT)'
error 2024-12-22 15:48:25z2m: Failed to configure 'Schakelaar_Bureau', attempt 1 (Error: ZCL command 0xa4c13864ed20c78a/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (198) Error: waiting for response TIMEOUT) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:485:19) at runNextTicks (node:internal/process/task_queues:60:5) at processTimers (node:internal/timers:509:9) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:446:29) at Object.configureMagicPacket (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:453:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2934:13) at Configure.configure (/app/lib/extension/configure.ts:130:13))
error 2024-12-22 15:48:28z2m: Failed to configure 'Schakelaar_Overloop', attempt 2 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
error 2024-12-22 15:48:28z2m: Failed to configure 'Schakelaar_Overloop', attempt 3 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
error 2024-12-22 15:49:07z2m: Failed to configure 'Schakelaar_Masterroom', attempt 1 (Error: Bind 0xa4c1388f37d382d9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
error 2024-12-22 15:49:07z2m: Failed to configure 'Schakelaar_Masterroom', attempt 2 (Error: Bind 0xa4c1388f37d382d9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
[... cut ...]
error 2024-12-22 15:49:23z2m: Failed to configure 'Schakelaar__Badkamer', attempt 3 (Error: Bind 0xa4c138b11d6cbb49/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at Immediate.<anonymous> (/app/lib/extension/configure.ts:75:17))`

h0m3d0m0, 2024-12-22:
No i have it on all my  wall switches, and one powerplugs 

`error 2024-12-22 15:48:15z2m: Failed to configure 'Schakelaar_Overloop', attempt 1 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
error 2024-12-22 15:48:23z2m: Publish 'set' 'state' to 'Schakelaar_Eetplaats' failed: 'Error: ZCL command 0x540f57fffe2fcac0/3 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (194) Error: waiting for response TIMEOUT)'
error 2024-12-22 15:48:23z2m: Publish 'set' 'state' to 'Schakelaar_Eetplaats' failed: 'Error: ZCL command 0x540f57fffe2fcac0/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (190) Error: waiting for response TIMEOUT)'
error 2024-12-22 15:48:23z2m: Publish 'set' 'state' to 'Powerplug_Tv_Boven' failed: 'Error: ZCL command 0x70ac08fffe5f8cb1/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (181) Error: waiting for response TIMEOUT)'
error 2024-12-22 15:48:25z2m: Failed to configure 'Schakelaar_Bureau', attempt 1 (Error: ZCL command 0xa4c13864ed20c78a/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (198) Error: waiting for response TIMEOUT) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:485:19) at runNextTicks (node:internal/process/task_queues:60:5) at processTimers (node:internal/timers:509:9) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:446:29) at Object.configureMagicPacket (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:453:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2934:13) at Configure.configure (/app/lib/extension/configure.ts:130:13))
error 2024-12-22 15:48:28z2m: Failed to configure 'Schakelaar_Overloop', attempt 2 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
error 2024-12-22 15:48:28z2m: Failed to configure 'Schakelaar_Overloop', attempt 3 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
error 2024-12-22 15:49:07z2m: Failed to configure 'Schakelaar_Masterroom', attempt 1 (Error: Bind 0xa4c1388f37d382d9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
error 2024-12-22 15:49:07z2m: Failed to configure 'Schakelaar_Masterroom', attempt 2 (Error: Bind 0xa4c1388f37d382d9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
[... cut ...]
error 2024-12-22 15:49:23z2m: Failed to configure 'Schakelaar__Badkamer', attempt 3 (Error: Bind 0xa4c138b11d6cbb49/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13) at Configure.configure (/app/lib/extension/configure.ts:130:13) at Immediate.<anonymous> (/app/lib/extension/configure.ts:75:17))`

Koenkk, 2024-12-22:
Could you provide the data/database.db entry of this device?

h0m3d0m0, 2024-12-22:
@Koenkk 
Its more then 1 device, alle my wall switches get this error when i restart Z2MQTT
This is my complete DB, do you need to know the device names of the wall switches?

[database.zip](https://github.com/user-attachments/files/18223151/database.zip)

h0m3d0m0, 2024-12-22:
this is the log file when i reboot z2mqtt

`[2024-12-22 15:48:12] warning:  z2m: `permit_join` set to  `true` in configuration.yaml.
[2024-12-22 15:48:12] warning:  z2m: Allowing new devices to join.
[2024-12-22 15:48:12] warning:  z2m: Set `permit_join` to `false` once you joined all devices.
[2024-12-22 15:48:15] error:    z2m: Failed to configure 'Schakelaar_Overloop', attempt 1 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
[2024-12-22 15:48:23] error:    z2m: Publish 'set' 'state' to 'Schakelaar_Eetplaats' failed: 'Error: ZCL command 0x540f57fffe2fcac0/3 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (194) Error: waiting for response TIMEOUT)'
[2024-12-22 15:48:23] error:    z2m: Publish 'set' 'state' to 'Schakelaar_Eetplaats' failed: 'Error: ZCL command 0x540f57fffe2fcac0/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (190) Error: waiting for response TIMEOUT)'
[2024-12-22 15:48:23] error:    z2m: Publish 'set' 'state' to 'Powerplug_Tv_Boven' failed: 'Error: ZCL command 0x70ac08fffe5f8cb1/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (181) Error: waiting for response TIMEOUT)'
[2024-12-22 15:48:26] error:    z2m: Failed to configure 'Schakelaar_Bureau', attempt 1 (Error: ZCL command 0xa4c13864ed20c78a/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (198) Error: waiting for response TIMEOUT)
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:485:19)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at processTimers (node:internal/timers:509:9)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:446:29)
    at Object.configureMagicPacket (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:453:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2934:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13))
[2024-12-22 15:48:28] error:    z2m: Failed to configure 'Schakelaar_Overloop', attempt 2 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
[2024-12-22 15:48:28] error:    z2m: Failed to configure 'Schakelaar_Overloop', attempt 3 (Error: Bind 0xa4c1384d17e179a9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
[2024-12-22 15:49:07] error:    z2m: Failed to configure 'Schakelaar_Masterroom', attempt 1 (Error: Bind 0xa4c1388f37d382d9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
[2024-12-22 15:49:07] error:    z2m: Failed to configure 'Schakelaar_Masterroom', attempt 2 (Error: Bind 0xa4c1388f37d382d9/1 genOnOff from '0x00212effff093317/1' failed (Status 'INVALID_EP')
[... cut ...]
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2935:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:75:17))
[2024-12-22 17:16:04] warning:  zhc:ota:common: Firmware on '0xf082c0fffea076bf' (TRADFRI control outlet) is newer than latest firmware online.
[2024-12-22 17:16:12] warning:  zhc:ota:common: Firmware on '0xf082c0fffea02bf7' (TRADFRI control outlet) is newer than latest firmware online.
`

Koenkk, 2024-12-23:
Do these devices report a state change when controlled physically?

h0m3d0m0, 2025-01-17:
@Koenkk  

Sorry for the late replay had some private problems.
Just did the update to 2.0 but still the same error 
To answer your question yes everything reflect correct, if i push the button is see a change in the fronend and vice versa

Grts,

fapgomes, 2025-01-18:
@h0m3d0m0 Just to throw off the problem I had, can you try see if the "0" is there:

https://github.com/Koenkk/zigbee2mqtt/issues/24585#issuecomment-2585389169

If is there, can you try to remove it?

https://github.com/Koenkk/zigbee2mqtt/issues/24585#issuecomment-2585885594

h0m3d0m0, 2025-01-18:
@fapgomes 

I dont have the 0 in de eplist

`{"id":1,"type":"Coordinator","ieeeAddr":"0x00212effff093317","nwkAddr":0,"manufId":4405,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[0,10,25],"outClus>`

MrLivelong, 2025-02-03:
I have the same problem too with multiple Aqara plugs and smoke/heat detectors.

gersonfer, 2025-02-15:
Same problem here with z2m version 1.41.0

fapgomes, 2025-02-16:
@gersonfer is the same problem that I have? can you try see if the "0" is there:

https://github.com/Koenkk/zigbee2mqtt/issues/24585#issuecomment-2585389169

If is there, can you try to remove it?

https://github.com/Koenkk/zigbee2mqtt/issues/24585#issuecomment-2585885594

closed 2025-04-26 by github-actions[bot]