Koenkk/zigbee2mqtt #24895: Can't get zbdongle-e to work

bryantlee, 2024-11-24:
### What happened?

This is a brand new deployment of HASS on a Raspberry Pi.  I cannot get zigbee2mqtt add-on to start

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

rcp-uart-802154-v4.4.4-zbdonglee-460800.gbl

### Adapter

SONOFF zbdongle-e

### Setup

HASS OS Addon

### Debug log

z2m logs
```[20:44:12] INFO: Preparing to start...
[20:44:14] INFO: Socat not enabled
[20:44:21] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-11-23 20:44:50] info: 	z2m: Logging to console, file (filename: log.log)
[2024-11-23 20:44:50] info: 	z2m: Starting Zigbee2MQTT version 1.41.0 (commit #unknown)
[2024-11-23 20:44:50] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2024-11-23 20:44:53] info: 	zh:ember: Using default stack config.
[2024-11-23 20:44:53] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-11-23 20:44:53] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-11-23 20:44:53] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-23 20:44:53] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-11-23 20:44:53] info: 	zh:ember:uart:ash: Serial port opened
[2024-11-23 20:44:53] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-23 20:44:56] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-23 20:44:56] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-23 20:44:58] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-23 20:44:58] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-23 20:45:01] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-23 20:45:01] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-23 20:45:03] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-23 20:45:03] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-23 20:45:06] error: 	z2m: Error while starting zigbee-herdsman
[2024-11-23 20:45:06] error: 	z2m: Failed to start zigbee
[2024-11-23 20:45:06] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-11-23 20:45:06] error: 	z2m: Exiting...
[2024-11-23 20:45:07] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:684:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1598:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)```

Firmware flash logs
```
blee@blee-macbook-pro-14 bin % ./universal-silabs-flasher --device /dev/tty.usbserial-202305091606511 flash --firmware /Users/blee/Downloads/rcp-uart-802154-v4.4.4-zbdonglee-460800.gbl --allow-cross-flashing
2024-11-23 16:03:00.912 blee-macbook-pro-14.local universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.4', ezsp_version=None, ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.MULTIPAN: 'multipan'>, fw_variant=None, baudrate=460800)
2024-11-23 16:03:00.912 blee-macbook-pro-14.local universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-11-23 16:03:04.949 blee-macbook-pro-14.local universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2024-11-23 16:03:07.009 blee-macbook-pro-14.local universal_silabs_flasher.flasher INFO Detected ApplicationType.CPC, version '4.3.1' at 460800 baudrate (bootloader baudrate None)
2024-11-23 16:03:09.570 blee-macbook-pro-14.local universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-11-23 16:03:11.614 blee-macbook-pro-14.local universal_silabs_flasher.flasher INFO Detected bootloader version '1.12.0'
2024-11-23 16:03:11.615 blee-macbook-pro-14.local universal_silabs_flasher.flasher INFO Detected ApplicationType.GECKO_BOOTLOADER, version '1.12.0' at 115200 baudrate (bootloader baudrate 115200)
rcp-uart-802154-v4.4.4-zbdonglee-460800.gbl  [####################################]  100%
```

HASSOS add-on config:
```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  adapter: ember
```

bryantlee, 2024-11-24:
I have also tried to directly address the port of the USB under serial, but same error occurs.  

This is using a local MQTT broker on HASS OS.

chris-1243, 2024-11-24:
> rcp-uart-802154-v4.4.4-zbdonglee-460800.gbl

You are trying to use a multi-protocol firmware. If I am not wrong, there are still issue with the multi-protocol addon and support in HASS OS.
I would suggest to flash a NCP firmware (zigbee only). You may have check there to get firmware for your Dongle-E

https://github.com/darkxst/silabs-firmware-builder/tree/4.4.4/firmware_builds/zbdonglee

https://github.com/Nerivec/silabs-firmware-builder/releases

vindaalex, 2024-11-24:
I just started using the e-donglen as well and got it first working by using `adapter: ezsp`. This however deprecated and then updated the e-dongle firmware via [sonoff guide](https://sonoff.tech/wp-content/uploads/2023/02/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing.pdf) to version 7.4.3 and then `adapter: ember` is working

bryantlee, 2024-11-25:
> > rcp-uart-802154-v4.4.4-zbdonglee-460800.gbl
> 
> 
> 
> You are trying to use a multi-protocol firmware. If I am not wrong, there are still issue with the multi-protocol addon and support in HASS OS.
> 
> I would suggest to flash a NCP firmware (zigbee only). You may check there to get firmware for your Dongle-E
> 
> 
> 
> https://github.com/darkxst/silabs-firmware-builder/tree/4.4.4/firmware_builds/zbdonglee
> 
> 
> 
> https://github.com/Nerivec/silabs-firmware-builder/releases

Thank you! You were spot on. I'm up and running. Thanks!