Koenkk/zigbee2mqtt #24898: scene_recall does not work for shades

RedNo7, 2024-11-24:
### What happened?

I have created a group of 8 shades
I have created 3 scenes within that group

However:
- The scenes save the state but not the '% Position'.
- Even though it saves and recalls the state correctly, the shades do not move.

Each shade (IKEA and Sonoff) works as expected individually.

### What did you expect to happen?

I expect the shade position to be recalled for each scene.
I expect the (actual) shades to move to the recalled position.

### How to reproduce it (minimal and precise)

Create a group
Add some shades
Set a scene
Move the shades
Recall the scene

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20221226

### Adapter

Sonoff dongle-P

### Setup

MQTT and HA in containers on an R-Pi (on a Debian variant called DietPI) 

### Debug log

[log.log](https://github.com/user-attachments/files/17892736/log.log)

closed 2025-03-22 by github-actions[bot]

cyanic-selkie, 2026-02-02:
I've encountered the same problem on 2.8.0.

RedNo7, 2026-02-03:
Interesting.  I wonder what's the point of a device supporting a group if doesn't also support a scene.

I assumed the group & scene logic was all in Z2M which translates the scene recall into a series of instructions to the individual devices in that group or scene i.e. the end device (battery or no) simply gets a command and doesn't even need to be aware about groups and/or scenes.